projectacrn/acrn-hypervisor #3721: [Compiling Issue] Error implicit declaration with VIRTIO_PCI_CONFIG_OFF

foreverchina, 2019-09-23:
Hi sir，

I used the latest acrn code to compling the apl-mrb images with the command "make O=build-sbl BOARD=apl-mrb",  it happen the error under the path: hw/pci/virtio/virtio.c. The detail steps is folling:
1、git clone the latest code.
2、type the compling command "make O=build-sbl BOARD=apl-mrb" in the root path.

the wrong message is folling:
devicemodel/include -I/root/hypervisor/acrn-hypervisor/build-sbl/misc/tools -fstack-protector -DDM_DEBUG -c hw/pci/virtio/virtio.c -o /root/hypervisor/acrn-hypervisor/build-sbl/devicemodel/hw/pci/virtio/virtio.o
hw/pci/virtio/virtio.c: In function ‘virtio_set_io_bar’:
hw/pci/virtio/virtio.c:220:2: error: implicit declaration of function ‘VIRTIO_PCI_CONFIG_OFF’ [-Werror=implicit-function-declaration]
  size = VIRTIO_PCI_CONFIG_OFF(1) + base->vops->cfgsize;
  ^
hw/pci/virtio/virtio.c: In function ‘virtio_set_modern_mmio_bar’:
hw/pci/virtio/virtio.c:1058:9: error: variable ‘cfg’ has initializer but incomplete type
  struct virtio_pci_cfg_cap cfg = {
         ^
hw/pci/virtio/virtio.c:1059:3: error: unknown field ‘cap’ specified in initializer
   .cap.cap_vndr = PCIY_VENDOR,
   ^
hw/pci/virtio/virtio.c:1059:3: error: excess elements in struct initializer [-Werror]
hw/pci/virtio/virtio.c:1059:3: error: (near initialization for ‘cfg’) [-Werror]
hw/pci/virtio/virtio.c:1060:3: error: unknown field ‘cap’ specified in initializer
   .cap.cap_next = 0,
   ^
hw/pci/virtio/virtio.c:1060:3: error: excess elements in struct initializer [-Werror]
hw/pci/virtio/virtio.c:1060:3: error: (near initialization for ‘cfg’) [-Werror]
hw/pci/virtio/virtio.c:1061:3: error: unknown field ‘cap’ specified in initializer
   .cap.cap_len = sizeof(cfg),
   ^
hw/pci/virtio/virtio.c:1061:24: error: invalid application of ‘sizeof’ to incomplete type ‘struct virtio_pci_cfg_cap’
   .cap.cap_len = sizeof(cfg),
                        ^
hw/pci/virtio/virtio.c:1061:3: error: excess elements in struct initializer [-Werror]
   .cap.cap_len = sizeof(cfg),
   ^
hw/pci/virtio/virtio.c:1061:3: error: (near initialization for ‘cfg’) [-Werror]
hw/pci/virtio/virtio.c:1062:3: error: unknown field ‘cap’ specified in initializer
   .cap.cfg_type = VIRTIO_PCI_CAP_PCI_CFG,
   ^
hw/pci/virtio/virtio.c:1062:3: error: excess elements in struct initializer [-Werror]
hw/pci/virtio/virtio.c:1062:3: error: (near initialization for ‘cfg’) [-Werror]
hw/pci/virtio/virtio.c:1058:28: error: storage size of ‘cfg’ isn’t known
  struct virtio_pci_cfg_cap cfg = {
                            ^
hw/pci/virtio/virtio.c: At top level:
cc1: error: unrecognized command line option "-Wno-address-of-packed-member" [-Werror]
cc1: error: unrecognized command line option "-Wno-stringop-truncation" [-Werror]
cc1: all warnings being treated as errors
make[1]: *** [/root/hypervisor/acrn-hypervisor/build-sbl/devicemodel/hw/pci/virtio/virtio.o] Error 1
make[1]: Leaving directory `/root/hypervisor/acrn-hypervisor/devicemodel'
make: *** [devicemodel] Error 2

Where can I find the definition VIRTIO_PCI_CONFIG_OFF in the full source code?

Thanks
Jackie

fuzhongl, 2019-09-24:
@foreverchina 
We can't reproduce this issue for our CI system; Can you please share more info about your build environment?
Thanks!

foreverchina, 2019-09-25:
@fuzhongl Thank you very much for your quickly response. My system info is following:
LSB Version:    :core-4.1-amd64:core-4.1-noarch
Distributor ID: CentOS
Description:    CentOS Linux release 7.7.1908 (Core)
Release:        7.7.1908
Codename:       Core
========================
Listing versions of doc build dependencies found on your system...

  breathe                   version: 4.13.1
  sphinx                    version: 2.2.0
  docutils                  version: 0.15.2
  sphinx-rtd-theme          version: 0.4.3
  kconfiglib                version: 12.14.0
  doxygen                   version: 1.8.17 (1024ab18b435dac7ba4ab94a087bb950b3bdf2c0)
========================
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-redhat-linux/4.8.5/lto-wrapper
Target: x86_64-redhat-linux
Configured with: ../configure --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-bootstrap --enable-shared --enable-threads=posix --enable-checking=release --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-linker-hash-style=gnu --enable-languages=c,c++,objc,obj-c++,java,fortran,ada,go,lto --enable-plugin --enable-initfini-array --disable-libgcj --with-isl=/builddir/build/BUILD/gcc-4.8.5-20150702/obj-x86_64-redhat-linux/isl-install --with-cloog=/builddir/build/BUILD/gcc-4.8.5-20150702/obj-x86_64-redhat-linux/cloog-install --enable-gnu-indirect-function --with-tune=generic --with-arch_32=x86-64 --build=x86_64-redhat-linux
Thread model: posix
gcc version 4.8.5 20150623 (Red Hat 4.8.5-39) (GCC)
========================
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-redhat-linux/4.8.5/lto-wrapper
Target: x86_64-redhat-linux
Configured with: ../configure --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-bootstrap --enable-shared --enable-threads=posix --enable-checking=release --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-linker-hash-style=gnu --enable-languages=c,c++,objc,obj-c++,java,fortran,ada,go,lto --enable-plugin --enable-initfini-array --disable-libgcj --with-isl=/builddir/build/BUILD/gcc-4.8.5-20150702/obj-x86_64-redhat-linux/isl-install --with-cloog=/builddir/build/BUILD/gcc-4.8.5-20150702/obj-x86_64-redhat-linux/cloog-install --enable-gnu-indirect-function --with-tune=generic --with-arch_32=x86-64 --build=x86_64-redhat-linux
Thread model: posix
gcc version 4.8.5 20150623 (Red Hat 4.8.5-39) (GCC)
=============================
Anyway, I can upload the full building log, but I don't know how to attach the attachement .Could you pls help to check it again?

thanks
Jackie

foreverchina, 2019-09-25:
@fuzhongl  Just now I update the gcc to  7.3.1, it get the same wrong, it can't pass the "virtio.c".
==================
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/opt/rh/devtoolset-7/root/usr/libexec/gcc/x86_64-redhat-linux/7/lto-wrapper
Target: x86_64-redhat-linux
Configured with: ../configure --enable-bootstrap --enable-languages=c,c++,fortran,lto --prefix=/opt/rh/devtoolset-7/root/usr --mandir=/opt/rh/devtoolset-7/root/usr/share/man --infodir=/opt/rh/devtoolset-7/root/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-shared --enable-threads=posix --enable-checking=release --enable-multilib --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-gcc-major-version-only --enable-plugin --with-linker-hash-style=gnu --enable-initfini-array --with-default-libstdcxx-abi=gcc4-compatible --with-isl=/builddir/build/BUILD/gcc-7.3.1-20180303/obj-x86_64-redhat-linux/isl-install --enable-libmpx --enable-gnu-indirect-function --with-tune=generic --with-arch_32=i686 --build=x86_64-redhat-linux
Thread model: posix
gcc version 7.3.1 20180303 (Red Hat 7.3.1-5) (GCC)
==================
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/opt/rh/devtoolset-7/root/usr/libexec/gcc/x86_64-redhat-linux/7/lto-wrapper
Target: x86_64-redhat-linux
Configured with: ../configure --enable-bootstrap --enable-languages=c,c++,fortran,lto --prefix=/opt/rh/devtoolset-7/root/usr --mandir=/opt/rh/devtoolset-7/root/usr/share/man --infodir=/opt/rh/devtoolset-7/root/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-shared --enable-threads=posix --enable-checking=release --enable-multilib --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-gcc-major-version-only --enable-plugin --with-linker-hash-style=gnu --enable-initfini-array --with-default-libstdcxx-abi=gcc4-compatible --with-isl=/builddir/build/BUILD/gcc-7.3.1-20180303/obj-x86_64-redhat-linux/isl-install --enable-libmpx --enable-gnu-indirect-function --with-tune=generic --with-arch_32=i686 --build=x86_64-redhat-linux
Thread model: posix
gcc version 7.3.1 20180303 (Red Hat 7.3.1-5) (GCC)

thanks
jackie

leilu, 2019-09-25:
@foreverchina , We build ACRN in Clear Linux, you may refer the [guide](https://projectacrn.github.io/latest/tutorials/building_acrn_in_docker.html) to use clearlinux docker image to build.

foreverchina, 2019-09-25:
@leilu Thank, I will follpw your instruction to comipling the imags.

thanks
jackie

foreverchina, 2019-09-25:
@fuzhongl @leilu  The docker work fine. it fixed  this issue. thanks a lot. but it seems there is no environment with compling 'doc' target .

thanks
jackie

closed 2019-09-25 by foreverchina

fuzhongl, 2019-09-25:
@foreverchina Currently, docker doesn't support compiling 'doc' target.
Code related changes don't need to compile 'doc'.
Do you plan to raise Doc PR?
Thanks!

foreverchina, 2019-09-25:
@fuzhongl Thanks for your info, No, until now I don't plan to raise a doc PR.

thanks
Jackie