projectacrn/acrn-hypervisor #3725: Add support for memory bandwidth allocation (MBA)

vijaydhanraj, 2019-09-23:
If memory bandwidth allocation is supported by the platform, add support in ACRN to limit the memory bandwidth for a particular partition.

Mingyuan18, 2019-09-27:
[External_System_ID] ACRN-5327

Mingyuan18, 2019-11-19:
@vijaydhanraj , Kindly ask how to deal with this issue. Will you submit Patch?

vijaydhanraj, 2019-11-20:
Hi @Mingyuan18, currently client platforms don't support memory bandwidth allocation(MBA) feature. Only server platforms support this. But there are few issues running acrn on OTS server platforms and need few patches such as multiple IOMMU https://github.com/projectacrn/acrn-hypervisor/issues/4134 which are being reviewed. Once this is complete, I can test and submit my patches. 

PS: My changes have already been reviewed in the technical forum and are complete. Testing is pending.

NanlinXie, 2020-06-11:
Patch merged and no regression. Close it.

closed 2020-06-11 by NanlinXie