projectacrn/acrn-hypervisor #3757: Not able to boot Ubuntu as SOS

PO-88, 2019-09-26:
Hi,

I am trying to get ACRN hypervisor working with Ubuntu as SOS, I have followed the below link for my initial set-up.
https://projectacrn.github.io/latest/tutorials/using_ubuntu_as_sos.html

However is was not successful so far, so I'm looking for any info i can get to go further.
I have few questions and few observations made. I would appreciate any help that comes my way.
Q1. As stated in the above link is it necessary that i use SOS kernel linux-iot-lts2018-sos-4.19.0-22.x86_64.rpm?
Observation made: with linux-iot-lts2018-sos-4.19.0-22.x86_64.rpm, i hit on a blank screen as discussed in other posts with other distro's(which actually dint help me with info)
I tried this linux-iot-lts2018-sos-4.19.73-92.x86_64.rpm and i was able to see some logs come on screen, but it gets stuck with below message
waiting for root device PARTUUID=<UUID of / (root) partition on my machine>
i have cross checked and made sure that i have the right UUID and right SOS kernel version specified in /etc/grub.d/40_custom entry.

Q2. After reboot on to Ubuntu and I checked boot order with efibootmgr -v, it gets changed on. Does this imply that i need to update my EFI firmware?

Q3. Is it possible to get the source code for SOS kernel linux-iot-lts2018-sos-4.19.73-92.x86_64.rpm, a link to this branch would be helpful.

My system configuration:
i3-3220 (x86)
Manufacturer Acer
BIOS version POI-A0
RAM used 3GB

Thanks in Advance.

coneypo, 2019-09-26:
**Apollo Lake** and **Kaby Lake** platforms have been verified by us, and please refer to:
https://projectacrn.github.io/latest/reference/hardware.html

**I3-3220** is **Ivy Bridge** rather than **Apolly Lake** or **Kaby Lake**;

Maybe you can try setting up Ubuntu as SOS or UOS with Apolly or Kaby lake hardware platform, tks;

Mingyuan18, 2019-09-27:
[External_System_ID] ACRN-5325

PO-88, 2019-10-09:
Thanks for the info coneypo, i had only seen the minimum system requirements earlier and sorry for replying this late.
I don't have access to Apollo/Kaby lake platforms as of now. So i was trying this out on the Ivy Bridge. I have few questions, it would be great if i get info for these questions.

Q1. What HW and firmware dependency do we have to get the ACRN hypervisor working? Why is it restricted for Apollo/Kaby lake platforms?

Q2. Is there a particular version of UEFI firmware that supports ACRN? I'm not able to understand the dependency of UFEI firmware and the acrn.efi file. (As grub sceen wont show up and the system reboots)

Q3. As ACRN can be build locally, is there a way to change the configuration to get this working on other platforms (like ivy-bridge)? (menuconfig has no much info on this)

Thanks in advance!!

> **Apollo Lake** and **Kaby Lake** platforms have been verified by us, and please refer to:
> https://projectacrn.github.io/latest/reference/hardware.html
>
> **I3-3220** is **Ivy Bridge** rather than [...]

closed 2020-06-28 by fuzhongl