projectacrn/acrn-hypervisor #3770: Warning when building the ACRN hypervisor `SDC (defined at arch/x86/Kconfig:7) set more than once`

gvancuts, 2019-10-02:
The following warnings are emitted when building the ACRN hypervisor (`acrn-2019w38.1-140000p-95-g28b50463`), using the default values (i.e. `make clean && make`):
```
/home/gvancuts/work/acrn/acrn-hypervisor/build/hypervisor/.config:40: warning: SDC (defined at arch/x86/Kconfig:7) set more than once. Old value: "y", new value: "y".
/home/gvancuts/work/acrn/acrn-hypervisor/build/hypervisor/.config:40: warning: SDC (defined at arch/x86/Kconfig:7) set more than once. Old value: "y", new value: "y".
/home/gvancuts/work/acrn/acrn-hypervisor/build/hypervisor/.config:40: warning: SDC (defined at arch/x86/Kconfig:7) set more than once. Old value: "y", new value: "y".
/home/gvancuts/work/acrn/acrn-hypervisor/build/hypervisor/.config:40: warning: SDC (defined at arch/x86/Kconfig:7) set more than once. Old value: "y", new value: "y".
```

Mingyuan18, 2019-10-08:
[External_System_ID] ACRN-5339

gvancuts, 2019-10-08:
I'm OK being assigned to this but I will not have time to look at it for a week at least.

fuzhongl, 2020-05-05:
@gvancuts Can this git issue be closed?
Thanks!

gvancuts, 2020-05-05:
I have not checked this lately but the build system has changed so much that I think it's safe to close it. I can always re-open if really needed.

closed 2020-05-05 by gvancuts