projectacrn/acrn-hypervisor #3773: suspicious logic in vhost.c

lyakh, 2019-10-02:
When calculating features from different sources a logic like `a &= ~(b ^ c)` is used, with a comment, which seems to me to simply mean, that it should clear all bits in `a`, which are not included in either `b` or `c`. If I understood the comment correctly, then the formula seems wrong to me and it should simply be `a &= b & c`
https://github.com/projectacrn/acrn-hypervisor/blob/28b50463c9d6ec99c545b588406f7153a73c4a8d/devicemodel/hw/pci/virtio/vhost.c#L614

Mingyuan18, 2019-10-08:
[External_System_ID] ACRN-5341

fuzhongl, 2020-05-05:
@lyakh 
Please check Dev  comments as following:

**there are fearure bits in dev->base->device_caps that are not included by either vhost_features or features, for example those feature bits only available between FE and virtio BE, those bits cannot be cleared.**

Will close this git issue as "reject".

closed 2020-05-05 by fuzhongl

lyakh, 2020-06-02:
> @lyakh
> Please check Dev comments as following:
>
> **there are fearure bits in dev->base->device_caps that are not included by either vhost_features or features, for example those feature bits only [...]
@fuzhongl just to make sure we really mean the same thing, as far as I understand the current logic is the following (for any specific bit):
```
vhost_features	features	result	comment
0				0		kept		// i.e. if feature absent in both masks, it will be kept in caps, e.g. it might stay on
0				1		cleared	// in this case the caps bit will be cleared
1				0		cleared	// cleared again
1				1		kept		// the caps bit will be kept
```
is this what is intended?