projectacrn/acrn-hypervisor #3776: ACRN on Intel Nuc 8 and other plateform

limbo127, 2019-10-05:
Hello, 
Can someone explain us how acrn is supported platefome ?
We want to use on Intel nuc generation 8, without success.
We are trying to install on new Pc portable generation 8 or more ( dell, hp) withtout success.
Is a lot dev is necessary ?  how can we help ?  
Any roadmap about support new intel processor ? 

Regards, 
Nicolas

gvancuts, 2019-10-07:
Hi Nicolas, we're not officially supporting this platform although I would expect that it should run ACRN with minor tweaks. A few things can go wrong when trying out a new platform inc. things like the memory size you are using. The easiest to debug these is if you have a functional serial port that can at least give some clues as to what is going on inside ACRN. Do you have that?

limbo127, 2019-10-07:
No,
not serial, and my question is also relative to new hardware, we are
project to install acrn on laptop for example, how can new hardware can be
integrated?
regards,
Nicolas

Mingyuan18, 2019-10-08:
[External_System_ID] ACRN-5342

gvancuts, 2019-10-08:
For the very first porting to a CPU that's not been validated before, I would try first to find a HW platform that has a serial port to help debug issues pertaining to this new architecture. Once we know it runs correctly, we can look at other factors that may affect its ability to run on a different one (e.g. laptop), such as amount of memory, EFI bios differences, etc.

Can you give us the exact model of this NUC8 that you're using (and its memory configuration), I may be able to locate one here and try it out on my side.

limbo127, 2019-10-08:
Nuc8bek i5 proc for example, but Intel says us that's nuc8 will be soon
replace by intel nuc 10, it's our problematic, how can use acrn as simple
as linux kvm for example.
nicolas

limbo127, 2019-10-09:
> ACRN is not a straight replacement for KVM, it depends on the
> features/use-case you have in mind. Each will have their pros and cons.
> ACRN is not focused on working "out-of-the-box", but rather focuses on
> use-cases where embedded (IoT, Edge) features are required, such as
> real-time behaviour, or where Functional Safety is required for part of the
> system. One of the downside is that configuring ACRN to run correctly on
> some HW requires more attention and pre-configuration.
>
> Can you tell us a bit more about your use-case so we can determine if ACRN
> is a good fit for you?

We wants deploy a lot of laptop with 2 vms ( win10 + clear Linux + android
) using with  kvmGt to get performance and using image vm.
Our tests on Linux + Kvm is good but acrn has a very small footprint, and
can be boot very quickly.
Linux + Kvm need to be manage , we are thinking that acrn less.
Regards,
Nicolas

fuzhongl, 2020-08-14:
No update for long time, will close it.

closed 2021-03-15 by fuzhongl