projectacrn/acrn-hypervisor #3777: Tutorial "Using partition mode on UP2" no longer valid

ttzeng, 2019-10-07:
The tutorial "[Using partition mode on UP2](https://projectacrn.github.io/latest/tutorials/using_partition_mode_on_up2.html)" no longer valid, the code snippets were moved to somewhere? Please update the document as I see "LOGICAL_PARTITION" scenario is still supported. Thanks.

Mingyuan18, 2019-10-08:
[External_System_ID] ACRN-5343

ttzeng, 2019-10-08:
Can't build the hypervisor for logical partition mode with the following error:
```
$ make SCENARIO=logical_partition hypervisor
...
scenario/logical_partition/vm_configurations.h: 10:10: fatal error: pci_devices.h: No such file or directory
```

jsun26intel, 2019-10-09:
Current code only support logical partition mode on kbl-nuc-i7/apl-mrb/dnv-cb2, so we did not provide pci_devices.h for other boards. And officially we only support logical partition on kbl-nuc-i7 if I remember right, so doc of  "Using partition mode on UP2" should be updated. (@NanlinXie please correct me if I am wrong, otherwise we should provide pci_devices.h for all supported boards)

If you want to support logical partition on UP2, please use acrn-config tool to generate pci_devices.h, or make by following command:
make hypervisor BOARD_FILE=$PWD/misc/acrn-config/xmls/board-xmls/apl-up2.xml SCENARIO_FILE=$PWD/misc/acrn-config/xmls/config-xmls/apl-up2/logical_partition.xml

fuzhongl, 2019-12-16:
@ttzeng  Any more comments about using acrn-config tool to support logical partition on UP2?
If no, could you please close this issue?
Thanks!

closed 2019-12-16 by ttzeng