projectacrn/acrn-hypervisor #3848: Ethernet on Zephyr UOS

marjano89, 2019-10-16:
Hi, 
I am struggling with setting up network on Zephyr UOS under ACRN.  I am building simple MQTT client app. First problem was that there was not net interface:

`[00:00:00.000,000] <err> net_if: There is no network interface to work with!
[00:00:00.090,000] <inf> net_config: Initializing network
[00:00:00.090,000] <err> net_config: No network interfaces
[00:00:00.090,000] <err> net_config: Network initialization failed (-19)
[00:00:00.110,000] <err> mqtt_client: Failed to subscribe to test item`

I edited simple config to support ETH_E1000 which leads to compilation errors.

My config looks like this
```
CONFIG_NETWORKING=y
CONFIG_NET_SOCKETS=y
CONFIG_NET_TCP=y
CONFIG_NET_LOG=y

CONFIG_PRINTK=y
CONFIG_STDOUT_CONSOLE=y

CONFIG_ENTROPY_GENERATOR=y
CONFIG_TEST_RANDOM_GENERATOR=y

# Enable IPv6 support
CONFIG_NET_IPV6=n
# Enable IPv4 support
CONFIG_NET_IPV4=y

# Enable the MQTT Lib
CONFIG_MQTT_LIB=y

CONFIG_NET_CONFIG_SETTINGS=y

CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.168.2.13"
CONFIG_NET_CONFIG_PEER_IPV4_ADDR="192.168.2.10"

CONFIG_MAIN_STACK_SIZE=2048

# For IPv6
CONFIG_NET_BUF_DATA_SIZE=256

CONFIG_NET_L2_ETHERNET=y
#CONFIG_NET_QEMU_ETHERNET=y

CONFIG_ETH_E1000=y

CONFIG_PCIE=y
```

build errors:

```
                 from /home/michal/zephyr/zephyrproject/zephyr/drivers/ethernet/eth_e1000.c:12:
/home/michal/zephyr/zephyrproject/zephyr/drivers/ethernet/eth_e1000.c: In function 'e1000_init':
/home/michal/zephyr/zephyrproject/zephyr/drivers/ethernet/eth_e1000.c:200:32: error: 'DT_ETH_E1000_IRQ_PRIORITY' undeclared (first use in this function); did you mean 'UART_1_IRQ_PRIORITY'?
  IRQ_CONNECT(DT_ETH_E1000_IRQ, DT_ETH_E1000_IRQ_PRIORITY,
                                ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/michal/zephyr/zephyrproject/zephyr/include/arch/x86/ia32/arch.h:210:21: note: in definition of macro 'Z_ARCH_IRQ_CONNECT'
     [priority] "i" (priority_p), \
                     ^~~~~~~~~~
/home/michal/zephyr/zephyrproject/zephyr/drivers/ethernet/eth_e1000.c:200:2: note: in expansion of macro 'IRQ_CONNECT'
  IRQ_CONNECT(DT_ETH_E1000_IRQ, DT_ETH_E1000_IRQ_PRIORITY,
  ^~~~~~~~~~~
/home/michal/zephyr/zephyrproject/zephyr/drivers/ethernet/eth_e1000.c:200:32: note: each undeclared identifier is reported only once for each function it appears in
  IRQ_CONNECT(DT_ETH_E1000_IRQ, DT_ETH_E1000_IRQ_PRIORITY,
                                ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/michal/zephyr/zephyrproject/zephyr/include/arch/x86/ia32/arch.h:210:21: note: in definition of macro 'Z_ARCH_IRQ_CONNECT'
     [priority] "i" (priority_p), \
                     ^~~~~~~~~~
/home/michal/zephyr/zephyrproject/zephyr/drivers/ethernet/eth_e1000.c:200:2: note: in expansion of macro 'IRQ_CONNECT'
  IRQ_CONNECT(DT_ETH_E1000_IRQ, DT_ETH_E1000_IRQ_PRIORITY,
  ^~~~~~~~~~~
/home/michal/zephyr/zephyrproject/zephyr/drivers/ethernet/eth_e1000.c:200:14: error: 'DT_ETH_E1000_IRQ' undeclared (first use in this function); did you mean 'ETH_E1000_PRIV_H'?
  IRQ_CONNECT(DT_ETH_E1000_IRQ, DT_ETH_E1000_IRQ_PRIORITY,
              ^~~~~~~~~~~~~~~~
/home/michal/zephyr/zephyrproject/zephyr/include/arch/x86/ia32/arch.h:212:16: note: in definition of macro 'Z_ARCH_IRQ_CONNECT'
     [irq] "i" (irq_p)); \
                ^~~~~
/home/michal/zephyr/zephyrproject/zephyr/drivers/ethernet/eth_e1000.c:200:2: note: in expansion of macro 'IRQ_CONNECT'
  IRQ_CONNECT(DT_ETH_E1000_IRQ, DT_ETH_E1000_IRQ_PRIORITY,
  ^~~~~~~~~~~
/home/michal/zephyr/zephyrproject/zephyr/drivers/ethernet/eth_e1000.c:202:4: error: 'DT_ETH_E1000_IRQ_FLAGS' undeclared (first use in this function); did you mean 'ETH_E1000_PRIV_H'?
    DT_ETH_E1000_IRQ_FLAGS);
    ^~~~~~~~~~~~~~~~~~~~~~
/home/michal/zephyr/zephyrproject/zephyr/include/arch/x86/ia32/arch.h:214:9: note: in definition of macro 'Z_ARCH_IRQ_CONNECT'
        (flags_p)); \
         ^~~~~~~
/home/michal/zephyr/zephyrproject/zephyr/drivers/ethernet/eth_e1000.c:200:2: note: in expansion of macro 'IRQ_CONNECT'
  IRQ_CONNECT(DT_ETH_E1000_IRQ, DT_ETH_E1000_IRQ_PRIORITY,
```

Does it means there is no network driver for ACRN or my config is broken?

When I build for qemu_x86, it builds without error but when I run app on ACRN, I have following output:
```
[00:00:00.000,000] <err> net_if: Iface 0x0012b220 driver API init NULL
[00:00:00.000,000] <inf> net_config: Initializing network
[00:00:00.000,000] <inf> net_config: IPv4 address: 192.168.2.13
```
Please help me solve the issue.

Mingyuan18, 2019-10-17:
[External_System_ID] ACRN-5441

fuzhongl, 2019-10-23:
@marjano89 We haven't enable Ethernet for pre-launched  Zephyr UOS.

gvancuts, 2019-10-24:
> @marjano89 We haven't enable Ethernet for pre-launched Zephyr UOS.

Hi @fuzhongl , is it a simple case of modifying the ACRN configuration to assign the Ethernet device to the Zephyr User VM, or is there more needed there?

I also suppose that the Zephyr Ethernet needs to support the Ethernet controller on the board, and I don't know if that's the case.

marjano89, 2019-10-24:
Is there any possibility to add this support? I am working under SDC2 project with Zephyr as an application part which shall be a data provider for AGL UOS (Instrument Cluster ) and Celadon (IVI) and Zephyr is a key part.

XX-MEN, 2019-11-06:
I will also give a warm welcome for Ethernet support. Can you enable it? I'm preparing a demo for CES 2020[https://www.ces.tech/](url)