projectacrn/acrn-hypervisor #3849: HW Support NUC7i7DNK

paspndevops, 2019-10-16:
Hi i am new for ACRN Hypervisor could you please confirm above HW(NUC) will support to start porting .Also guide me how to start with link.

Thanks in advance

fuzhongl, 2019-10-17:
@paspndevops Following is supported HW and GSG
https://projectacrn.github.io/latest/reference/hardware.html
https://projectacrn.github.io/latest/getting-started/apl-nuc.html

paspndevops, 2019-10-22:
@fuzhongl BIOS FW update from DNKBLi7v.86A.0056.2018.1003.1622 ==> to ==> **2019.0710.1150**
Tried with clr-linux-server and ubuntu but no luck 

could you pls suggest basic procedure to start working for SOS/UOS on NUC

**root@intel-NUC7i7DNKE:~/acrn-hypervisor# efibootmgr -v**
BootCurrent: 0003
Timeout: 1 seconds
BootOrder: 001D,0003,001B

Boot0003* ubuntu	HD(1,GPT,d09e6bb4-9604-426c-aa78-2b19cbc415ca,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)

Boot001B* UEFI : LAN : PXE IP4 Intel(R) Ethernet Connection I219-LM	PciRoot(0x0)/Pci(0x1f,0x6)/MAC(54b203091844,0)/IPv4(0.0.0.00.0.0.0,0,0)..BO
Boot001C* UEFI : LAN : PXE IP6 Intel(R) Ethernet Connection I219-LM	PciRoot(0x0)/Pci(0x1f,0x6)/MAC(54b203091844,0)/IPv6([::]:<->[::]:,0,0)..BO

_Boot001D* ACRN Hypervisor	HD(1,GPT,d09e6bb4-9604-426c-aa78-2b19cbc415ca,0x800,0x100000)/File(\EFI\acrn\acrn.efi)b.o.o.t.l.o.a.d.e.r.=.\.E.F.I.\.u.b.u.n.t.u.\.g.r.u.b.x.6.4...e.f.i._

fuzhongl, 2019-10-22:
@paspndevops Do you install Ubuntu on SATA SSD?
If yes, do you run:
sudo efibootmgr -c -l "\EFI\acrn\acrn.efi" -d /dev/sda -p 1 \
       -L "ACRN Hypervisor" -u "bootloader=\EFI\ubuntu\grubx64.efi"

Thanks!

fuzhongl, 2019-10-23:
@paspndevops  Could you please run following command to check the HW info:
dmidecode --type 1

Thanks!

fuzhongl, 2019-10-23:
@paspndevops What's your progress on NUC7i7DNKE?
Thanks!

paspndevops, 2019-10-23:
**root@intel-NUC7i7DNKE:~# uname -a**
Linux intel-NUC7i7DNKE 5.3.0_rc8-9.iot-lts2019 #1 SMP Fri Sep 13 18:53:33 PDT 2019 x86_64 x86_64 x86_64 GNU/Linux

**root@intel-NUC7i7DNKE:~# vi /boot/grub/grub.cfg** 

menuentry "ACRN ubuntu SOS" {
        recordfail
        load_video
        gfxmode $linux_gfx_mode
        insmod gzio
        if [ x$grub_platform = xxen ]; then insmod xzio; insmod lzopio; fi
        insmod part_gpt
        insmod ext2
        set root='hd0,gpt2'
        if [ x$feature_platform_search_hint = xy ]; then
          search --no-floppy --fs-uuid --set=root --hint-bios=hd0,gpt2 --hint-efi=hd0,gpt2 --hint-baremetal=ahci0,gpt2  a5165558-2bcc-4859-8d5f-946cedbd6a9b
        else
          search --no-floppy --fs-uuid --set=root a5165558-2bcc-4859-8d5f-946cedbd6a9b
        fi
                linux   /boot/acrn/org.clearlinux.iot-lts2019.5.3.0_rc8-9 root=UUID=a5165558-2bcc-4859-8d5f-946cedbd6a9b rw  quiet splash $vt_handoff
        initrd  /boot/initrd.img-4.15.0-66-generic

========================================================
root@intel-NUC7i7DNKE:~# dmesg | grep ACRN
root@intel-NUC7i7DNKE:~#

paspndevops, 2019-11-06:
root@clrdev~/uos # ./launch_uos_31440.sh
cpu1 online=0
cpu2 online=0
cpu3 online=0
cpu4 online=0
cpu5 online=0
cpu6 online=0
cpu7 online=0
passed gvt-g optargs low_gm 64, high_gm 448, fence 8
SW_LOAD: get ovmf path /usr/share/acrn/bios/OVMF.fd, size 0x200000
logger: name=console, level=4
logger: name=kmsg, level=3
logger: name=disk, level=5
vm_create: vm1
VHM api version 1.0
vm_setup_memory: size=0x80000000
open hugetlbfs file /run/hugepage/acrn/huge_lv1/vm1/D279543825D611E8864ECB7A18B34643
open hugetlbfs file /run/hugepage/acrn/huge_lv2/vm1/D279543825D611E8864ECB7A18B34643
level 0 free/need pages:1/1 page size:0x200000
level 1 free/need pages:2/2 page size:0x40000000

try to setup hugepage with:
level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x0
total_size 0x180000000

mmap ptr 0x0x7f0494bf1000 -> baseaddr 0x0x7f04c0000000
mmap 0x80000000@0x7f04c0000000
touch 2 pages with pagesz 0x40000000
mmap 0x200000@0x7f05bfe00000
touch 1 pages with pagesz 0x200000

really setup hugepage with:
level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x0
vm_init_vdevs
pci init hostbridge
pci init pci-gvt
modify_bar_registration: bypass for pci-gvt
modify_bar_registration: bypass for pci-gvt
modify_bar_registration: bypass for pci-gvt
polling 36...
Listening 36...
pci init virtio-blk
pci init virtio-net
pci init virtio-console
pci init virtio-hyper_dmabuf
pci init virtio-rnd
tpm: init_vtpm2:Invalid socket path!
/tmp/dm.XEQobdH 53: [0001] ProcessorId : FF
Error 6313 - Invalid field label detected ^ (found "ProcessorId" expected "Processor ID")
/tmp/dm.XEQobdH 57: [0001] Interrupt : 01
Error 6313 - ^ Invalid field label detected (found "Interrupt" expected "Interrupt Input LINT")
acpi_build failed, error=-1
pci deinit hostbridge
pci deinit pci-gvt
modify_bar_registration: bypass for pci-gvt
modify_bar_registration: bypass for pci-gvt
modify_bar_registration: bypass for pci-gvt
pci deinit virtio-blk
pci deinit virtio-net
vtnet tx thread closing...
pci deinit virtio-console
pci deinit virtio-hyper_dmabuf
pci deinit virtio-rnd
mngr_client_new: Failed to accept from fd 36, err: Invalid argument
Stop listening 36...
Stop polling 36...

fuzhongl, 2019-11-06:
@paspndevops 
There is the known issue for CL 31440 because of iasl version.
Please use CL 31470 which include following commit:
* 9143e563 - (2019-10-25 15:12:00 +0800) dm: update ACPI with latest ASL standard 
Thanks!

fuzhongl, 2020-05-04:
@paspndevops Do you still have any issue on NUC7i7DNK?
If no, could you please close this git issue?
Thanks!

closed 2020-06-28 by fuzhongl