projectacrn/acrn-hypervisor #3859: VM-Manager: the return value of "strtol" is not validated properly

yonghuah, 2019-10-18:
the return value of "strtol" is "long int", but in below cases,  its return value is assigned to a 

local variable with "int" type and compare it with "LONG_MAX" and "LONG_MIN",  it is invalid:

1)  _get_vmname_pid() @ misc/acrn-manager/acrn_vm_ops.c file_

_*pid = strtol(p, NULL, 10);_
_if ((errno == ERANGE && (*pid == LONG_MAX || *pid == LONG_MIN))_
_|| (errno != 0 && *pid == 0))_
_return -1;_

_..._

_2) static int parse_opt(int argc, char *argv[])  @ misc/acrn-manager/acrnd.c_

static int parse_opt(int argc, char *argv[])
{
int opt, ret = 0;

while ((opt = getopt(argc, argv, optString)) != -1) {
switch (opt) {
case 't':
logfile = 0;
break;
#ifdef MNGR_DEBUG
case 'd':
ret = strtol(optarg, NULL, 10);
if ((errno == ERANGE && (ret == LONG_MAX |

Mingyuan18, 2019-10-18:
[External_System_ID] ACRN-5300

Mingyuan18, 2019-10-23:
fix is integrated and verified, so close this issue.

closed 2019-10-23 by Mingyuan18