projectacrn/acrn-hypervisor #3869: NUC Stoped at EFI

paspndevops, 2019-10-18:
Hi,

I am trying to install ACRN on intel nuc ( NUC7i7DNK(BOARD-NUC7i7DNB) BIOS FW-DNKBLi7v.86A.0056.2018.1003.1622 ) without success.

1 - clearlinux is installed and working well, then installed acrn (27960)

2 - Nuc is stopping during EfI, screen display Intel Boot + number of precessor at top

"detected 4 processes, 4 enabled
current on process 0"

paspndevops, 2019-10-18:
VMX 	Enabled
VT-D 	Enabled
Hyper-Threading 	Disabled
Secure Boot 	Disabled+

gvancuts, 2019-10-21:
Hi @paspndevops , can you confirm these couple of points:
* How much RAM is installed on your system [1]
* Do you get any additional message if you enter an EFI shell and run `acrn.efi` directly from there?

[1] See also this guide if you need to reconfigure the amount of RAM memory to match your system: https://projectacrn.github.io/latest/faq.html#how-do-i-configure-acrn-s-memory-use

paspndevops, 2019-10-21:
**How much RAM is installed on your system [1]**
config HV_RAM_SIZE
		default 0x0b800000
	
config PLATFORM_RAM_SIZE
	hex "Size of the physical platform RAM"
	default 0x400000000
	
config SOS_RAM_SIZE
	hex "Size of the Service OS (SOS) RAM"
	default 0x400000000

 config UOS_RAM_SIZE
	hex "Size of the User OS (UOS) RAM"
	default 0x200000000
	
    **Do you get any additional message if you enter an EFI shell and run acrn.efi directly from there?**
--> make all BOARD=nuc7i7dnb SCENARIO=industry
--> Grub menu with the new “ACRN ubuntu SOS” entry  

"detected 8 processes, 8 enabled
current on process 0"

ttzeng, 2019-10-22:
@paspndevops you may also disable UART for a try by adding "uart=disabled" to the EFI boot entry. The UART controller on NUC7i7DNB may be different than the default one. Check the step # 3 to supply the right UART on your platform. FYI.

paspndevops, 2019-10-22:
@ttzeng 

1. Could you pls suggest me simple method to demonstrate ACRN on my NUC .
2. Recently i updated NUC BIOS FW 

- Now NUC installed with VMware & virtualBox installed on  "Ubuntu 18.04.3 LTS" 
- CPU Model ==> model name	: Intel(R) Core(TM) i7-8650U CPU @ 1.90GHz

root@intel-NUC7i7DNKE:~/acrn-hypervisor# lshw
intel-nuc7i7dnke            
    description: Computer
    product: NUC7i7DNKE
    vendor: Intel Corporation
    version: J85069-204
    serial: DW1703483900155
    width: 64 bits
    capabilities: smbios-3.1 dmi-3.1 smp vsyscall32
    configuration: boot=normal family=Intel NUC uuid=471F5354-B7D6-CC46-ABA2-54B203091844
  *-core
       description: Motherboard
       product: NUC7i7DNB
       vendor: Intel Corporation
       physical id: 0
       version: J83500-204
       serial: BTDN82500261
       slot: Default string
     *-firmware
          description: BIOS
          vendor: Intel Corp.
          physical id: 0
          **version: DNKBLi7v.86A.0066.2019.0710.1150
          date: 07/10/2019**
          size: 64KiB
          capacity: 15MiB
          capabilities: pci upgrade shadowing cdboot bootselect socketedrom edd int13floppy1200 int13floppy720 int13floppy2880 int5printscreen int14serial int17printer acpi usb biosbootspecification uefi
     *-memory
          description: System Memory
          physical id: 35
          slot: System board or motherboard
          size: 12GiB

paspndevops, 2019-10-22:
@ttzeng Thanks for your quick reply  ..

**Could you confirm Which HARDWARE you recommend  for successful operation. Here i attached 'lspci' output for your reference.** 

root@intel-NUC7i7DNKE:~/acrn-hypervisor# lspci
00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers (rev 08)
00:02.0 VGA compatible controller: Intel Corporation UHD Graphics 620 (rev 07)
00:08.0 System peripheral: Intel Corporation Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th Gen Core Processor Gaussian Mixture Model
00:14.0 USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21)
00:14.2 Signal processing controller: Intel Corporation Sunrise Point-LP Thermal subsystem (rev 21)
00:15.0 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #0 (rev 21)
00:15.1 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #1 (rev 21)
00:16.0 Communication controller: Intel Corporation Sunrise Point-LP CSME HECI #1 (rev 21)
00:16.3 Serial controller: Intel Corporation Device 9d3d (rev 21)
00:17.0 SATA controller: Intel Corporation Sunrise Point-LP SATA Controller [AHCI mode] (rev 21)
00:1c.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #3 (rev f1)
00:1f.0 ISA bridge: Intel Corporation Intel(R) 100 Series Chipset Family LPC Controller/eSPI Controller - 9D4E (rev 21)
00:1f.2 Memory controller: Intel Corporation Sunrise Point-LP PMC (rev 21)
00:1f.3 Audio device: Intel Corporation Sunrise Point-LP HD Audio (rev 21)
00:1f.4 SMBus: Intel Corporation Sunrise Point-LP SMBus (rev 21)
00:1f.6 Ethernet controller: Intel Corporation Ethernet Connection I219-LM (rev 21)
01:00.0 Network controller: Intel Corporation Wireless 8265 / 8275 (rev 78)

paspndevops, 2019-10-22:
@ttzeng /etc/grub.d/40_custom  

menuentry 'ACRN ubuntu SOS' {
        recordfail
        load_video
        insmod gzio
        insmod part_gpt
        insmod ext2
        linux  /boot/acrn/org.clearlinux.iot-lts2018-sos.4.19.78-95 pci_devices_ignore=(0:18:1) console=tty0 console=ttyS0 root=PARTUUID=a5165558-2bcc-4859-8d5f-946cedbd6a9b rw rootwait ignore_loglevel no_timer_check consoleblank=0 i915.nuclear_pageflip=1 i915.avail_planes_per_pipe=0x01010F i915.domain_plane_owners=0x011111110000 i915.enable_gvt=1 i915.enable_guc=0 hvlog=2M@0x1FE00000
}

ttzeng, 2019-10-22:
@paspndevops I believe [NUC7i7DNHE](https://ark.intel.com/content/www/us/en/ark/products/130393/intel-nuc-kit-nuc7i7dnhe.html) or [NUC7i5DNHE](https://ark.intel.com/content/www/us/en/ark/products/122488/intel-nuc-kit-nuc7i5dnhe.html) are full validated.

However, while looking the spec of NUC7i7DNHE, I found Intel Core i7 8650U are in the list, so your current NUC system might be able to run.

Per your `lspci` output, it looks like your serial port may be at 0:16.3, which may not be the same as the [default configuration for NUC7i7DNHE](https://github.com/projectacrn/acrn-hypervisor/blob/master/hypervisor/arch/x86/configs/nuc7i7dnb.config). So, again, please try to disable the serial console from the EFI boot entry:
```
$ sudo efibootmgr -c -l "\EFI\acrn\acrn.efi" -d /dev/sda -p 1 -L "ACRN" -u "uart=disabled"
```
That should avoid the hypervisor hang while initializing the serial port. You may need to remove the previous ACRN boot entry by `efibootmgr -b _n_ -B`, where _n_ is the number associated with the old ACRN boot entry. You can get the number by `efibootmgr -v`

paspndevops, 2019-10-23:
@ttzeng  Now I am using **NUC7iDNKE** and i tried disabling serial console EFI boot entry after that i am not able to boot again into EFI, Please find below Log FYI:

**Boot001B*** ACRN NUC Hypervisor   HD(1,GPT,d09e6bb4-9604-426c-aa78-2b19cbc415ca,0x800,0x100000)/File(\EFI\acrn\acrn.efi)b.o.o.t.l.o.a.d.e.r.=.\.E.F.I.\.o.r.g...c.l.e.a.r.l.i.n.u.x.\.b.o.o.t.l.o.a.d.e.r.x.6.4...e.f.i. .u.a.r.t.=.d.i.s.a.b.l.e.d.

**_Boot001C*** ACRN  HD_(1,GPT,d09e6bb4-9604-426c-aa78-2b19cbc415ca,0x800,0x100000)/File(\EFI\acrn\acrn.efi)u.a.r.t.=.d.i.s.a.b.l.e.d.

ttzeng, 2019-10-23:
@paspndevops Which version of `acrn.efi` are you using? from the beginning of this thread, I suppose you are using `acrn.efi` from the Clear Linux 27960, and your system has 12GB memroy installed. Are these correct?

The last option I have is to use the later `acrn.efi` executable, for example v1.3 with Clear Linux 31080, could you do one more try? Thanks.

closed 2019-10-23 by paspndevops