projectacrn/acrn-hypervisor #3912: EFI firmware :Not Found

paspndevops, 2019-10-23:
**NAME**="Ubuntu"
**VERSION**="18.04.3 LTS (Bionic Beaver)"
Manufacturer: Intel Corporation
Product Name: NUC7i7DNKE
Version: J85069-204
Serial Number: DW1703483900155
Family: Intel NUC

After Configure the EFI firmware to boot the ACRN hypervisor by default with v1.3:

efibootmgr -c -l "\EFI\acrn\acrn.efi" -d /dev/sda -p 1 -L "ACRN" -u "uart=disabled"

Detected 8 process, 1 enabled
: Not Found
Hit any key to exit

**a bootable device has not been detected** 

==================================================

**cat /etc/grub.d/40_custom**
menuentry 'ACRN ubuntu SOS' {
        recordfail
        load_video
        insmod gzio
        insmod part_gpt
        insmod ext2
        linux  /boot/acrn/org.clearlinux.iot-lts2019.5.3.0_rc8-9 pci_devices_ignore=(0:18:1) console=tty0 console=ttyS0 root=PARTUUID=<UUID of rootfs partition> rw rootwait ignore_loglevel no_timer_check consoleblank=0 i915.nuclear_pageflip=1 i915.avail_planes_per_pipe=0x01010F i915.domain_plane_owners=0x011111110000 i915.enable_gvt=1 i915.enable_guc=0 hvlog=2M@0x1FE00000

ttzeng, 2019-10-24:
@paspndevops it looks like you are booting ACRN with GRUB, you will need the ELF32 executable `acrn.32.out` instead, as the `acrn.efi` will load the wrapper for systemd-boot bootloader `bootloaderx64.efi` by default. It looks like you installed Ubuntu instead Clear Linux on your NUC, so systemd-boot is not installed.

Mingyuan18, 2019-10-24:
[External_System_ID] ACRN-5512

fuzhongl, 2019-10-24:
@paspndevops 
Do you copy acrn.efi to /boot/EFI/acrn/acrn.efi? 
Do you install  in the SATA SSD or NVMe SSD?
# For SATA
sudo efibootmgr -c -l "\EFI\acrn\acrn.efi" -d /dev/sda -p 1 \
       -L "ACRN Hypervisor" -u "bootloader=\EFI\ubuntu\grubx64.efi"
# For NVMe
sudo efibootmgr -c -l "\EFI\acrn\acrn.efi" -d /dev/nvme0n1 -p 1 \
       -L "ACRN Hypervisor" -u "bootloader=\EFI\ubuntu\grubx64.efi"

Please use following CMD to check the memory of your NUC:
dmidecode --type 17

Thanks!

paspndevops, 2019-10-24:
@fuzhongl 
Yes copied acrn.efi to /boot/efi/EFI/acrn/acrn.efi and executed 

sudo efibootmgr -c -l "\EFI\acrn\acrn.efi" -d /dev/sda -p 1
-L "ACRN Hypervisor" -u "bootloader=\EFI\ubuntu\grubx64.efi"

same not able to boot into EFI

fuzhongl, 2019-10-24:
@paspndevops  Please use following cmd to check the memory size of your NUC:
dmidecode --type 17

Thanks!

paspndevops, 2019-10-24:
Handle 0x0036, DMI type 17, 40 bytes
Array Handle:0x0035
----
----
Handle 0x0037, DMI type 17, 40 bytes 
Handle 0x0038, DMI type 17, 40 bytes
Handle 0x0039, DMI type 17, 40 bytes

fuzhongl, 2019-10-24:
@paspndevops Thanks for update.
ACRN related kernel patches aren't upstreamed to  org.clearlinux.iot-lts2019.5.3.0 yet.
Could you please try with our acrn-kernel code?
https://github.com/projectacrn/acrn-kernel
Thanks!

paspndevops, 2019-10-24:
@fuzhongl 
Thanks for quick reply  Shell i try clr-Linux on virtual-box on top of UBUNTU 18.04 
https://projectacrn.github.io/latest/getting-started/apl-nuc.html

Also please suggest me which build need to try

Thanks

fuzhongl, 2019-10-24:
@paspndevops Please try CL 30180 build; it is the verified Build for V1.3 release.
Thanks!

paspndevops, 2019-10-25:
@fuzhongl 
Hi i Tried with 30180 and executed acrn_quick_setup.sh successfully after reboot(selected iot-sos kernel) it was struck  at 

> acrn-br0: port 2(enp0s3) entered blocking state
>  acrn-br0: port 2(enp0s3) entered forwarding state
> hrtimer: interrupt took 7149418 ns

fuzhongl, 2019-10-25:
@paspndevops  Could you please share full log?
Thanks!

fuzhongl, 2019-10-26:
@paspndevops 
acrn_quick_setup.sh will copy acrn.efi and acrn-dm to the right directory.
You can following

closed 2019-10-31 by paspndevops