projectacrn/acrn-hypervisor #3939: zero-copy non-functional with vhost

lyakh, 2019-10-25:
The vhost-net driver can use zero-copy for sending data. It has to be enabled using the `experimental_zcopytx` module parameter. However, enabling it and adding debugging to actually check whether zero-copy data transmission was successful, reveals, that it's always failing and the driver reverts to the default copying procedure:
```
[ 3778.439847] vhost_net_tx_err(): 1 : 0
[ 3778.446522] handle_tx_zerocopy(): ztx 12 @ 00000000eb8152b6
[ 3778.447097] vhost_net_tx_err(): 2 : 64
[ 3778.450652] handle_tx_zerocopy(): ztx 12 @ 00000000cdda46b1
[ 3778.451286] vhost_net_tx_err(): 3 : 128
```
i.e. every attempt to use zero-copy increments the zero-copy error count (the first number in `vhost_net_tx_err()` lines), 64 packets later (the second number in the same lines) it's attempted again and it fails again, etc.

Mingyuan18, 2019-10-28:
[External_System_ID] ACRN-5529

ssqre, 2019-11-08:
TAP device does not support this.  when call tun_get_user from tun_chr_write_iter, zerocopy is false. This is not a hypervisor bug.

lyakh, 2019-11-08:
@ssqre hi, thanks for checking. However I'm not sure I fully understand your analysis. Have you set the experimental_zcopytx module parameter? I don't understand why the TAP device should be related to the networking data path, to how data is copied between the SOS and the UOS?

ssqre, 2019-11-08:
If you want to use vhost zerocopy, you should make sure the lower device also support this. when vhost send data to tap device, ... -->tun_chr_write_iter --> tun_get_user will be called. And zerocopy is always false regardless of experimental_zcopytx is 1 or 0. This will cause vq->heads[i].len == VHOST_DMA_FAILED_LEN to be true in vhost.

lyakh, 2019-11-08:
@ssqre interesting, thanks, I'll have to check that. Do you happen to know how they then implemented and tested the VirtIO zero-copy feature with vhost-net? Isn't it usually used together with a tun / tap device?

ssqre, 2019-11-11:
@lyakh  tun/tap and vhost-net are standard linux kernel driver. Maybe you can find a solution in Linux community.

Mingyuan18, 2019-11-19:
@lyakh ，Kindly ask, can you close this issue?

fuzhongl, 2020-05-05:
@lyakh Can this git issue be closed now?
Thanks!

lyakh, 2020-05-05:
> @lyakh Can this git issue be closed now?
> Thanks!

@fuzhongl I don't know if I find time to further investigate this issue any time soon, so, sure, if nobody else sees this issue let's close it for now.

closed 2020-05-05 by lyakh