projectacrn/acrn-hypervisor #3946: Cannot boot VxWorks as UOS on KabyLake

LiangZhou9527, 2019-10-29:
Hello guys,

I'm trying to run VxWorks as UOS but failed. Here's the console log (attached vxworks_error.log) when I run the script launch_vxworks.sh .
Could you guy guide me the direction to go further?
Thank you in advance.

My Env: 
KabyLake with i7-7700@3.60Ghz
VMX Enabled
VT-D Enabled
Hyper-Threading Disabled
The clear Linux as SOS can be running:
$ uname -r
4.19.78-95.iot-lts2018-sos

The VxWorks I'm using is VxWorks7, and I can boot the VxWorks image as an native OS on the same board.

[vxworks_error.log](https://github.com/projectacrn/acrn-hypervisor/files/3781492/vxworks_error.log)

Mingyuan18, 2019-10-29:
[External_System_ID] ACRN-5539

gvancuts, 2019-10-29:
A bit of a shot in the dark here but could it be related to this PR #3935 ?

LiangZhou9527, 2019-10-30:
> A bit of a shot in the dark here but could it be related to this PR #3935 ?

Thanks for your comments.

I'm not familiar with the ACRN internal. 
Do you have any suggestion to make the UOS VxWorks go further?

fuzhongl, 2019-10-30:
@LiangZhou9527 Which build are you using?
We can't reproduce this issue on KBLNUC7i7DNH; please check following info:
root@10239227155sos-dom0~ # /usr/bin/acrn-dm -v
DM version is: 1.4-unstable-d5c3523d (daily tag:acrn-2019w43.3-140000p), build by root@2019-10-25 07:52:52
root@10239227155sos-dom0~ # swupd info --allow-insecure-http 
Warning: This is an insecure connection
The --allow-insecure-http flag was used, be aware that this poses a threat to the system

Distribution:      Clear Linux OS
Installed version: 31310
Version URL:       http://linux-ftp.sh.intel.com/pub/mirrors/clearlinux/update
Content URL:       http://linux-ftp.sh.intel.com/pub/mirrors/clearlinux/update

LiangZhou9527, 2019-10-30:
> Distribution: Clear Linux OS
> Installed version: 31310

@fuzhongl Thank you for your information. Currently this board is used by other purpose, I'll check the info you asked when got chance and update you soon.

LiangZhou9527, 2019-11-08:
Hi @fuzhongl  ,
I tried a new installation version 31310 which is the same as yours , but still got failure which is some bit different than my previous trying.

Any suggestion please?

liang@clr-fd28f8f2534a42c19acf20d10f59fd28~/vxworks $ sudo ./launch_vxworks.sh
cpu1 online=1
cpu2 online=1
cpu3 online=1
SW_LOAD: get ovmf path /usr/share/acrn/bios/OVMF.fd, size 0x200000
vm_create: vxworks_vm1
VHM api version 1.0
vm_setup_memory: size=0x80000000
open hugetlbfs file /run/hugepage/acrn/huge_lv1/vxworks_vm1/D279543825D611E8864ECB7A18B34643
open hugetlbfs file /run/hugepage/acrn/huge_lv2/vxworks_vm1/D279543825D611E8864ECB7A18B34643
level 0 free/need pages:0/1 page size:0x200000
level 1 free/need pages:0/2 page size:0x40000000
to reserve more free pages:
to reserve pages (+orig 0): echo 2 > /sys/kernel/mm/hugepages/hugepages-1048576kB/nr_hugepages
to reserve pages (+orig 0): echo 1 > /sys/kernel/mm/hugepages/hugepages-2048kB/nr_hugepages
now enough free pages are reserved!

try to setup hugepage with:
	level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
	level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x0
total_size 0x180000000

mmap ptr 0x0x7f31213cc000 -> baseaddr 0x0x7f3140000000
mmap 0x80000000@0x7f3140000000
touch 2 pages with pagesz 0x40000000
mmap 0x200000@0x7f323fe00000
touch 1 pages with pagesz 0x200000

really setup hugepage with:
	level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
	level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x0
vm_init_vdevs
No correct pm notify channel given
pci init hostbridge
pci init virtio-blk
pci init virtio-console
tpm: init_vtpm2:Invalid socket path!
                                    polling 34...
                                                 Listening 34...
                                                                /tmp/dm.XY6eHrC    110:     Processor (CPU0, 0x00, 0x00000000, 0x00) {}
                                                       Warning  3168 -                          ^ Legacy Processor() keyword detected. Use Device() keyword instead.

    /tmp/dm.XY6eHrC    111:     Processor (CPU1, 0x01, 0x00000000, 0x00) {}
                                                                           Warning  3168 -                          ^ Legacy Processor() keyword detected. Use Device() keyword instead.

                        /tmp/dm.XY6eHrC    112:     Processor (CPU2, 0x02, 0x00000000, 0x00) {}
               Warning  3168 -                          ^ Legacy Processor() keyword detected. Use Device() keyword instead.

                                            acrn_sw_load
                                                        SW_LOAD: partition blob /usr/share/acrn/bios/OVMF.fd size 2097152 copy to guest 0xffe00000
                                                                  SW_LOAD: build e820 7 entries to addr: 0x7f31400ef008
SW_LOAD: entry[0]: addr 0x0000000000000000, size 0x00000000000a0000,  type 0x1
SW_LOAD: entry[1]: addr 0x00000000000a0000, size 0x0000000000060000,  type 0x2
SW_LOAD: entry[2]: addr 0x0000000000100000, size 0x000000007ff00000,  type 0x1
SW_LOAD: entry[3]: addr 0x0000000080000000, size 0x0000000008000000,  type 0x2
SW_LOAD: entry[4]: addr 0x00000000df000000, size 0x0000000001000000,  type 0x2
SW_LOAD: entry[5]: addr 0x00000000e0000000, size 0x0000000020000000,  type 0x2
SW_LOAD: entry[6]: addr 0x0000000100000000, size 0x0000000040000000,  type 0x2
SW_LOAD: ovmf_entry 0xfffffff0
                              add_cpu
                                     out instr on NMI port (0x97) not supported
                                                                               out instr on NMI port (0x97) not supported
                                         out instr on NMI port (0x97) not supported

fuzhongl, 2019-11-08:
@LiangZhou9527 
We can't reproduce this issue on our side. Maybe this issue is related with Vxwork image.
Thanks!

LiangZhou9527, 2019-11-08:
the launch script for VxWorks I used is located under SOS: /usr/share/acrn/samples/nuc/launch_vxworks.sh , I also attached it.
[launch_vxworks.zip](https://github.com/projectacrn/acrn-hypervisor/files/3822772/launch_vxworks.zip)

fuzhongl, 2019-11-08:
@LiangZhou9527 Checked with developer; Vxworks needs additional patches for ACRN; and the version should be SR0540. It isn't available for community until now.
Sorry about this.

LiangZhou9527, 2019-11-08:
@fuzhongl  Thanks a lot for your confirmation.

closed 2020-05-05 by fuzhongl