projectacrn/acrn-hypervisor #3958: linux-iot-lts2018-sos-4.19.78-95.x86_64

paspndevops, 2019-10-29:
Login Prompt appeared successfully but keyboard not responding for input entry 

[image]

paspndevops, 2019-10-29:
@fuzhongl 

[image]

fuzhongl, 2019-10-29:
@paspndevops USB ports are assigned to UOS in your script? Could you please share the launch script? 
Thanks!

fuzhongl, 2019-10-29:
@paspndevops  Which Clear Linux build are you using?
Ant steps to reproduce this issue?
Thanks!

paspndevops, 2019-10-30:
@fuzhongl 

clr-os **31380** i tried and Kernel " **linux-iot-lts2018-sos-4.19.78-95.x86_64**"

fuzhongl, 2019-10-30:
@paspndevops  clr-os 31380 isn't the workable CL build, please try the latest build or 31310(verified on our side).
Thanks!

Mingyuan18, 2019-10-30:
[External_System_ID] ACRN-5558

paspndevops, 2019-10-30:
@fuzhongl @Mingyuan18 

Could you please confirm on which NUC(Model) 31310 verified now I am using ( NUC7i7DNK(BOARD-NUC7i7DNB)

fuzhongl, 2019-10-30:
@paspndevops We verified this build  on KBLNUC7i7DNHE. 
Thanks.

paspndevops, 2019-12-12:
@fuzhongl 

Is there any specific reason to use above HW why i am not able work with NUC7i7DNK and intel laptop.

Thanks !

fuzhongl, 2019-12-12:
@paspndevops 
We don't have KBLNUC7i7DNK; KBLNUC7i7DNHE is our QA verified platform.
Thanks!

paspndevops, 2019-12-12:
@fuzhongl 
Thanks for your quick reply could you please tell me is there any hardware dependency in acrn.efi or sos kernel.

fuzhongl, 2019-12-12:
@paspndevops 
Following is verified platform:
https://projectacrn.github.io/latest/reference/hardware.html
Thanks!

fuzhongl, 2020-05-05:
@paspndevops Do you copy the lib/modules to your board for this issue?

closed 2020-06-28 by fuzhongl