projectacrn/acrn-hypervisor #3960: UP2 Setup - "ACRN HVLog: not running under acrn hypervisor!"

KalevGonvick, 2019-10-29:
Hello,
I am having a few issues trying to set everything up on my UP2 board.

I have followed the setup for the UP2 board found [here](https://projectacrn.github.io/latest/getting-started/up2.html#getting-started-guide-for-up2-board).

After following the guide, using the command `$ sudo dmesg | grep ACRN` returns:

- [5.617767] ACRNTrace: acrn_trace: not support acrn hypervisor!

-  [5.617769] ACRN HVLog: not running under acrn hypervisor!

The only thing that I could not find out how to modify was the `acrn.conf` file when setting the root. The UP2 tutorial mentions an acrn.conf file, but the NUC tutorial does not mention it at all.

Any help for this novice would be greatly appreciated,

Kalev

Mingyuan18, 2019-10-30:
[External_System_ID] ACRN-5557

fuzhongl, 2019-11-01:
@KalevGonvick Could you please share HW info with "lshw" command? Which Clear Linux build used for this issue?
Thanks!

fuzhongl, 2019-11-05:
@KalevGonvick Please refer following link to build acrn.efi for apl-up2: with BOARD=apl-up2
https://projectacrn.github.io/latest/getting-started/building-from-source.html#
Thanks!

KalevGonvick, 2019-11-18:
@fuzhongl helped solve this error over email. The solution was to build from source, and to change the boot order in the BIOS after configuring 'efibootmgr'.

closed 2019-11-18 by KalevGonvick