projectacrn/acrn-hypervisor #3980: invalid ovmf param ./OVMF.fd

paspndevops, 2019-10-31:
root@clrdev~ # sh acrn_quick_setup.sh -u 31440

Configure kernel-ios-lts2018 as /root/clear-31440-kvm.img default boot kernel
Upgrade User VM done...
Now you can run this command to start User VM...
sudo /root/launch_uos_31440.sh
root@clrdev~ # /root/launch_uos_31440.sh
cpu1 online=1
cpu2 online=1
cpu3 online=1
cpu4 online=1
cpu5 online=1
cpu6 online=1
cpu7 online=1
passed gvt-g optargs low_gm 64, high_gm 448, fence 8
SW_LOAD ERR: image file failed to open
acrn-dm: invalid ovmf param ./OVMF.fd
root@clrdev~ # 

=========================================================

root@clrdev~ # dmesg | grep ACRN
[    0.000000] Hypervisor detected: ACRN
[    1.034252] ACRNTrace: Initialized acrn trace module with 8 cpu
[    1.036685] ACRN HVLog: Failed to init last hvlog devs, errno -19
[    1.036849] ACRN HVLog: Initialized hvlog module with 8 cpu
root@clrdev~ # cat /dev/acrn_
acrn_hvlog_cur_0  acrn_hvlog_cur_5  acrn_trace_2      acrn_trace_7
acrn_hvlog_cur_1  acrn_hvlog_cur_6  acrn_trace_3      acrn_vhm
acrn_hvlog_cur_2  acrn_hvlog_cur_7  acrn_trace_4      
acrn_hvlog_cur_3  acrn_trace_0      acrn_trace_5      
acrn_hvlog_cur_4  acrn_trace_1      acrn_trace_6

ttzeng, 2019-10-31:
@paspndevops I suspect your script is out-of-dated, the latest script should use the OVMF.fd firmware in the rootfs according #3677. Instead of using the firmware in the local folder, please modify the path in the script to use the firmware included in the releases. FYI.

paspndevops, 2019-10-31:
@ttzeng 
Thanks for quick reply but my SOS is working after did the following modification :

"make BOARD=nuc7i7dnb FIRMWARE=uefi"

Could you help me what are modifications need to do in **acrn_quick_setup.sh**

paspndevops, 2019-10-31:
@ttzeng @gvancuts 

**root@clrdev~/uos # ./launch_uos_31440.sh** 
cpu1 online=0
cpu2 online=0
cpu3 online=0
cpu4 online=0
cpu5 online=0
cpu6 online=0
cpu7 online=0
passed gvt-g optargs low_gm 64, high_gm 448, fence 8
**SW_LOAD: get ovmf path /usr/share/acrn/bios/OVMF.fd, size 0x200000**
logger: name=console, level=4
logger: name=kmsg, level=3
logger: name=disk, level=5
vm_create: vm1
VHM api version 1.0
vm_setup_memory: size=0x80000000
open hugetlbfs file /run/hugepage/acrn/huge_lv1/vm1/D279543825D611E8864ECB7A18B34643
open hugetlbfs file /run/hugepage/acrn/huge_lv2/vm1/D279543825D611E8864ECB7A18B34643
level 0 free/need pages:1/1 page size:0x200000
level 1 free/need pages:2/2 page size:0x40000000

try to setup hugepage with:
	level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
	level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x0
total_size 0x180000000

mmap ptr 0x0x7f0494bf1000 -> baseaddr 0x0x7f04c0000000
mmap 0x80000000@0x7f04c0000000
touch 2 pages with pagesz 0x40000000
mmap 0x200000@0x7f05bfe00000
touch 1 pages with pagesz 0x200000

really setup hugepage with:
	level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
	level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x0
vm_init_vdevs
pci init hostbridge
pci init pci-gvt
modify_bar_registration: bypass for pci-gvt
modify_bar_registration: bypass for pci-gvt
modify_bar_registration: bypass for pci-gvt
polling 36...
Listening 36...
pci init virtio-blk
pci init virtio-net
pci init virtio-console
pci init virtio-hyper_dmabuf
pci init virtio-rnd
tpm: init_vtpm2:Invalid socket path!
 /tmp/dm.XEQobdH     53: [0001]		ProcessorId : FF
                                                                                                Error    6313 - Invalid field label detected ^  (found "ProcessorId" expected "Processor ID")
/tmp/dm.XEQobdH     57: [0001]		Interrupt : 01
         Error    6313 -                            ^ Invalid field label detected (found "Interrupt" expected "Interrupt Input LINT")
acpi_build failed, error=-1
pci deinit hostbridge
pci deinit pci-gvt
modify_bar_registration: bypass for pci-gvt
modify_bar_registration: bypass for pci-gvt
modify_bar_registration: bypass for pci-gvt
pci deinit virtio-blk
pci deinit virtio-net
vtnet tx thread closing...
pci deinit virtio-console
pci deinit virtio-hyper_dmabuf
pci deinit virtio-rnd
mngr_client_new: Failed to accept from fd 36, err: Invalid argument
Stop listening 36...
Stop polling 36...
root@clrdev~/uos #

Mingyuan18, 2019-10-31:
[External_System_ID] ACRN-5562

fuzhongl, 2019-10-31:
@paspndevops  Any more issue left?
Thanks!

paspndevops, 2019-10-31:
@fuzhongl 

Default acrn_quick_setup.sh is struck at boot-loader if i recompiled acrn-hypervisor with "make BOARD=nuc7i7dnb FIRMWARE=uefi"  then it was booting .

**/usr/lib/acrn/acrn.nuc6cayh.sdc.efi** 

Could you help me what are modifications need to do in acrn_quick_setup.sh

lirui34, 2019-10-31:
Here is the workaround need to do in ```acrn_quick_setup.sh`:
`117        efibootmgr -c -l "\EFI\acrn\acrn.efi" -d $partition -p 1 -L "ACRN" >/dev/null`
to
`117        efibootmgr -c -l "\EFI\acrn\acrn.efi" -d $partition -p 1 -L "ACRN" -u "uart=disabled" >/dev/null`

Now we are working on refining the script, you may use this WA until the next release 😄

ttzeng, 2019-11-01:
@paspndevops I met the same issue as yours on a fresh installation with the following errors:
```
/tmp/dm.XEQobdH 53: [0001] ProcessorId : FF
Error 6313 - Invalid field label detected ^ (found "ProcessorId" expected "Processor ID")
/tmp/dm.XEQobdH 57: [0001] Interrupt : 01
Error 6313 - ^ Invalid field label detected (found "Interrupt" expected "Interrupt Input LINT")
acpi_build failed, error=-1
```
It looks like we hit the [ACPI version mis-match issue](https://github.com/projectacrn/acrn-hypervisor/issues/3933) after Clear Linux upgrade the `iasl` ACPI source compiler. After rebuilding the DM, the issue is solved. You will need to use the latest DM includes [this commit](https://github.com/projectacrn/acrn-hypervisor/commit/9143e56336366110ba8d88ccf8d5353469b87739), or update the Clear Linux to 31470 build (though it's coming soon).

paspndevops, 2019-11-01:
@ttzeng 

I tried as you suggested with 31470 even though i am getting same error could you please help me to run any UOS on my NUC.

/tmp/dm.XEQobdH 53: [0001] ProcessorId : FF
Error 6313 - Invalid field label detected ^ (found "ProcessorId" expected "Processor ID")
/tmp/dm.XEQobdH 57: [0001] Interrupt : 01
Error 6313 - ^ Invalid field label detected (found "Interrupt" expected "Interrupt Input LINT")
acpi_build failed, error=-1
pci deinit hostbridge
pci deinit pci-gvt
modify_bar_registration: bypass for pci-gvt
modify_bar_registration: bypass for pci-gvt
modify_bar_registration: bypass for pci-gvt
pci deinit virtio-blk
pci deinit virtio-net
vtnet tx thread closing...
pci deinit virtio-console
pci deinit virtio-hyper_dmabuf
pci deinit virtio-rnd
mngr_client_new: Failed to accept from fd 36, err: Invalid argument
Stop listening 36...
Stop polling 36...

fuzhongl, 2019-11-01:
@paspndevops Please check iasl version with "iasl -v" in sos; it should be 20191018 if you want to use latest acrn-dm.
If the iasl version isn't 20191018; please use v1.3 release build: CL 31080.
Thanks!

ttzeng, 2019-11-02:
@paspndevops Did you re-install CL 31470 or `swupd update` from your previous installation? If the `/etc/os-releases` indicates the release is 31470, what's the md5sum of `/usr/bin/acrn-dm'?

I didn't update my device to CL 31470 as it wasn't available at that time, so I download the [ACRN binary package](https://cdn.download.clearlinux.org/releases/31470/clear/x86_64/os/Packages/acrn-hypervisor-bin-2019w44.3.140000p-245.x86_64.rpm) , extract the files using the `rpm2cpio xxx.rpm | cpio -idmv` command, and move the extracted `acrn-dm` to `/usr/bin/acrn-bin`. The issue is done, and the CL kvm image can be booted up successfully.

fuzhongl, 2019-11-02:
@paspndevops  The tag of acrn-hypervisor is 2019w44.3.140000 in CL 31470; It should be ok for UOS launch.
Please using following cmd to upgrade to CL 31470:
“swupd repair --picky -V 31470”
Thanks!

paspndevops, 2019-11-04:
@fuzhongl 

thank you very much for your continues support :

After update to CL 31470 successfully i am running /root/launch_uos_31470.sh getting following error:

acrn_dm:  Invalid guest vCPUs
vhm-ioreq: destroy ioreq client 2 for ioeventfd-1

paspndevops, 2019-11-04:
root@clr-bcb66b15ba444557a68bdadab9e6270f~ # md5sum /usr/bin/acrn-dm 
**49fe0446842fd00fa73db0271532e7b0**  /usr/bin/acrn-dm

root@clr-bcb66b15ba444557a68bdadab9e6270f~ # acrn-dm -v
DM version is: 1.4-unstable-”**2019w44.3.140000p_245**” (daily tag:”2019w44.3.140000p”), build by mockbuild@2019-10-31 14:18:45

paspndevops, 2019-11-04:
@fuzhongl 
sh launch_uos_31470.sh 
cpu1 online=1
cpu2 online=1
cpu3 online=1
cpu4 online=1
cpu5 online=1
cpu6 online=1
cpu7 online=1
passed gvt-g optargs low_gm 64, high_gm 448, fence 8
SW_LOAD: get ovmf path /usr/share/acrn/bios/OVMF.fd, size 0x200000
pm by vuart node-index = 0
logger: name=console, level=4
logger: name=kmsg, level=3
logger: name=disk, level=5
vm_create: vm1
VHM api version 1.0
Invalid guest vCPUs (0)

fuzhongl, 2019-11-04:
@paspndevops Which launch script do you use?
There is  -c parameter in the script, right?
please use the latest launch script in github:
https://github.com/projectacrn/acrn-hypervisor/blob/master/devicemodel/samples/nuc/launch_uos.sh
Thanks!

paspndevops, 2019-11-04:
@fuzhongl 

-c and new launch file giving same error
cpu7 online=1
passed gvt-g optargs low_gm 64, high_gm 448, fence 8
SW_LOAD: get ovmf path /usr/share/acrn/bios/OVMF.fd, size 0x200000
pm by vuart node-index = 0
logger: name=console, level=4
logger: name=kmsg, level=3
logger: name=disk, level=5
vm_create: vm1
VHM api version 1.0
Invalid guest vCPUs (0)

paspndevops, 2019-11-04:
@fuzhongl 

In **NUC7i7BNH** device SOS and UOS working fine with CL 31470
only change i renamed acrn_efi_path=/usr/lib/acrn/acrn.nuc6cayh.sdc.efi   ==  >> **acrn_efi_path=/usr/lib/acrn/acrn.nuc7i7dnb.sdc.efi** 

In NUC7i7DNK still showing **Invalid guest vCPUs (0)**
SOS works after git checkout **acrn-2019w35.1-140000p**

Could you Help us to resolve this issue in **NUC7i7DNK**

Thanks

fuzhongl, 2019-11-04:
@paspndevops Please disable Hyper-thread in BIOS setting. CPU sharing feature needs this modification.
Thanks!

paspndevops, 2019-11-04:
@fuzhongl 

No change same error after disabling HT and I tried with single core also.

fuzhongl, 2019-11-04:
> @fuzhongl
> 
> No change same error after disabling HT and I tried with single core also.

@paspndevops 
Please use CL 31470 also for  NUC7i7DNK.
 After disable HT on  NUC7i7DNK; there should no cpu4 and more
cpu1 online=1
cpu2 online=1
cpu3 online=1
cpu4 online=1
cpu5 online=1
cpu6 online=1
cpu7 online=1

closed 2019-11-20 by paspndevops