projectacrn/acrn-hypervisor #4010: Bootning in blind mode

paspndevops, 2019-11-06:
Hi

As we tried https://projectacrn.github.io/latest/tutorials/running_ubun_as_user_vm.html

on clrlinux 31470 

errot: no suitable video mode found
error: no video mode activated.
error: can't find command 'hwmatch' .errir no suitable video mode found

Thanks in advance

ttzeng, 2019-11-06:
@paspndevops did you attach 2nd display while launching the Ubuntu user VM? ACRN leverages [GVT-g](https://01.org/igvt-g) technology to share the Intel GPU among VMs. It's recommended to connect dual displays to your target device to be aware of the status of the Service VM and User VM.

paspndevops, 2019-11-06:
@ttzeng 

Thanks for quick reply but i could't understand could you brief the steps to solve this issue .

What are the steps need to follow  ?

Thanks again

paspndevops, 2019-11-06:
@ttzeng 

I am using NUC7i7BNH which was having only one HDMI

paspndevops, 2019-11-06:
@ttzeng 
**After modifying Launch script for to enable console (-s 9,virtio-console,@stdio:stdio_port )**

# **sh launch_fedora.sh** 
cpu1 online=1
vtcon: stdio is used by other device
   pci virtio-console init failed
  modify_bar_registration: bypass for pci-gvt
    modify_bar_registration: bypass for pci-gvt
                                                                                                                                                    modify_bar_registration: bypass for pci-gvt
                                                                                                                                                                                                   vtnet tx thread closing...
                                                                                                                                                                                                                             vm_reset_ptdev_intx:0x28-fb, ioapic virpin=0.
                            mngr_client_new: Failed to accept from fd 34, err: Invalid argument
                                                                                               Stop listening 34...
                                                                                                                   Stop polling 34...
                                                                                                                                     No correct pm notify channel given
                                                                                                                                                                       Unable to init vdev (22)

Mingyuan18, 2019-11-07:
[External_System_ID] ACRN-5610

gvancuts, 2019-11-07:
@paspndevops what is the version of the User VM you're trying to launch? Ubuntu, Fedora, which version?

This could be an upstream bug such as this one: https://bugs.launchpad.net/ubuntu/+source/grub2-signed/+bug/1840560

closed 2019-11-20 by paspndevops