projectacrn/acrn-hypervisor #4014: Launch two UOSes with two video outputs on Intel NUC i7DNHE

marjano89, 2019-11-06:
Hello,

Is there a possibility to launch SOS and two rich UOSes (Celadon and AGL) while having only two HDMI ports without physical display port on Intel NUC i7DNHE? For my purposes I shall not use SOS screen at all, only console via SSH. Can I reassign the primary display from SOS?

I found this info:

> The i915 GPU supports three hardware pipes to drive the displays, however only certain products are designed with circuitry needed to connect to three external displays. On a system supporting two external displays, because the primary display is assigned to the Service VM at boot time, you may remove the -s 2,pci-gvt -G "$3" options in one of the previous VM-launching example scripts to completely disable the GVT-g feature for that VM. Refer the GVT-g high-level design for detailed information.

Currently I can only see one UOS on external HDMI and SOS console on another one. If it is not possible without 3rd physical output, will solution with [USB3.0 HDMI expander](https://www.gorite.com/intel-nuc-dual-hdmi-lid-for-7th-gen-dawson-canyon-nuc ) work as a workaround for my problem?

Mingyuan18, 2019-11-07:
[External_System_ID] ACRN-5610

fuzhongl, 2019-11-25:
@marjano89 
Please change i915.domain_plane_owners=0x011111110000 
to 
i915.domain_plane_owners=0x011111112222
in acrn.conf

Then vm1 will use PipeB and vm2 will use PipeA.
Please check detail information with following link:
https://projectacrn.github.io/latest/user-guides/kernel-parameters.html#i915-avail-planes-owners

Thanks!

extramask93, 2020-01-09:
@fuzhongl 
I've took over @marjano89 job and setting domain_plane_owners did help somewhat. Now the primary screen is indeed  not given to the SOS ( remains black ) but is not assigned to the second UOS either. So the problem remains. There are our configs:  https://github.com/extramask93/acrn_config

fuzhongl, 2020-01-09:
@extramask93 Do you use Celadon as UOS?
Celadon is hard code for domain_plane_owners; please contact with Celadon opensource team about how to modify it.
Thanks!

fuzhongl, 2020-01-10:
@extramask93 Checked with Celadon expert; You need modify following file:
groups/graphics/mesa/BoardConfig.mk 
Seeing your setting in launch_android_2.sh is **launch_win 2 1 "64 448 8" 0x00000F**
Do you set following parameter in one Celadon image:

BOARD_KERNEL_CMDLINE += **i915.avail_planes_per_pipe=0x00000F**   i915.enable_hangcheck=0 i915.enable_guc_loading=0   i915.enable_guc_submission=0 i915.enable_guc=0 intel_iommu=off nohpet   tsc=reliable
--
If not, please do it and re-build the image; this image will use PipeA for display.
Thanks!

extramask93, 2020-01-10:
@fuzhongl 
I didn't set it in the BoardConfig.mk. But now having rebuilded the img with provided params for the second Celadon instance -  it falls into a bootloop. Here is the console log:
https://pastebin.com/NkCLyfsR
Can You help me with that?
Thank You !

fuzhongl, 2020-05-05:
Sorry, I'm not familiar with Celadon.
Could you please ask help in Celadon open source project?
Thanks!

fuzhongl, 2020-06-28:
Celadon project related configuration.

closed 2020-06-28 by fuzhongl