projectacrn/acrn-hypervisor #4061: Some scripts are missing license and copyright header

dbkinder, 2019-11-08:
Some scripts in the repo are missing a copyright and license header such as:
```
# Copyright (C) 2019 Intel Corporation.
# SPDX-License-Identifier: BSD-3-Clause
```

```
hypervisor/scripts/genld.sh
devicemodel/samples/apl-mrb/launch_uos.sh
devicemodel/samples/nuc/launch_uos.sh
devicemodel/samples/nuc/launch_win.sh
devicemodel/samples/nuc/launch_hard_rt_vm.sh
devicemodel/samples/nuc/launch_zephyr.sh
devicemodel/samples/nuc/launch_vxworks.sh
devicemodel/samples/apl-up2/launch_uos.sh
```

NanlinXie, 2019-11-08:
@lirui, Please update the script, thanks!

lirui34, 2019-11-11:
[External_System_ID] ACRN-5641

Mingyuan18, 2019-11-20:
No regression issue found, so close this issue.
ACRN tag: acrn-2019w46.2-140000p
CL: 31630

closed 2019-11-20 by Mingyuan18