projectacrn/acrn-hypervisor #4072: hv: add printf "not support the value of vuart index parameter" in function vuart_register_io_handler

xiajidong, 2019-11-11:
When call vuart_register_io_handler function,the  parameter vuart_idx is greater than or equal to 2.should print the not support vuart index value whitch will not register the vuart.

Mingyuan18, 2019-11-12:
@xiajidong , Please add your email information to your git account. Thank you

Mingyuan18, 2019-11-12:
[External_System_ID] ACRN-5669

xiajidong, 2019-11-12:
> @xiajidong , Please add your email information to your git account. Thank you

xiajidong, 2019-11-12:
> @xiajidong , Please add your email information to your git account. Thank you

@Mingyuan18 ,I have already add my email information to my git account in Settings,in Setting options have email information，and I add it in there，is anyother place which I should to add？
Thanks
Best Regards

Mingyuan18, 2019-11-12:
@xiajidong ，Do you plan to submit this patch?

xiajidong, 2019-11-12:
Yes，I plan to submit this patch.
Thanks,
Best Regards

xiajidong, 2019-11-12:
> @xiajidong ，Do you plan to submit this patch?

@Mingyuan18 ，Yes,I plan to submit this patch.and any other things I can do ?
Thanks，
Best Regards

fuzhongl, 2020-05-05:
@xiajidong Can this git issue be closed now?
Thanks!

xiajidong, 2020-05-06:
ok,I will close it now,ths.

closed 2020-05-06 by xiajidong