projectacrn/acrn-hypervisor #4099: Boot issue on non Apollo/Kaby lake

PO-88, 2019-11-12:
Hi,

As we don’t have access to Apollo/Kaby lake for now, we are trying to get ACRN working on IvyBridge. As the dependency is with x86, we expect this to be working. But unfortunately we run into problems.

I have gone through the docs and followed instructions under section “**Board and VM configuration workflow**” from below link and was successful in generating the board.xml and was able to build ACRN with the board src and scenario src xml generated.
https://projectacrn.github.io/latest/tutorials/acrn_configuration_tool.html#vm-config-workflow

I did build the kernel from below link under section “**Update kernel image and modules of Pre-launched VM**”
https://projectacrn.github.io/latest/tutorials/using_partition_mode_on_nuc.html

I have updated the grub with below menu entry:
menuentry 'ACRN hypervisor Logical Partition Scenario' --class ubuntu --class gnu-linux --class gnu --class os $menuentry_id_option 'gnulinux-simple-e23c76ae-b06d-4a6e-ad42-46b8eedfd7d3' {
        recordfail	
        load_video
        gfxmode $linux_gfx_mode
        insmod gzio
        insmod part_gpt
        insmod ext2

        echo 'Loading hypervisor SDC scenario ...'
        multiboot --quirk-modules-after-kernel /boot/acrn.32.out
        module /boot/bzImage Linux_bzImage
}
I'm following direct boot mode and I expect to see the ACRN shell after this and then later I can launch the VM kernels. But with the above config it only prints “Loading hypervisor SDC scenario” and boot stalls. 
I’m not sure if uart is creating the problem somehow, as from the reading I did I got to know uart would stall the boot if the serial port isn't specified right. 
With this I’m currently blocked, is it possible to get any info on this to unblock and proceed further?
Any suggestions would be helpful, also please correct me if my understand with the set-up is wrong.

Thanks in advance
P

cpuinfo attached.
[cpuinfo.txt](https://github.com/projectacrn/acrn-hypervisor/files/3835449/cpuinfo.txt)

Mingyuan18, 2019-11-13:
[External_System_ID] ACRN-5680

jsun26intel, 2019-11-17:
@PO-88 
could you please attach board.xml/sdc.xml? We need to check your serial settings.

PO-88, 2019-11-18:
@jsun26intel 

Earlier machine we tried was with Sandy Bridge (i5-2540M), with the serial log we figured out that there was no SMAP support and it failed to boot. We did some reading and got to know SMAP is 
supported form Broadwell processors.

We were able to get hold of processor with coffeelake arch, but we still end up with boot issue.
Unfortunately there is no serial port available on this machine and we are clueless.

I have attached the board.xml, sdc.xml generated on coffeelake and cpuinfo for the same.
With sdc.xml I had to update vUART base to INVALID_COM_BASE as the generated file was with empty fiels and the board_cfg_gen.py would fail.(updated lines in file sdc.xml 28, 23, 61, 66)

[xmls.zip](https://github.com/projectacrn/acrn-hypervisor/files/3856891/xmls.zip)
[cpuinfo_i3_8100.txt](https://github.com/projectacrn/acrn-hypervisor/files/3856892/cpuinfo_i3_8100.txt)

PO-88, 2019-11-20:
@jsun26intel
On coffeelake I just deviated with the boot mode, with de-privilaged boot mode I was able to get the ACRN installed where ubuntu is used as sos.
Followed below link:
https://projectacrn.github.io/latest/tutorials/using_ubuntu_as_sos.html#start-the-user-os-uos

With dmesg I'm able to see logs with ACRN hypervisor getting detected
# dmesg | grep -i ACRN
>>[    0.000000] Hypervisor detected: ACRN
>>[    1.107659] ACRNTrace: acrn_trace_init, cpu_num 4
>>[    1.109729] ACRN HVLog: acrn_hvlog_init
I'm not able to launch the, user OS as the ACRN shell commands are not available.
When I run the launch acript, i get this:
./launch_uos.sh 
cpu1 online=0
cpu2 online=0
cpu3 online=0
./launch_uos.sh: line 105: acrn-dm: command not found

Am i missing something here?
Thanks!!

PO-88, 2019-11-20:
> @PO-88
> could you please attach board.xml/sdc.xml? We need to check your serial settings.

@jsun26intel

Earlier machine we tried was with Sandy Bridge (i5-2540M), with the serial log we figured out that there was no SMAP support and it failed to boot. We did some reading and got to know SMAP is
supported form Broadwell processors.

We were able to get hold of processor with coffeelake arch, but we still end up with boot issue.
Unfortunately there is no serial port available on this machine and we are clueless.

I have attached the board.xml, sdc.xml generated on coffeelake and cpuinfo for the same.
With sdc.xml I had to update vUART base to INVALID_COM_BASE as the generated file was with empty fiels and the board_cfg_gen.py would fail.(updated lines in file sdc.xml 28, 23, 61, 66)

[cpuinfo_i3_8100.txt](https://github.com/projectacrn/acrn-hypervisor/files/3868018/cpuinfo_i3_8100.txt)
[xmls.zip](https://github.com/projectacrn/acrn-hypervisor/files/3868021/xmls.zip)

PO-88, 2019-11-22:
Got this working with de-privileged boot mode.
Was able to boot ACRN hypervisor on coffeelake.
 
[    0.000000] Hypervisor detected: ACRN
[    0.737444] ACRNTrace: Initialized acrn trace module with 4 cpu

Next step is to get the UOS working.
We also need to check why direct boot mode did not work.
Closing this post for now.
Thanks

closed 2019-11-22 by PO-88