projectacrn/acrn-hypervisor #4116: How to set CPU Core UOS

paspndevops, 2019-11-13:
Hi,

Could you please help me how to set single core for UOS vm .i am using VERSION_ID=31620.'Launch_fedora.sh'

HW Deatils:
       product: NUC7i7BNB
       vendor: Intel Corporation
       physical id: 0
       version: J31145-304
       serial: GEBN72200F0J
       slot: Default string
     *-firmware
          description: BIOS
          vendor: Intel Corp.
          physical id: 0
          version: BNKBL357.86A.0080.2019.0725.1139
          date: 07/25/2019
          size: 64KiB
          capacity: 8128KiB

Mingyuan18, 2019-11-14:
[External_System_ID] ACRN-5682

fuzhongl, 2019-11-14:
@paspndevops Which UOS are you using? The default is one core for non-RT VM.
Thanks!

paspndevops, 2019-11-14:
@fuzhongl 

we are using fedora 30 as a UOS now we would like to run 'two' VM's working in parallel 
One is fedora 30 and other one is Ubuntu 18.04

Thanks in advance

fuzhongl, 2019-11-14:
@paspndevops The fedora 30 and Ubuntu 18.04 will both use PipeA.
So you must set one to headless.
Thanks!

paspndevops, 2019-11-14:
@fuzhongl 

Thanks for your reply i am using both VM's as headless could you please help me.

How to start both headless VM's in my SOS with two HDMI's

Thanks

fuzhongl, 2019-11-14:
@paspndevops 
losetup –f –P --show  /your_path_to/uos.img
Mount the rootfs partition and  vim /mnt/etc/default/grub 
Update: GRUB_CMDLINE_LINUX_DEFAULT="nomodeset"  and GRUB_TERMINAL=console
sudo update-grub

Thanks!

paspndevops, 2019-11-14:
@fuzhongl 

Could you please help on following points:

- How to set two cores for fedora 30 headless UOS server 
- where to modify UOS launcher script to set hdmi1/hdmi2

Here i attached my working launcher script  
Thanks
[launch_fedora.txt](https://github.com/projectacrn/acrn-hypervisor/files/3845201/launch_fedora.txt)

fuzhongl, 2019-11-14:
@paspndevops 
1) you can refer this commit to set two cores for fedora 30, please using UUID 495ae2e5-2603-4d64-af76-d4bc5a8ec0e5 in launch script:
d5c3523d304ee6ae02fb04972590457f00cc63ea 

2)you need modify  acrn.conf:
i915.domain_plane_owners=0x011111110000  -> i915.domain_plane_owners=0x011100001111
With this setting, SOS will use PipeB.
Thanks！

paspndevops, 2019-12-12:
@fuzhongl 

Sorry for late reply i tried to both UOS as headless and unable to set hdmi 2 for 2nd VM.
cat /proc/cmdline:

BOOT_IMAGE=/boot/acrn/org.clearlinux.iot-lts2018-sos.4.19.78-98 root=UUID=25bc697e-3e39-4c0a-8b48-a75fdd191fd0 pci_devices_ignore=(0:18:1) console=tty0 console=ttyS0 rw rootwait ignore_loglevel no_timer_check consoleblank=0 i915.nuclear_pageflip=1 i915.avail_planes_per_pipe=0x01010F i915.domain_plane_owners=0x011111110000 i915.enable_gvt=1 i915.enable_guc=0 hvlog=2M@0x1FE00000 quiet splash vt.handoff=1

My quires are:
1. Specific reason for to use only NUC, shell i use my intel laptop for acrn
2. How to set HDMI-2 for UOS2
3. How to get less CPU latency in UOS(fedora)

Thanks in advance,

fuzhongl, 2019-12-12:
@paspndevops  for item 2,please modify in acrn.conf as following 
from: 
i915.domain_plane_owners=0x011111110000 
to: 
i915.domain_plane_owners=0x011100001111

paspndevops, 2019-12-13:
@fuzhongl 

could you please guide me how to configure device manager(acrn-dm) for the below device:

$ lspci -nn -s 0:02.0
00:02.0 VGA compatible controller [0300]: Intel Corporation Broadwell-U
Integrated Graphics [8086:193b] (rev 09)

[   16.750769] usbcore: registered new interface driver myriad_usb_vsc
[219380.723614] myriad_usb_vsc 12-1:1.0: USB VSC device now attached to USBVSC-1

Thanks!

fuzhongl, 2019-12-13:
@paspndevops  Do you want to passthrough it to Guest VM?
We perfer gvt-g, not gvt-d.
Thanks!

paspndevops, 2019-12-13:
@fuzhongl 

Could you please provide Guest VM modified "acrn-dm" arguments 

acrn-dm -A -m $mem_size -s 0:0,hostbridge \
  -s 2,pci-gvt -G "$2" \
  -s 5,virtio-console,@stdio:stdio_port \
  -s 6,virtio-hyper_dmabuf \
  -s 3,virtio-blk,/root/ubuntu_images/uosFC.img \
  -s 4,virtio-net,tap0 \
  -s 7,virtio-rnd \
  --ovmf /usr/share/acrn/bios/OVMF.fd \
  $pm_channel $pm_by_vuart $pm_vuart_node \
  $logger_setting \
  --mac_seed $mac_seed \
  -U d2795438-25d6-11e8-864e-cb7a18b34643 \

Thanks!

fuzhongl, 2019-12-13:
@paspndevops 
Please check following link:
https://projectacrn.github.io/latest/user-guides/acrn-dm-parameters.html
Thanks!

closed 2019-12-16 by paspndevops