projectacrn/acrn-hypervisor #4121: Machine Check Error on Page Size Change.(release v1.4)

binbinwu1, 2019-11-14:
Machine Check Error on Page Size Change
Instruction fetch may cause machine check error if page size
and memory type was changed without invalidation on some
processors. Malicious guest kernel could trigger this issue.

Mingyuan18, 2019-11-14:
[External_System_ID] ACRN-5687

Mingyuan18, 2019-12-09:
The patch was merged and no regression issue found in weekly test.

closed 2019-12-09 by Mingyuan18

jzySaber1996, 2024-05-10:
Hello, we're investigating the issue reports in security. Since we observed that this issue may relate to a potential vulnerability, has it been disclosed in CVE already? Hope to receive your reply.

binbinwu1, 2024-05-10:
> Hello, we're investigating the issue reports in security. Since we observed that this issue may relate to a potential vulnerability, has it been disclosed in CVE already? Hope to receive your reply.

Yes, Machine Check Error on Page Size Change is assigned as CVE-2018-12207.
You can refer to the page https://www.intel.com/content/www/us/en/developer/articles/troubleshooting/software-security-guidance/technical-documentation/machine-check-error-avoidance-page-size-change.html for details.