projectacrn/acrn-hypervisor #4123: Creating Ubuntu SOS not working - black screen

ErinRJ, 2019-11-18:
Following the instructions found at [Using Ubuntu as the Service OS](https://projectacrn.github.io/latest/tutorials/using_ubuntu_as_sos.html), we were unable to create an SOS with Ubuntu.

Board used: UP2

After following the instructions and booting to the SOS, it just shows a black screen. When rebooting the board, and trying to access the original Ubuntu installed on the board, it can no longer access it.

Since we're using a UP2 board instead of a NUC, we tried the installation with two different configurations:  Using "/mnt", and using "/boot", since the [UP2 Instructions](https://projectacrn.github.io/1.2/getting-started/up2.html) specified using mnt instead of boot. Regardless, both had the black screen SOS problem.

Please let me know if you need any more information and thank you for your time,

Erin

Mingyuan18, 2019-11-18:
[External_System_ID] ACRN-5701

fuzhongl, 2019-11-18:
@ErinRJ Does the native Ubuntu work well on your board?
If yes, could you please share the acrn-dm version info?
Thanks!

ErinRJ, 2019-11-28:
HI Fuzhongl! 
Thank you for your prompt support.
Before configuring the ACRN SOS, Ubuntu worked well and as expected on the board.

Unfortunately, we've actually switched over to a different part of our project for now so we won't be using this operating system configuration at the moment.
If we end up using Ubuntu in the future, I'll create a new issue thread if need be, but for now I'll close it.

Thanks again,
Erin

closed 2019-11-28 by ErinRJ