projectacrn/acrn-hypervisor #4125: vm1 is running, can't create twice!

paspndevops, 2019-11-18:
root@clrlinux~/rpm2 # sh launch_Fed.sh 
cpu1 online=0
cpu2 online=0
cpu3 online=0
**vm1 is running, can't create twice!**

root@clrlinux~/rpm2 # ^C
root@clrlinux~/rpm2 # **acrn-dm -v**
DM version is: 1.4-unstable-”2019w45.5.140000p_250” (daily tag:”2019w45.5.140000p”), build by mockbuild@2019-11-12 19:18:43

==> How to stop vm1 (Every time need to restart system )
==> How to run two VM's in clr-linux(31640)
==>VM1 running successfully 

Thanks 
[launch_FedLog.txt](https://github.com/projectacrn/acrn-hypervisor/files/3857079/launch_FedLog.txt)
[launch_uos_31470Log.txt](https://github.com/projectacrn/acrn-hypervisor/files/3857081/launch_uos_31470Log.txt)

paspndevops, 2019-11-18:
Hi,

Now i am able to run two VM's and 2nd VM giving following error:

[    6.053933] Kernel Offset: 0x35000000 from 0xffffffff81000000 (relocation range: 0xffffffff80000000-0xffffffffbfffffff)
[    6.053933] ---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(253,0) ]---

=============================================================

acrn-dm -A -m $mem_size -s 0:0,hostbridge -s 1:0,lpc -l com1,stdio \
 -s 2,pci-gvt -G "$2" \
 -s 5,virtio-console,@pty:pty_port \
 -s 6,virtio-hyper_dmabuf \
 -s 3,virtio-blk,/root/rpm2/uos.img \
 -s 4,virtio-net,tap1 \
 -s 7,virtio-rnd \
 $logger_setting \
 --mac_seed $mac_seed \
 -U 38158821-5208-4005-b72a-8a609e4190d0 \
 -k /usr/lib/kernel/default-native \
 -B "root=/dev/vda rw rootwait maxcpus=$2 nohpet console=tty0 console=hvc0 \
 console=ttyS0 no_timer_check ignore_loglevel log_buf_len=16M \
 consoleblank=0 tsc=reliable i915.avail_planes_per_pipe=$4 \
 i915.enable_hangcheck=0 i915.nuclear_pageflip=1 i915.enable_guc_loading=0 \
 i915.enable_guc_submission=0 i915.enable_guc=0" $vm_name

gvancuts, 2019-11-18:
That looks like an issue with your virtual storage device. Can you verify that:
1. /root/rpm2/uos.img exists and is valid
2. If there are partitions in that img, make sure to adjust "root=/dev/vda" accordingly, i.e. root=/dev/vda3 if your rootfs is on a third partition inside the img file.

paspndevops, 2019-11-18:
@gvancuts 

Thanks for your reply ..

UUID :
ubuntu-16.04 ==> 495ae2e5-2603-4d64-af76-d4bc5a8ec0e5
But now i am using ' ubuntu-18.04 ' as a 2nd VM what is the UUID for 18.04

gvancuts, 2019-11-18:
Hi @paspndevops , I'm confused by your answer/question. I don't think it's a problem with the UUID.

The kernel is not able to mount the rootfs that is on the VM2 virtual disk, this could be caused by a problem with the /root/rpm2/uos.img file (not there, corrupted) or simply because you are missing the partition number on which the rootfs is located. I don't recall off the top of my head what partition Ubuntu uses by default for the rootfs but if it's the second, you'd want "root=/dev/vda2" as an example in your launch script.

paspndevops, 2019-11-18:
@gvancuts 

I am using following acrn-dm successfully Could you please provide 2nd VM script 👍 

1st VM i would like to use clear-27550-kvm.img
2nd VM is clear-31440-kvm.img  

acrn-dm -A -m $mem_size -c $2 -s 0:0,hostbridge -s 1:0,lpc -l com1,stdio \
 -s 2,pci-gvt -G "$3" \
 -s 5,virtio-console,@pty:pty_port \
 -s 6,virtio-hyper_dmabuf \
 -s 3,virtio-blk,clear-27550-kvm.img \
 -s 4,virtio-net,tap0 \
 $logger_setting \
 --mac_seed $mac_seed \
 -U d2795438-25d6-11e8-864e-cb7a18b34643 \
 -k /usr/lib/kernel/default-iot-lts2018 \
 -B "root=/dev/vda3 rw rootwait maxcpus=$2 nohpet console=tty0 console=hvc0 \
 console=ttyS0 no_timer_check ignore_loglevel log_buf_len=16M \
 consoleblank=0 tsc=reliable i915.avail_planes_per_pipe=$4 \
 i915.enable_hangcheck=0 i915.nuclear_pageflip=1 i915.enable_guc_loading=0 \
 i915.enable_guc_submission=0 i915.enable_guc=0" $vm_name

gvancuts, 2019-11-18:
You'll need to recompile ACRN to use the SDC2 scenarion for that, have you done that already? See https://projectacrn.github.io/latest/tutorials/using_sdc2_mode_on_nuc.html for more details (it also includes the UUIDs you should use).

Please note that you will need to adjust these two lines in the launch script for the second VM:
`-s 3,virtio-blk,ubuntu-16.04.img \` -> `-s 3,virtio-blk,clear-27550-kvm.img \` 
And
`root=/dev/vda` -> `root=/dev/vda3`

In that sense, assuming you are already using SDC2, your script above is correct except for the UUID which you should change to `495ae2e5-2603-4d64-af76-d4bc5a8ec0e5`

Mingyuan18, 2019-11-19:
[External_System_ID] ACRN-5703

fuzhongl, 2019-11-19:
@paspndevops
==> How to stop vm1 (Every time need to restart system )
S5 feature is added since v1.3; please follow the link:
https://projectacrn.github.io/latest/tutorials/enable_s5.html

==> How to run two VM's in clr-linux(31640)
There are both   "vm_name=vm$1" in your two launch script; please change one to difference: something like "vm_name=vm_Fed" 

Thanks!

paspndevops, 2019-11-19:
@fuzhongl 
thanks for your reply now both VM's are working fine but if both are clr-linux.
Now i replace 2nd VM as headless fedora but it was giving error as follows:

Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(253,3)

Just i replace clr-linux.kvm.img to ubuntu working Image

I am using /usr/lib/kernel/default-native  and root=/dev/vda3

Thanks in advance

fuzhongl, 2019-11-19:
@paspndevops
Does headless fedora UOS work if you only lunch it?
Thanks!

paspndevops, 2019-11-19:
@fuzhongl 

- Please find below log advice me what are the modification required for my VM2 launcher :
- Also let me know how to limit the CPU's for each VM (single core for each VM)

cpu1 online=0
cpu2 online=0
cpu3 online=0
passed gvt-g optargs low_gm 64, high_gm 448, fence 8
SW_LOAD: get ovmf path /usr/share/acrn/bios/OVMF.fd, size 0x200000
pm by vuart node-index = 0
logger: name=console, level=4
logger: name=kmsg, level=3
logger: name=disk, level=5
vm_create: vm2
VHM api version 1.0
vm_setup_memory: size=0x80000000
open hugetlbfs file /run/hugepage/acrn/huge_lv1/vm2/D279543825D611E8864ECB7A18B34643
open hugetlbfs file /run/hugepage/acrn/huge_lv2/vm2/D279543825D611E8864ECB7A18B34643
level 0 free/need pages:0/1 page size:0x200000
level 1 free/need pages:0/2 page size:0x40000000
to reserve more free pages:
to reserve pages (+orig 1): echo 3 > /sys/kernel/mm/hugepages/hugepages-1048576kB/nr_hugepages
to reserve pages (+orig 36): echo 37 > /sys/kernel/mm/hugepages/hugepages-2048kB/nr_hugepages
now enough free pages are reserved!

try to setup hugepage with:
	level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
	level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x0
total_size 0x180000000

mmap ptr 0x0x7ff180c25000 -> baseaddr 0x0x7ff1c0000000
mmap 0x80000000@0x7ff1c0000000
touch 2 pages with pagesz 0x40000000
mmap 0x200000@0x7ff2bfe00000
touch 1 pages with pagesz 0x200000

really setup hugepage with:
	level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
	level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x0
vm_init_vdevs
polling 38...
Listening 38...
pci init hostbridge
pci init lpc
pci init pci-gvt
modify_bar_registration: bypass for pci-gvt
modify_bar_registration: bypass for pci-gvt
modify_bar_registration: bypass for pci-gvt
pci init virtio-blk
pci init virtio-net
pci init virtio-console
pci init virtio-hyper_dmabuf
pci init virtio-rnd
tpm: init_vtpm2:Invalid socket path!
                                    /tmp/dm.Xj1r0mS     24:     Device (PCI0)
                                                                             Warning  3073 -        Multiple types ^  (Device object requires either a _HID or _ADR, but not both)

                  /tmp/dm.Xj1r0mS    827:     Processor (CPU0, 0x00, 0x00000000, 0x00) {}
         Warning  3168 -                          ^ Legacy Processor() keyword detected. Use Device() keyword instead.

                                      acrn_sw_load
                                                  SW_LOAD: partition blob /usr/share/acrn/bios/OVMF.fd size 2097152 copy to guest 0xffe00000
                                                            SW_LOAD: build e820 7 entries to addr: 0x7ff1c00ef008
SW_LOAD: entry[0]: addr 0x0000000000000000, size 0x00000000000a0000,  type 0x1
SW_LOAD: entry[1]: addr 0x00000000000a0000, size 0x0000000000060000,  type 0x2
SW_LOAD: entry[2]: addr 0x0000000000100000, size 0x000000007ff00000,  type 0x1
SW_LOAD: entry[3]: addr 0x0000000080000000, size 0x0000000008000000,  type 0x2
SW_LOAD: entry[4]: addr 0x00000000df000000, size 0x0000000001000000,  type 0x2
SW_LOAD: entry[5]: addr 0x00000000e0000000, size 0x0000000020000000,  type 0x2
SW_LOAD: entry[6]: addr 0x0000000100000000, size 0x0000000040000000,  type 0x2
SW_LOAD: ovmf_entry 0xfffffff0
                              add_cpu
                                     out instr on NMI port (0x61) not supported
                                                                               out instr on NMI port (0x61) not supported
                                         out instr on NMI port (0x61) not supported
   Unhandled ps2 mouse command 0xe1
                                   Unhandled ps2 mouse command 0x88

Fedora 30 (Server Edition)
[root@localhost ~]# lsblk
NAME            MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
vda             252:0    0   20G  0 disk 
├─vda1          252:1    0  200M  0 part /boot/efi
├─vda2          252:2    0    1G  0 part /boot
└─vda3          252:3    0 18.8G  0 part 
  ├─fedora-root 253:0    0   15G  0 lvm  /
  └─fedora-swap 253:1    0    2G  0 lvm  [SWAP]

fuzhongl, 2019-11-19:
@paspndevops  LVM stands for Logical Volume Management.
ACRN still don't support this feature; please create Fedora image with ext4 format rootfs manually.
Thanks!

paspndevops, 2019-11-19:
@fuzhongl

Also let me know how to limit the CPU's for each VM (single core for each VM)

Thanks !

fuzhongl, 2019-11-19:
@paspndevops 
By default, there is one core per UOS; except for RTVM with UUID: 495ae2e5-2603-4d64-af76-d4bc5a8ec0e5
Thanks!

fuzhongl, 2019-11-20:
@paspndevops Does Fedora UOS launch successfully on you side?
Thanks!

paspndevops, 2019-11-20:
@fuzhongl 
Yes now it was working fine thanks for help 

Now i am trying to launch UOS on UBUNTU SOS showing invalid vCPU's(0)

Thanks !

fuzhongl, 2019-11-20:
@paspndevops  Could you please raise another git issue for this issue with right title? since the root cause isn't same with before issue.
Thanks!

Mingyuan18, 2019-12-09:
@paspndevops ,Could you please close this issue and create another Git issue to track the new problem? Thanks!

paspndevops, 2019-12-09:
Launcher script modified issue resolved thanks again

closed 2019-12-09 by paspndevops