projectacrn/acrn-hypervisor #4126: Trying to run ACRN on a laptop

yfliuu, 2019-11-18:
Hi!

I am recently trying to run ACRN release 1.4 (BOARD=generic, SCENARIO=hybrid) on top of a Dell Latitude E5270 laptop (Intel Core i5-6300U@ 2.4Ghz*4, HD Graphics 520, 8GB memory) for my research purposes. 

I have followed the "Using Ubuntu 18.04 as the Service OS" tutorial and after tweaking some memory settings and change some EFI boot code (boot arguments parsing in efi_main) it boots and loads grubx64.efi. The GRUB menu showed up but it was EXTREMELY slow: it took about 10 seconds for the count down timer to decrement from, say, 5s to 4s. The keyboard also didn't work. 

I understand that the hardware I am using is not officially supported. Still, could you please point out where things could possibly go wrong?

Mingyuan18, 2019-11-19:
[External_System_ID] ACRN-5704

fuzhongl, 2019-11-19:
@yfliuu Could you please attend TCM on 11/20/2019?
We will talk about config-tool on how to enable acrn on different platform.
Thanks!

yfliuu, 2019-11-19:
@fuzhongl Unfortunately I will not be available that time. Will there be any recordings of the conference?

yfliuu, 2019-11-21:
OK, I have figured out that the slow countdown problem is a GRUB2 problem. Some have reported similar issues [here](https://askubuntu.com/questions/947458/grub2-countdown-slow) and [here](https://forums.linuxmint.com/viewtopic.php?t=209231).

After the count down timer expires (Be patient even after it expires!)  finally Ubuntu boots successfully. The interesting thing is, builtin keyboard of the laptop doesn't work. You have to use a USB keyboard. Other than that everything seems fine, dmesg shows the expected ACRN logs.

Any suggestions on making builtin keyboard work?

skisssbb, 2020-05-12:
@fuzhongl would you mind clarify which parameters from 
https://github.com/projectacrn/acrn-hypervisor/blob/master/hypervisor/arch/x86/Kconfig
need to be specified in order for the built-in keyboard to work?
Since 
```
make menuconfig -C hypervisor BOARD=nuc6cayh SDC
```
throws an error `Can not find executable menuconfig`

fuzhongl, 2020-05-12:
@skisssbb Please using make menuconfig and select following item to build kernel:
Device Drivers --->
HID support --->
USB HID support --->
<> USB HID transport layer
[] PID device support
[*] /dev/hiddev raw HID device support

Thanks!

gvancuts, 2020-05-12:
> @fuzhongl would you mind clarify which parameters from
> https://github.com/projectacrn/acrn-hypervisor/blob/master/hypervisor/arch/x86/Kconfig
> need to be specified in order for the built-in keyboard to work?

@fuzhongl is talking about drivers to be enabled in the Service VM (SOS) kernel, you can use the one we maintain in https://github.com/projectacrn/acrn-kernel. The procedure to do that is roughly as follows:
```
$ git clone https://github.com/projectacrn/acrn-kernel.git
Cloning into 'acrn-kernel'...
...
$ cd acrn-kernel
$ cp kernel_config_uefi_sos .config
$ make olddefconfig
$ make menuconfig -> make recommended changes there and save
$ make
$ make modules_install INSTALL_MOD_PATH=out/
```
Then transfer your kernel binary *and* modules to your target

> Since
> 
> ```
> make menuconfig -C hypervisor BOARD=nuc6cayh SDC
> ```

This is not entirely correct, please use `make menuconfig -C hypervisor BOARD=nuc6cayh SCENARIO=SDC`

> 
> throws an error `Can not find executable menuconfig`

Did you install the `kconfiglib` Python package? (e.g. `sudo pip3 install kconfiglib`)

If you did, as a user (`--user`), please make sure the path to that module is correctly set-up in your environment.

skisssbb, 2020-05-18:
@gvancuts I followed the instructions, then transfer my kernel binary and modules to the target with:
```
cp -r out/lib/modules/5.4.28-PKT-200203T060100Z/ /lib/modules/
cp arch/x86/boot/bzImage /boot/acrn/5.4.28-PKT-200203T060100Z
```
And also adjusted `/etc/grub.d/40_custom`.

The menuconfig is configured as follow:
> Device Drivers --->
> HID support --->
> USB HID support --->
> <M> USB HID transport layer
> [] PID device support
> [*] /dev/hiddev raw HID device support
> USB HID Boot Protocol drivers  --->
>> <\*> USB HIDBP Keyboard (simple Boot) support 
>> <\*> USB HIDBP Mouse (simple Boot) support

If `<>  USB HID transport layer` is unselected, `[*] /dev/hiddev raw HID device support` doesnt show up. I also tried the scenario `<*> USB HID transport layer`.

However, the built-in keyboard and mouse didnt work. I used the screen-keyboard with my USB mouse to check which Kernel is running and the status of ACRN and got followings:
```
dmesg | grep ACRN
> [    1.545934] ACRNTrace: acrn_trace: not support acrn hypervisor!
> [    1.545935] ACRN HVLog: acrn_hvlog: not running under acrn
uname -a hypervisor!
> 5.4.28-PKT-200203T060100Z
```
You can find the .config file and  `make log` here: https://file.io/k1IqhPdO

gvancuts, 2020-05-18:
I cannot see the log output on file.io, can you share it some other way?

Can you share the content of your `/etc/grub.d/40_custom`?

You're not running under ACRN, what does `efibootmgr -v` returns?

skisssbb, 2020-05-18:
https://filebin.net/r0zmu12o6nzmmoxb
`efibootmgr -v` returns:

>BootCurrent: 0000
Timeout: 0 seconds
No BootOrder is set; firmware will attempt recovery
Boot0000* ubuntu	HD(4,GPT,09dd38cd-927d-4cf3-998f-328abf51c695,0x2baa3800,0x47800)/File(\EFI\ubuntu\shimx64.efi)
Boot0001* ubuntu	HD(4,GPT,09dd38cd-927d-4cf3-998f-328abf51c695,0x2baa3800,0x47800)/File(\EFI\ubuntu\shimx64.efi)
Boot0002* ubuntu	HD(4,GPT,09dd38cd-927d-4cf3-998f-328abf51c695,0x2baa3800,0x47800)/File(\EFI\ubuntu\shimx64.efi)
Boot0003* ubuntu	HD(4,GPT,09dd38cd-927d-4cf3-998f-328abf51c695,0x2baa3800,0x47800)/File(\EFI\ubuntu\shimx64.efi)
Boot0004* ubuntu	HD(4,GPT,09dd38cd-927d-4cf3-998f-328abf51c695,0x2baa3800,0x47800)/File(\EFI\ubuntu\shimx64.efi)
Boot0005* ACRN Hypervisor	HD(1,GPT,14add027-8bc1-412e-a6d8-8f22e4cac5fa,0x800,0xe8e1000)/File(\EFI\acrn\acrn.efi)b.o.o.t.l.o.a.d.e.r.=.\.E.F.I.\.u.b.u.n.t.u.\.g.r.u.b.x.6.4...e.f.i. .

But i choose ACRN Hypervisor to boot at grub menu during start up. That's why the built in keyboard doesnt work

gvancuts, 2020-05-18:
Thanks @skisssbb 

I don't see the `[*] /dev/hiddev raw HID device support` option enabled in your kernel config - see
```#
# USB HID support
#
# CONFIG_USB_HID is not set
# CONFIG_HID_PID is not set

#
# USB HID Boot Protocol drivers
#
CONFIG_USB_KBD=y
CONFIG_USB_MOUSE=y
# end of USB HID Boot Protocol drivers
# end of USB HID support
```

I see this on my kernel config after enabling that:
```
#                                                                                                                       
# USB HID support                                                                                                       
#                                                                                                                       
CONFIG_USB_HID=y                                                                                                        
# CONFIG_HID_PID is not set                                                                                             
CONFIG_USB_HIDDEV=y                                                                                                     
# end of USB HID support  
```

skisssbb, 2020-05-19:
Hi @gvancuts , thank you for the quick response. I have tried the whole process from scratch (clone the kernel and build) with the new .config file: https://filebin.net/n9hlg58170c6pjyr/configFile?t=11z2fasg
Still the built-in keyboard doesnt seem to work. I just bought a usb keyboard and going to use it to bypass the problem for now

skisssbb, 2020-05-19:
Hi @gvancuts , thank you for the quick response. I have tried the whole process from scratch (clone the kernel and build) with the new .config file: https://filebin.net/n9hlg58170c6pjyr/configFile?t=11z2fasg
Still the built-in keyboard doesnt seem to work. I just bought a usb keyboard and going to use it to bypass the problem for now.

However, ACRN doesn't seem to run. I got following output:

```
$ sudo dmesg | grep ACRN
[    2.007727] ACRNTrace: acrn_trace: not support acrn hypervisor!
[    2.007728] ACRN HVLog: acrn_hvlog: not running under acrn hypervisor!

$ uname -a
Linux bi-HP-EliteBook-2570p 4.19.120-108.iot-lts2018-sos #1 SMP PREEMPT Mon May 4 18:13:04 PDT 2020 x86_64 x86_64 x86_64 GNU/Linux

$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04 LTS
Release:	20.04
Codename:	focal
```

Hardware: HP Elitebook 2570p (Intel® Core™ i5-3320M CPU @ 2.60GHz × 4)
Ubuntu20.04 LTS
Kernel: linux-iot-lts2018-sos-4.19.120-108.x86_64.rpm (From https://download.clearlinux.org/releases/current/clear/x86_64/os/Packages/ as provided by the tutorial)
ACRN: Current master branch commit 9d3f2066c1c81fbcf2fdfa1431cd56f4e74f9174

Any idea what why it doesnt work?

gvancuts, 2020-05-20:
> Hi @gvancuts , thank you for the quick response. I have tried the whole process from scratch (clone the kernel and build) with the new .config file: [...]

@fuzhongl , are you confident this is the driver that was missing?

I don't have such system available to test so I'm not sure how to help. If you have a native installation that works, you can try to figure out what driver is used for the device.

gvancuts, 2020-05-20:
> Hardware: HP Elitebook 2570p (Intel® Core™ i5-3320M CPU @ 2.60GHz × 4)

That's a different laptop than the one you were using before, you were using a Dell before. Is the same problem seen on both laptops?

This one is based on an Ivy-Bridge processor. AFAIK, we have never tested that family of processor with ACRN.

What does `efibootmgr -v` return on the laptop?

Do you have access to an EFI shell to try run `acrn.efi` manually from there?

fuzhongl, 2020-05-21:
@skisssbb Sorry, I mis-understanding your question about **builtin keyboard**
I thought it as  build-in USB keyboard in kernel before.
Let me have a try and back to you.

Thanks!

fuzhongl, 2020-05-21:
@skisssbb Checked with HV expert, this kernel build-in keyboard config is related with hardware vendor, Please boot from native(make sure the build-in keyboard works ); then share the log of `dmesg`.
Thanks!

skisssbb, 2020-05-21:
Hi @gvancuts, @yfliuu, the one who opened the issue uses Dell laptop, not me. His processor is Skylake.
As for `efibootmgr -v`, whenever I try to set the boot order with `sudo efibootmgr -o 0005` and reboot, it always go back to following output:
```
BootCurrent: 0000
Timeout: 0 seconds
No BootOrder is set; firmware will attempt recovery
Boot0000* ubuntu	HD(4,GPT,09dd38cd-927d-4cf3-998f-328abf51c695,0x2baa3800,0x47800)/File(\EFI\ubuntu\shimx64.efi)
Boot0001* ubuntu	HD(4,GPT,09dd38cd-927d-4cf3-998f-328abf51c695,0x2baa3800,0x47800)/File(\EFI\ubuntu\shimx64.efi)
Boot0002* ubuntu	HD(4,GPT,09dd38cd-927d-4cf3-998f-328abf51c695,0x2baa3800,0x47800)/File(\EFI\ubuntu\shimx64.efi)
Boot0003* ubuntu	HD(4,GPT,09dd38cd-927d-4cf3-998f-328abf51c695,0x2baa3800,0x47800)/File(\EFI\ubuntu\shimx64.efi)
Boot0004* ubuntu	HD(4,GPT,09dd38cd-927d-4cf3-998f-328abf51c695,0x2baa3800,0x47800)/File(\EFI\ubuntu\shimx64.efi)
Boot0005* ACRN Hypervisor	HD(1,GPT,14add027-8bc1-412e-a6d8-8f22e4cac5fa,0x800,0xe8e1000)/File(\EFI\acrn\acrn.efi)b.o.o.t.l.o.a.d.e.r.=.\.E.F.I.\.u.b.u.n.t.u.\.g.r.u.b.x.6.4...e.f.i. .
```

I tried to choose the EFI during boot via `F10` but the boot process is stuck at
```
detected 4 processes, 4 enabled
current on process 0
```
Issue #3488 and #4149 seem to have the same problem. I tried setting BIOS configuration as described in #3488 to:
> Item 	Setting
> VMX 	Enabled
> VT-D 	Enabled
> Hyper-Threading 	Disabled
> Secure Boot 	Disabled

However, its still stuck at ... when I choose to boot `acrn.efi` manually.
```
detected 4 processes, 4 enabled
current on process 0
```
Hi @fuzhongl, I dont quite understand what booting from native mean. I tried to boot with 3 different kernels: Ubuntu Kernel:`5.4.0-31-generic`, Clear Linux Kernel: `4.19.120-108.iot-lts2018-sos` and the kernel pulled from Github managed by the team: `5.4.39-PKT-200203T060100Z-97611-g095509221660`. ACRN doesn't run successfully for any scenario. Here are the log files: https://filebin.net/ob36p2nh71ru9osi

Im going to try the tutorial https://projectacrn.github.io/latest/tutorials/using_ubuntu_as_sos.html again with the exact Ubuntu/ACRN Hypervisor and Kernel described in the tutorial and keep you guys updated.

yfliuu, 2020-05-21:
> detected 4 processes, 4 enabled
current on process 0

I've had similar issue before. This means ACRN did not boot successfully. I managed to boot successfully on ACRN 1.4 but failed on 1.5 (and I did not try newer version). 

By the way I had tried to list the discovered devices on native Ubuntu and Ubuntu on ACRN and the keyboard (trackpad too) was missing on the latter one. Seems that ACRN did not expose serial devices to the VMs (correct me if I'm wrong). Unfortunately I am not able to reproduce the result now because I've reinstalled everything...

fuzhongl, 2020-05-21:
@yfliuu The build-in keyboard is serial device for your desktop, will check with ACRN expert about how to expose it to VMs.
Thanks for your update! 
@skisssbb You issue is ACRN can't boot, correct?
Please boot with `Ubuntu Kernel:5.4.0-31-generic` and share `lshw` log to us.
Thanks!

skisssbb, 2020-05-25:
@yfliuu can you clarify on how you were able to boot ACRN 1.4 successfully? I redid the whole process exactly like https://projectacrn.github.io/latest/tutorials/using_ubuntu_as_sos.html#install-ubuntu-natively down to the versions, however I now get `Kernel Panic error`:
[image]
When I try to `boot with EFI file acrn.efi`, i got stuck with the same thing:

> detected 4 processes, 4 enabled
current on process 0

@fuzhongl I reinstalled everything with Ubuntu 18.04.4 LTS like in the tutorial. Here is the `lshw` log for Ubuntu Kernel: `4.15.0-101-generic`
https://filebin.net/ckckaauefuag3ijq

yfliuu, 2020-05-25:
@skisssbb Seems like you boot directly from kernel and the kernel did not find init executable (or, a hard disk).

In my case with 1.4, I have to modify some code to make it work. In `misc/efi-stub/boot.c`, function `efi_main`, the `if (bootloader_name)` branch (line 402 in original release_1.4 branch) failed to identify correct bootloader name because on my machine the bootloader string did not end with '\0'. In that case it will fall back to the default bootloader name (which is defined on line 350). I have to change that default name to load my bootloader. Everything after that seemed fine. 

With 1.5, that problem was fixed because you can specify default bootloader name in `make menuconfig` step. I didn't touch any other code but 1.5 failed on my machine. I've tried to insert some printings but that trick only works before ACRN sets up virtual memory. 

Try to boot with `acrn.efi`. That will launch ACRN first and ACRN will launch whatever OS you've installed/configured as SOS. If you stuck on that, there's no point trying other options because ACRN wasn't even running.

ACRN does output error messages in early boot process through serial output (because ACRN targets devboards). If you can somehow get your serial output, then it will be a lot easier to debug the problem. But I doubt if any modern laptop still supports this without some external devices. 

From my debugging experiences, one possibility is that it was stuck on `detect_hardware_support` function in `cpu_caps.c`. I.e., some features required by your CPU were not supported (and unfortunately the error messages were sent to serial output).

fuzhongl, 2020-06-02:
@yfliuu 
The drivers for serio mouse/keyboard are not enabled in ACRN SOS kernel config. 
Please refer following steps to enable it with `make menuconfig`:

   Input device support  --->
       Hardware I/O ports  --->
           <*> Serial I/O support
           <*> i8042 PC Keyboard controller (NEW)
           <*> Serial port line discipline (NEW)

Thanks!

closed 2020-06-28 by fuzhongl