projectacrn/acrn-hypervisor #4127: Can't create Celadon image for user os

AlexMotyka, 2019-11-18:
Following the instructions for [Using Celadon as the User OS](https://projectacrn.github.io/latest/tutorials/using_celadon_as_uos.html) I was unable to build a Celadon image. 

Operating System used to create the image: Ubuntu 18.04

I tried numerous times to apply the patch links provided using git apply with no success. I attempted manually applying the commits but this did not work either. 

Lastly I tried running "make SPARSE_IMG=true gptimage -j $(nproc)" at step 4 of the **Build Celadon from source**  without applying the patches however this also failed to make the image, giving me the error: 
"ninja: build stopped: subcommand failed.
14:11:49 ninja failed with: exit status 1

####failed to build some targets (01:47 (mm:ss))#### "

Is there any solution available for creating an image of Caledon? If possible would anyone be able to provide me with a Caledon image that will be compatible with ACRN?

Any help is appreciated and thank you for your time! Please let me know if I could provide any more information about my situation.

All the best,

Alex

Mingyuan18, 2019-11-19:
[External_System_ID] ACRN-5705

ttzeng, 2019-11-19:
@AlexMotyka please make sure you have all required `apt-get` dependencies list in the [build from source](https://01.org/projectceladon/documentation/getting_started/build-source) guide, it's no harm to `sudo apt-get install .....` again to verify. In addition, I remember the `flex` package is also required in the later builds, but it's not documented. You may give it a try.

fuzhongl, 2020-05-05:
@AlexMotyka Could you please ask help in Celadon open source project for build fail issue?

fuzhongl, 2021-03-15:
Close it since no update for long time.

closed 2021-03-15 by fuzhongl