projectacrn/acrn-hypervisor #4132: ACRN Device Model (acrn-dm) version information is incorrect in Clear Linux

gvancuts, 2019-11-19:
The ACRN Device Model (`acrn-dm`) installed via the `service-os` bundle reports an incorrect version of itself. E.g.:
```
$ acrn-dm -v
DM version is: 1.4-unstable-”2019w45.5.140000p_250” (daily tag:”2019w45.5.140000p”), build by mockbuild@2019-11-12 19:18:43
```
However, tag 2019w45.5.140000p does not exist in the tree, acrn-2019w45.5-140000p is (I suppose) the correct tag information.

Note that this does not happen if you build `acrn-dm` manually, in that case, the version and tag info is correct, see:
```
$ ./build/devicemodel/acrn-dm -v
DM version is: 1.4-unstable-da469d9e (daily tag:acrn-2019w46.4-140000p), build by geoffroy@2019-11-19 14:16:53
```

gvancuts, 2019-11-19:
/cc @miguelinux

gvancuts, 2019-11-20:
@miguelinux , it looks like a problem that could (at least partially) stem from the wrong version info located in the `versions` file: https://github.com/clearlinux-pkgs/acrn-hypervisor/blob/master/versions

Perhaps this would help:
```
diff --git a/versions b/versions
index c005a7d..a928d13 100644
--- a/versions
+++ b/versions
@@ -1 +1 @@
-2019w47.1.140000p
+acrn-2019w47.1-140000p
```

Mingyuan18, 2019-11-25:
[External_Systen_ID] ACRN-5769

miguelinux, 2020-03-31:
@gvancuts I think we can close this one

look at https://github.com/clearlinux-pkgs/acrn-hypervisor/commit/4f12c1370433c9c472c927b30639252b3e192f18

gvancuts, 2020-04-08:
Yes, thanks @miguelinux !

closed 2020-04-08 by gvancuts