projectacrn/acrn-hypervisor #4139: mngr_client_new: Failed to accept from fd 38

paspndevops, 2019-11-21:
Ubuntu as SOS giving following error:

mngr_client_new: Failed to accept from fd 38, err: Invalid argument
Stop listening 38...
                Stop polling 38...
-------------------------------------
root@intel-desktop:~# acrn-dm -v
DM version is: 1.4-unstable-f657bae0 (daily tag:acrn-2019w47.3-140000p), build by root@2019-11-21 14:05:05
---------------------------------------------

file:///root/launch_uos.sh

Thanks!

paspndevops, 2019-11-21:
[launch_uos.txt](https://github.com/projectacrn/acrn-hypervisor/files/3874433/launch_uos.txt)

paspndevops, 2019-11-22:
@gvancuts 
Hi,
Could you please reply for above error i attached UOS script.

Thanks!

gvancuts, 2019-11-22:
Hi @paspndevops , can you copy/paste the entire output of the launch script?

Can you also confirm what version of `acrn.efi` you are using and how you compiled/installed it?

paspndevops, 2019-11-22:
@gvancuts 
DM version is: 1.4-unstable-f657bae0 (daily tag:acrn-2019w47.3-140000p), build by root@2019-11-21 14:05:05

cpu1 online=1
cpu2 online=1
cpu3 online=1
passed gvt-g optargs low_gm 64, high_gm 448, fence 8
SW_LOAD: get ovmf path /usr/share/acrn/bios/OVMF.fd, size 0x200000
pm by vuart node-index = 0
logger: name=console, level=4
logger: name=kmsg, level=3
logger: name=disk, level=5
vm_create: vm1
VHM api version 1.0
vm_setup_memory: size=0x80000000
open hugetlbfs file /run/hugepage/acrn/huge_lv1/vm1/D279543825D611E8864ECB7A18B34643
open hugetlbfs file /run/hugepage/acrn/huge_lv2/vm1/D279543825D611E8864ECB7A18B34643
level 0 free/need pages:0/1 page size:0x200000
level 1 free/need pages:0/2 page size:0x40000000
to reserve more free pages:
to reserve pages (+orig 0): echo 2 > /sys/kernel/mm/hugepages/hugepages-1048576kB/nr_hugepages
to reserve pages (+orig 0): echo 1 > /sys/kernel/mm/hugepages/hugepages-2048kB/nr_hugepages
now enough free pages are reserved!

try to setup hugepage with:
	level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
	level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x0
total_size 0x180000000

mmap ptr 0x0x7ff9216aa000 -> baseaddr 0x0x7ff940000000
mmap 0x80000000@0x7ff940000000
touch 2 pages with pagesz 0x40000000
mmap 0x200000@0x7ffa3fe00000
touch 1 pages with pagesz 0x200000

really setup hugepage with:
	level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
	level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x0
vm_init_vdevs
pci init hostbridge
pci init lpc
pci init pci-gvt
modify_bar_registration: bypass for pci-gvt
modify_bar_registration: bypass for pci-gvt
modify_bar_registration: bypass for pci-gvt
polling 38...
Listening 38...
pci init virtio-blk
pci init virtio-net
pci init virtio-console
pci init virtio-hyper_dmabuf
pci init virtio-rnd
tpm: init_vtpm2:Invalid socket path!
                                    /tmp/dm.XglrV0D     46: [0004]		Address : fec00000
                 Error    6303 -                          ^ Integer too large for target (00000000FEC00000 - max 1 bytes)

                                         /tmp/dm.XglrV0D     58: [0001]		Subtable Type : 02
                 Error    6302 -        Flag value is too large ^  (Maximum 1 bit)

  /tmp/dm.XglrV0D     62: [0004]		Interrupt : 00000009
                                                                    Error    6302 -    Flag value is too large ^  (Maximum 2 bit)

                                                 /tmp/dm.XglrV0D     63: [0002]Flags (decoded below) : 000D
                           Error    6302 -                Flag value is too large ^  (Maximum 2 bit)

                    acpi_build failed, error=-1
                                               pci deinit hostbridge
                                                                    pci deinit lpc
  pci deinit pci-gvt
                    modify_bar_registration: bypass for pci-gvt
                                                               modify_bar_registration: bypass for pci-gvt
                          modify_bar_registration: bypass for pci-gvt
                                                                     pci deinit virtio-blk
          pci deinit virtio-net
                               vtnet tx thread closing...
                                                         pci deinit virtio-console
  pci deinit virtio-hyper_dmabuf
                                pci deinit virtio-rnd
                                                     mngr_client_new: Failed to accept from fd 38, err: Invalid argument
                                        Stop listening 38...
                                                            Stop polling 38...

paspndevops, 2019-11-22:
@gvancuts 

  290  cd acrn-hypervisor-master/
  291  ls
  292  make all BOARD=nuc6cayh SCENARIO=sdc
  293  make install

Thanks!

paspndevops, 2019-11-22:
@gvancuts 
root@intel-desktop:~# bash launch_uos.sh 
cpu1 online=0
cpu2 online=0
cpu3 online=0
passed gvt-g optargs low_gm 64, high_gm 448, fence 8
SW_LOAD: get ovmf path /usr/share/acrn/bios/OVMF.fd, size 0x200000
pm by vuart node-index = 0
logger: name=console, level=4
logger: name=kmsg, level=3
logger: name=disk, level=5
vm_create: vm1
VHM api version 1.0
vm_setup_memory: size=0x80000000
open hugetlbfs file /run/hugepage/acrn/huge_lv1/vm1/D279543825D611E8864ECB7A18B34643
open hugetlbfs file /run/hugepage/acrn/huge_lv2/vm1/D279543825D611E8864ECB7A18B34643
level 0 free/need pages:1/1 page size:0x200000
level 1 free/need pages:2/2 page size:0x40000000

try to setup hugepage with:
	level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
	level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x0
total_size 0x180000000

mmap ptr 0x0x7f7dae391000 -> baseaddr 0x0x7f7dc0000000
mmap 0x80000000@0x7f7dc0000000
touch 2 pages with pagesz 0x40000000
mmap 0x200000@0x7f7ebfe00000
touch 1 pages with pagesz 0x200000

really setup hugepage with:
	level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
	level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x0
vm_init_vdevs
pci init hostbridge
pci init lpc
pci init pci-gvt
modify_bar_registration: bypass for pci-gvt
modify_bar_registration: bypass for pci-gvt
polling 38...
modify_bar_registration: bypass for pci-gvt
Listening 38...
pci init virtio-blk
pci init virtio-net
pci init virtio-console
pci init virtio-hyper_dmabuf
pci init virtio-rnd
tpm: init_vtpm2:Invalid socket path!
/tmp/dm.XorFnzV     46: [0004]		Address : fec00000
Error    6303 -                          ^ Integer too large for target (00000000FEC00000 - max 1 bytes)
/tmp/dm.XorFnzV     58: [0001]		Subtable Type : 02
Error    6302 -        Flag value is too large ^  (Maximum 1 bit)
/tmp/dm.XorFnzV     62: [0004]		Interrupt : 00000009
Error    6302 -    Flag value is too large ^  (Maximum 2 bit)
/tmp/dm.XorFnzV     63: [0002]Flags (decoded below) : 000D
Error    6302 -                Flag value is too large ^  (Maximum 2 bit)
 acpi_build failed, error=-1
pci deinit hostbridge
pci deinit lpc
  pci deinit pci-gvt
modify_bar_registration: bypass for pci-gvt
modify_bar_registration: bypass for pci-gvt
modify_bar_registration: bypass for pci-gvt
pci deinit virtio-blk
pci deinit virtio-net
vtnet tx thread closing...
pci deinit virtio-console
pci deinit virtio-hyper_dmabuf
pci deinit virtio-rnd
mngr_client_new: Failed to accept from fd 38, err: Invalid argument
Stop listening 38...
Stop polling 38...

gvancuts, 2019-11-22:
Thanks @paspndevops , I can see a couple of potential issues:
- The `iasl` version is not up to date, can you run `/usr/sbin/iasl -v` to check its version? (you need version 20190816 with the latest ACRN)
- You are using the wrong board when building ACRN, please use `nuc7i7dnb` instead of `nuc6cayh`

Mingyuan18, 2019-11-25:
[External_System_ID] ACRN-5769

paspndevops, 2019-11-25:
@gvancuts 

Thank you very much now ubuntu as SOS with clr-linux as uOS is working fine.

fuzhongl, 2019-11-25:
@paspndevops Could you please close this git issue?
Thanks!

closed 2019-11-25 by paspndevops