projectacrn/acrn-hypervisor #4148: AGL as User OS - Weston GUI not starting

KalevGonvick, 2019-11-25:
Hi I am having an issue with starting the Weston GUI on our AGL UOS.

There are no errors when launching the UOS.
We followed [this](https://projectacrn.github.io/latest/tutorials/agl-vms.html).
Changing the weston.ini and rebooting has no effect.

- Board: **APL-UP2**
- Service OS: **Clear Linux -  31470**
- ACRN Hypervisor: **1.0**
- ACRN Kernel: revision **acrn-2019w39.1-140000p**
- AGL Release: **Funky Flounder (6.02)**

The SOS is built from source for the apl-up2 board using the SDC scenario. I am not sure if this would have any effect, but the guide uses ACRN Hypervisor release 1.3, while we used 1.0.

**SIDE NOTE:**  Should the AGL VM have afm-utils already installed? I can't seem to find it.

Mingyuan18, 2019-12-02:
[External_System_ID] ACRN-5783

fuzhongl, 2019-12-02:
@KalevGonvick  Please share following log:
 uname -a and cat /proc/cmdline both in sos and uos
Thanks!

gvancuts, 2019-12-02:
@KalevGonvick , in addition the details asked by @fuzhongl , can you also explain how you installed version 1.0 of ACRN?

I believe the launch script you are using is really meant for systems running SBL (and launching a User VM using vSBL), @fuzhongl , can you confirm (or infirm) this? What Bios version is on your UP2.

You may also need to adjust all display output from the tutorial you followed as you're using an UP2 and the tutorial is for a NUC.

fuzhongl, 2019-12-02:
> I believe the launch script you are using is really meant for systems running SBL (and launching a User VM using vSBL), @fuzhongl , can you confirm (or infirm) this? What Bios version is on your UP2.
> 
@gvancuts  You are right, KalevGonvick  is  launching a User VM using vSBL.

@KalevGonvick Could you please try with UEFI version; instead of SBL?
Thanks!

KalevGonvick, 2019-12-02:
@fuzhongl @gvancuts 

- I launched the UOS using the script found on the tutorial found [here](https://projectacrn.github.io/latest/tutorials/agl-vms.html). I just thought that since that script was for the `NUC7i7DNHE`, that I should modify the apl-up2 script. I have not actually used the SBL script to launch, just the NUC one from that page.

- I installed ACRN by building from source using `BOARD=apl-up2` and `SCENERIO=sdc`. I used the command `efibootmgr -c -l "\EFI\acrn\acrn.efi" -d /dev/mmcblk0 -p 1 -L "ACRN Hypervisor" \
    -u "bootloader=\EFI\org.clearlinux\bootloaderx64.efi uart=bdf@0:18.1"'` after mounting `/dev/mmcblk0p1` to `/mnt` and placing my built `acrn.efi` file.

- BIOS version is 2.18.1263

- SOS uname -a result: `Linux clr-c599eb195aed40618117f07b34e226ff 4.19.78-95.iot-lts2018-sos #1 SMP PREEMPT Fri Oct 11 13:09:14 PDT 2019 x86_64 GNU/Linux`

- SOS cat /proc/cmdline result: `initrd=\EFI\org.clearlinux\freestanding-00-intel-ucode.cpio initrd=\EFI\org.clearlinux\freestanding-i915-firmware.cpio.xz root=PARTUUID=a027c374-f287-466a-b634-de212b5cfc35 quiet console=tty0 console=ttyS0,115200n8 consoleblank=0 cryptomgr.notests hvlog=2M@0x1FE00000 i915.avail_planes_per_pipe=0x01010F i915.domain_plane_owners=0x011111110000 i915.enable_guc=0 i915.enable_gvt=1 i915.nuclear_pageflip=1 ignore_loglevel init=/usr/lib/systemd/systemd-bootchart intel_iommu=igfx_off memmap=2M$0x1FE00000 no_timer_check no_timer_check noreplace-smp rcu_nocbs=0-64 rcupdate.rcu_expedited=1 rootfstype=ext4,btrfs,xfs rootwait tsc=reliable rw`

- AGL UOS uname -a result: `Linux intel-corei7-64 4.19.68-uos #1 SMP PREEMPT Thu Nov 28 03:12:45 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux`

- AGL UOS cat /proc/cmdline result: `root=/dev/vda2 rw rootwait maxcpus=1 nohpet console=tty0 console=hvc0   console=ttyS0 no_timer_check ignore_loglevel log_buf_len=16M   consoleblank=0 tsc=reliable i915.avail_planes_per_pipe=0x000F00   i915.enable_hangcheck=0 i915.nuclear_pageflip=1 i915.enable_guc_loading=0   i915.enable_guc_submission=0 i915.enable_guc=0`

gvancuts, 2019-12-02:
Hi @KalevGonvick , I assume that the AGL User VM starts up correctly, am I right? Is the issue that you do not see the Weston UI? If true, we need to look at the AcrnGT parameters, they are documented [here](https://projectacrn.github.io/latest/user-guides/kernel-parameters.html). And we also need to check the content of the weston.ini file. Do you have any logs of what happens inside the User OS (weston log, journalctl)?

KalevGonvick, 2019-12-02:
The AGL UOS Seems to launch fine. There is definitely an issue with Weston not being able to launch.

Here is the log:

[14:22:29.125] Command line: /usr/bin/weston --idle-time=0 --tty=1 --log=/run/platform/display/weston.log
[14:22:29.125] OS: Linux, 4.19.68-uos, #1 SMP PREEMPT Thu Nov 28 03:12:45 UTC 2019, x86_64
[14:22:29.125] warning: XDG_RUNTIME_DIR "/run/platform/display" is not configured
correctly.  Unix access mode must be 0700 (current mode is 770),
and must be owned by the user (current owner is UID 200).
Refer to your distribution on how to get it, or
http://www.freedesktop.org/wiki/Specifications/basedir-spec
on how to implement it.
[14:22:29.133] Using config file '/etc/xdg/weston/weston.ini'
[14:22:29.141] Output repaint window is 7 ms maximum.
[14:22:29.143] Loading module '/usr/lib/libweston-2/drm-backend.so'
[14:22:29.215] initializing drm backend
[14:22:29.225] logind: not running in a systemd session
[14:22:29.225] logind: cannot setup systemd-logind helper (-61), using legacy fallback
[14:22:29.265] using /dev/dri/card0
[14:22:29.276] Loading module '/usr/lib/libweston-2/gl-renderer.so'
[14:22:29.547] EGL client extensions: EGL_EXT_client_extensions
               EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses
               EGL_KHR_debug EGL_EXT_platform_wayland EGL_MESA_platform_gbm
[14:22:29.558] warning: neither EGL_EXT_swap_buffers_with_damage or EGL_KHR_swap_buffers_with_damage is supported. Performance could be affected.
[14:22:29.558] EGL_KHR_surfaceless_context available
[14:22:29.648] EGL version: 1.4 (DRI2)
[14:22:29.648] EGL vendor: Mesa Project
[14:22:29.648] EGL client APIs: OpenGL OpenGL_ES 
[14:22:29.648] EGL extensions: EGL_ANDROID_native_fence_sync EGL_EXT_buffer_age
               EGL_EXT_image_dma_buf_import EGL_KHR_config_attribs
               EGL_KHR_create_context EGL_KHR_fence_sync
               EGL_KHR_get_all_proc_addresses EGL_KHR_gl_renderbuffer_image
               EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_cubemap_image
               EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap
               EGL_KHR_no_config_context EGL_KHR_reusable_sync
               EGL_KHR_surfaceless_context EGL_KHR_wait_sync
               EGL_MESA_configless_context EGL_MESA_drm_image
               EGL_MESA_image_dma_buf_export EGL_WL_bind_wayland_display
[14:22:29.648] GL version: OpenGL ES 3.2 Mesa 17.1.7
[14:22:29.648] GLSL version: OpenGL ES GLSL ES 3.20
[14:22:29.648] GL vendor: Intel Open Source Technology Center
[14:22:29.648] GL renderer: Mesa DRI Intel(R) HD Graphics 505 (Broxton) 
[14:22:29.648] GL extensions: GL_EXT_blend_minmax GL_EXT_multi_draw_arrays
               GL_EXT_texture_filter_anisotropic
               GL_EXT_texture_compression_dxt1 GL_EXT_texture_format_BGRA8888
               GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24
               GL_OES_element_index_uint GL_OES_fbo_render_mipmap
               GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives
               GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float
               GL_OES_texture_float_linear GL_OES_texture_half_float
               GL_OES_texture_half_float_linear GL_OES_texture_npot
               GL_OES_vertex_half_float GL_EXT_texture_sRGB_decode
               GL_OES_EGL_image GL_OES_depth_texture
               GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV
               GL_OES_get_program_binary GL_APPLE_texture_max_level
               GL_EXT_discard_framebuffer GL_EXT_read_format_bgra
               GL_EXT_frag_depth GL_NV_fbo_color_attachments
               GL_OES_EGL_image_external GL_OES_EGL_sync
               GL_OES_vertex_array_object GL_OES_viewport_array
               GL_ANGLE_texture_compression_dxt3
               GL_ANGLE_texture_compression_dxt5 GL_EXT_robustness
               GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers
               GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil
               GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range
               GL_KHR_debug GL_KHR_robustness
               GL_KHR_texture_compression_astc_ldr
               GL_OES_depth_texture_cube_map GL_OES_surfaceless_context
               GL_EXT_color_buffer_float GL_EXT_separate_shader_objects
               GL_EXT_shader_framebuffer_fetch GL_EXT_shader_integer_mix
               GL_EXT_tessellation_point_size GL_EXT_tessellation_shader
[... 50 lines cut ...]
- wayland-0.lock
- weston.log (the log printed above)

Weston.ini looks like this:

[core]
shell=ivi-shell.so
backend=drm-backend.so
require-input=false
modules=systemd-notify.so

[output]
name=HDMI-A-1
transform=270
[id-agent]
default-id-offset=1000

[ivi-shell]
ivi-input-module=ivi-input-controller.so
ivi-module=ivi-controller.so
id-agent-module=simple-id-agent.so

[shell]
locking=true
panel-position=none

fuzhongl, 2019-12-03:
@KalevGonvick You can see the console of AGL, right?
If yes, please try following for Weston GUI:
1) $ vim /etc/xdg/weston/weston.ini
    Make these changes to weston.ini:
   Comment ivi-shell.so out
   Set the name of output is HDMI-A-2(if still not work, try HDMI-A-3)
2) After that, set up an environment variable and restart the weston service:
$ export XDG_RUNTIME_DIR=/run/platform/display
$ systemctl restart weston

Thanks!

KalevGonvick, 2019-12-08:
Hello, sorry for the late reply.

I will be away from home for a week or so, for now I will not be able to
try this out.

ErinRJ, 2019-12-31:
Hi Fuzhongl,

Sorry for the late reply again. I'm working with Kalev on this and I currently have the board.
We can see the AGL console as a command line type terminal right now.

Steps performed:
1. "shell=ivi-shell.so" was commented out. HDMI-A-1 was changed to HDMI-A-2, then to HDMI-A-3.
2. the two commands given to setup the environment variables and restart the weston service were executed.
When executing "systemctl restart weston", it doesn't appear to do anything, the console just moves on to the next line as if Enter was pressed. Is there something else we need to do to make the GUI appear after the changes to the weston.ini file are made?

Thanks for your time, and happy new year!

Erin

We also tried restarting the UOS once the changes were made, however this made no difference.

fuzhongl, 2020-01-01:
Hi Erin
Happy New Year!
https://projectacrn.github.io/latest/tutorials/agl-vms.html
As said in this link, the ACRN hypervisor is 1.3, not 1.0. and the Clear Linux OS is 31080 
We verified this guide on Kaby Lake NUC7i7DNHE; will try it on UP2 later.

Thanks!

KalevGonvick, 2020-01-16:
Hi again,

AGL GUI still does not work with the 1.3 hypervisor and Clear Linux 31080. We can only see the base shell.

The same errors occur as the logs above.