projectacrn/acrn-hypervisor #4149: Kaby Lake** NUC8i7HNB Build

khyberspache, 2019-11-25:
I've spent a while trying to get this running on a **Kaby Lake** NUC8i7HNB (it's the first generation of NUC8s, so it *should* work), but I have not been able to Get any configuration to boot past the below.  I attached the board config as a file.
```
detected 4 processes, 4 enabled
current on process 0
```
I have tried:
1. Default quick install script with a NUC7 board config (it failed as I assumed it would).
2. Custom built several EFIs...
- Building the acrn hypervisor with a generic board config w/ 32 GB RAM configuration
- Built a board specific configuration using the `board_config` tool and the SDC scenario (tried several variations of this)

General build approach is to use the quick config script to download the correct service OS version and replace the EFI image with my custom built images and attempt to boot (I'm only trying to get the SDC scenario booted, so this should be supported).  Are there other installation methods to try?

I've follow pretty much every build instruction in the Docs to no avail.  What would you recommend as next steps?  I did notice the board config produced by the script will cause the board_cfg_gen script to fail because of these two lines:
```
	Region 0: [virtual] Memory at 2000000000 (64-bit, prefetchable) [size=4G]
	Region 2: [virtual] Memory at 2100000000 (64-bit, prefetchable) [size=2M]
```
That is why I tried to do the generic board approach.  I tried patching the cfg_script to handle those lines (and tried both removing the lines and modifying them to non-virtual address spaces).

Pretty much have no idea where to go from here... any tips are appreciated. 

[NUC8i7HNB-Full.txt](https://github.com/projectacrn/acrn-hypervisor/files/3888515/NUC8i7HNB-Full.txt)

_Originally posted by @khyberspache in https://github.com/projectacrn/acrn-hypervisor/issues/3488#issuecomment-558321533_

jsun26intel, 2019-11-26:
Using configuration of generic board might work, but it would also lose some features.

Current code does not support well on 64bit BAR device. Could you please have a check in BIOS setting that find a configurable item which could change the memmap region from "above 4G" to "below 4G"? Usually it is a configuration of video card, if you set too large Aperture Size (e.g. > 1G), you might get 64bit BAR for some devices. So better to make sure the Aperture Size is <= 1G.

Another workaround is manually modify the BAR address to 0 in board file, this would block ACRN running on logical scenario or hybrid scenario, but SDC/Industry scenario is not relied on the BAR values. 

BTW, the patch which fix 64bit BAR parse error issue will be ready soon.

khyberspache, 2019-11-26:
@jsun26intel Thanks for getting back to me.  I dug through the HNKBLi7 BIOS and wasn't able to find anywhere to manual set iGD Aperture Size (you can only disable or enable) - there wasn't anywhere to adjust device memory mapping in general (strange).

I've pull 16Gb of RAM out of the NUC and will attempt to get the `BOARD=generic` SDC configuration to boot with as close to "stock" deployment as possible. If that doesn't work, I will set the BAR addresses in a fresh board config and try to deploy a new SDC scenario using that config.

Is there a PR out for the 64bit BAR parse error? I'd be interested in reviewing to see if that solve any of my problems.

khyberspache, 2019-11-26:
Update: Generic board config doesn't work.  Just remains stuck on the `detected 4 process, 4 enabled`.

Going to attempt to build a custom board config, zero out the 64bit BAR, and attempt to boot.

khyberspache, 2019-11-26:
Update: Generated another custom board config and attempted to build a stock SDC `acrn.efi` image and still hit the same error (tried zero-ing as you recommended).

Not sure where to go from here...

Mingyuan18, 2019-12-02:
[External_System_ID] ACRN-5787

fuzhongl, 2019-12-02:
@khyberspache  Could you please try -u "uart=disabled" in efibootmgr?
Thanks!

khyberspache, 2019-12-02:
@fuzhongl I will try that today

khyberspache, 2019-12-05:
Didn't have any luck with my existing EFI images. Ended up purchasing a NUC7 to get it working... will periodically poke at this.

khyberspache, 2019-12-12:
@fuzhongl I think I narrowed in on the issue - appears to be related to >1TB NVMe drives. I got a NUC7i7HNB board and tried to build an image using all of the same settings (with the 1TB NVMe Drive) and I couldn't get a working EFI image build.  

Once I pulled the NVMe drive, I could boot.

jsun26intel, 2019-12-13:
@khyberspache do you mean you replaced the 1T NVMe to a small one? Or you changed the root device with a SATA disk? If the native kernel support 1T, then ACRN kernel should support too.

BTW, please try the latest target board parser, 64bit BAR issue has been fixed. But with 64bit BAR you can only run ACRN on SDC/SDC2/INDUSTRY scenarios.

khyberspache, 2019-12-13:
I mean I just pulled out the 1T NVMe drive from the NUC7.  I had been installing to the SATA drive (/dev/sda) per the instructions, but having the NVMe drive in the NUC7 would prevent it from booting altogether.  

Haven't tested this in the NUC8, but I imagine it's the same issue.

fuzhongl, 2019-12-14:
@khyberspache  Do you set the boot order in BIOS? SATA or NVMe?
Thanks!

khyberspache, 2019-12-14:
I'm fairly certain I did... I'll double check and get back to you.

khyberspache, 2019-12-14:
@fuzhongl Double checked the BIOS boot order and it was SATA then NVMe.  Think I narrowed the issue down to specific hardware.  Using the NUC7i7DNHE platform, I swapped out the Intel 960P 1T NVMe M.2 drive for an Intel 760P 256G NVMe M.2 drive and was able to boot using my modified EFI image.  I'm going to try booting in the NUC8 using the 7-series drive as well.

Do you know why a 9 series drive would not work?

fuzhongl, 2019-12-15:
@khyberspache Sorry, we don't have  Intel 960P 1T NVMe M.2 SSD. Do you try to boot native with this 1T NVMe M.2 SSD? if it boots wells, please share "uname -a" and "cat /proc/cmdline"
Thanks!

khyberspache, 2019-12-18:
I had no problem booting into Clear Linux 31670 nor installing to the 1T SSD. I'll try to get you that data soon...

closed 2020-06-28 by fuzhongl