projectacrn/acrn-hypervisor #4161: The `acrnprobe.service` does not start correctly

gvancuts, 2019-11-28:
The `acrnprobe` service installed by the `service-os` bundle does not start correctly. One of the `Requires=` statement in the file is invalid.

```
$ sudo systemctl status acrnprobe
Password: 
● acrnprobe.service - ACRN crashlog probe
   Loaded: loaded (/usr/lib/systemd/system/acrnprobe.service; disabled; vendor preset: disabled)
   Active: inactive (dead)

Nov 28 16:37:32 kata-acrn-2 systemd[1]: /usr/lib/systemd/system/acrnprobe.service:4: Failed to add dependency on usercrash_s, ignoring: Invalid argument
Nov 28 16:37:44 kata-acrn-2 systemd[1]: /usr/lib/systemd/system/acrnprobe.service:4: Failed to add dependency on usercrash_s, ignoring: Invalid argument
Nov 28 16:51:43 kata-acrn-2 systemd[1]: /usr/lib/systemd/system/acrnprobe.service:4: Failed to add dependency on usercrash_s, ignoring: Invalid argument

```

```
$ cat /usr/lib/systemd/system/acrnprobe.service 
[Unit]
Description=ACRN crashlog probe
Requires=telemd.socket
Requires=usercrash_s
After=acrnlog.service
After=usercrash.service
After=prepare.service

[Service]
Type=simple
ExecStart=/usr/bin/acrnprobe

[Install]
WantedBy=multi-user.target
```

fuzhongl, 2020-05-04:
@wenlingz Please help to reproduce this issue?
Thanks!

gvancuts, 2021-02-23:
This is still an issue. The `acrnprobe` service does not start correctly.

It's not only the `Requires=usercrash_s` line that's causing the problem now though. The `acrnprobe` binary does not start anymore. The patch that removed the code for `libtelemetry` (https://github.com/projectacrn/acrn-hypervisor/commit/baea9ecb4477156585e277d6c91e934dcda44424) was incomplete and more is needed to make it completely clean.

@chengangc @dizhang417 I believe you are the owners of that tool, can you please take a look?

@NanlinXie , I'm surprised that our validation never caught the fact that the service does not even start. This bug was originally filed at the end of November in 2019. I can only assume that we do not validate `acrnprobe` at all, do we?