projectacrn/acrn-hypervisor #4195: HV: Add support to assign non-contiguous HPA regions for pre-launched VM

vijaydhanraj, 2019-12-05:
On some platforms, HPA regions for Virtual Machine can not be contiguous because of E820 reserved type or PCI hole. In such cases, pre-launched VMs need to be assigned non-contiguous memory regions and this patch addresses it.
To keep things simple, the current design has the following assumptions,
1. HPA2 always will be placed after HPA1
2. HPA1 and HPA2 don’t share a single ve820 entry. (Create multiple entries if needed but not shared)
3. Only support 2 non-contiguous HPA regions (can extend at a later point for multiple non-contiguous HPA)

vijaydhanraj, 2019-12-05:
[External_System_ID] ACRN-5792

NanlinXie, 2020-06-11:
Patch merged and no regression. Close it.

closed 2020-06-11 by NanlinXie