projectacrn/acrn-hypervisor #4247: Generate Scenario for VM0 communites with VM1 is incorrect

lirui34, 2019-12-13:
1.  git clone `https://github.com/projectacrn/acrn-hypervisor.git`
2. follow guide to launch config app: [https://projectacrn.github.io/latest/tutorials/acrn_configuration_tool.html#use-the-acrn-configuration-app]_
3. import default board info from misc/acrn-config/xmls/board-xmls/whl-ipc-i5.xml_
4. select scenario: industry_
```
|Modify Scenario Setting: VM0 communicate with VM1 only(ttyS1 0x2f8)
 config1:
     VM0: vuart1 base=SOS_COM2_BASE; target_vm_id=1
     VM1: vuart1 base=COM2_BASE; target_vm_id=0
 Notes: Default config0: 
     VM0: vuart1 base=SOS_COM2_BASE; target_vm_id=2
     VM1: vuart1 base=INVALID_COM_BASE
     VM2: vuart1 base=COM2_BASE; target_vm_id=0|
```
5. click "generate board src" and "generate scenario src"_
6. git show VM2's vuart1base is incorrect !

lirui34, 2019-12-13:
[External_System_ID] ACRN-5810

Mingyuan18, 2019-12-20:
Verify pass
HV tag: acrn-2019w51.2-140000p

closed 2019-12-20 by Mingyuan18