projectacrn/acrn-hypervisor #4249: Generated Launchscript but WebUI prompt error msg after we just select passthru-devices:audio_codec

lirui34, 2019-12-13:
**Steps**
_1,_ _cd acrn-hypervisor/misc/acrn-config/config_app_
_2, python3 app.py ->open WebUI_
_3, click "ImportBoardInfo" to select whl-ipc-i5.xml_
_4, select Scenario Setting:_ _industry_
_5, select Launch Setting:_ industry__launch_1uos_hardrt_
_6, select passthrough-devices: audio_codec only_
_7,_ Click "Generate Launch Script", WebUI will show error
8, But actually there's  ../acrn-hypervisor/misc/acrn-config/xmls/config-xmls/whl-ipc-i5/output/launch_uos_id2.sh
9, and We check launch_uos_id2.sh, it hasnot passthru-audio_codec code,just has audio_codes's pid&bdf

***Expected result***
_step8, should not generate launchscrip?_

***Actual result***
step8, But actually there's  ../acrn-hypervisor/misc/acrn-config/xmls/config-xmls/whl-ipc-i5/output/launch_uos_id2.sh
step9, and We check launch_uos_id2.sh, it hasnot passthru-audio_codec code,just has audio_codes's pid&bdf

lirui34, 2019-12-13:
[External_System_ID] ACRN-5837

Mingyuan18, 2019-12-20:
Verify pass: HV tag: acrn-2019w51.2-140000p

closed 2019-12-20 by Mingyuan18