projectacrn/acrn-hypervisor #4260: webcam switch between 2 UOS

paspndevops, 2019-12-16:
Hi,

How to switch my webcam from uos1 to uos2 

- i am using both uos's are headless 
- how to switch VM to VM 

Please guide me the steps for launcher script

Example:
**Bus 001 Device 002: ID 03f0:034a  xxxxx**

"ffmpeg -f video4linux2 -s 640x480 -i /dev/video0 -ss 0:0:2 -frames 1 /tmp/out.jpg"

Thanks !

Mingyuan18, 2019-12-17:
[External_System_ID] 5862

fuzhongl, 2019-12-17:
@paspndevops  Is it a USB webcam, or build-in webcam? 
For USB webcam, USB mediator can help on it.
Thanks!

paspndevops, 2019-12-19:
@fuzhongl 
It was USB Webcam could you please brief on USB mediator and how it will help.

Thank!

fuzhongl, 2019-12-19:
@paspndevops Please using dmesg in sos to check the USB port info of Webcam.
then add USB mediator option in UOS launch script;  
please refer following link for detail:

[](https://projectacrn.github.io/latest/tutorials/running_deb_as_user_vm.html)
`3Assign USB ports to the Debian VM in order to use the mouse and keyboard before the launch:

$ vim launch_debian.sh

<Add below as the acrn-dm parameter>
-s 7,xhci,1-2:1-3:1-4:1-5 \'  #it will assign USN port 2 3 4 and 5 to UOS.

closed 2019-12-19 by paspndevops