projectacrn/acrn-hypervisor #4278: Xenomai will get general protection fault when run /usr/xenomai/bin/latency -c1

Mingyuan18, 2019-12-19:
[   18.772166] general protection fault: 0000 1 SMP
  2 [   18.776261] Modules linked in: x86_pkg_temp_thermal
  1 [   18.782507] CPU: 1 PID: 1369 Comm: sampling-1366 Tainted: G        W       4.9.90-xenomai-3.0.7+ #23
37 [   18.789847] I-pipe domain: Linux
  1 [   18.792996] task: ffff88003c9a5100 task.stack: ffffc900017f4000
  2 [   18.798311] RIP: 0010:[<ffffffff81058c35>]  [<ffffffff81058c35>] xnarch_switch_fpu+0xc5/0x1d0
  3 [   18.805598] RSP: 0018:ffffc900017f7e28  EFLAGS: 00010206
  4 [   18.809765] RAX: 00000000ffffffff RBX: ffffc900011d3040 RCX: ffff88003c9a5ac0
  5 [   18.816074] RDX: 00000000ffffffff RSI: ffffc900011d3040 RDI: ffff88003ed251c0
  6 [   18.822404] RBP: ffffc900017f7e40 R08: 0000000000000000 R09: ffffffff81a15678
  7 [   18.828654] R10: 0000002a5e00c1d9 R11: 0000000000000000 R12: ffff88003c9a5100
  8 [   18.834930] R13: ffff88003c9a5100 R14: 0000000000000000 R15: ffffc900011d3040
  9 [   18.841326] FS:  00007f2873a9f700(0000) GS:ffff88003ed00000(0000) knlGS:0000000000000000
10 [   18.847576] CS:  0010 DS: 0000 ES: 0000 CR0: 000000008005003b
11 [   18.852781] CR2: 00007ffd18326080 CR3: 000000003adf9000 CR4: 00000000003406b0
12 [   18.859012] Stack:
13 [   18.861093]  ffffc900011d3040 ffff88003ed25000 ffff88003c9a5100 ffffc900017f7e78
14 [   18.870550]  ffffffff81130e58 ffffc900017f7f58 0000000000000006 0000000000000002
15 [   18.879925]  ffffc900011d3040 ffffffff81158460 ffffc900017f7eb0 ffffffff811507b6
16 [   18.889341] Call Trace:
17 [   18.891429]  [<ffffffff81130e58>] xnthread_harden+0x98/0x250
18 [   18.896635]  [<ffffffff81158460>] ? CoBaLt_thread_setschedparam_ex+0x60/0x60
19 [   18.902908]  [<ffffffff811507b6>] ipipe_syscall_hook+0x136/0x2f0
20 [   18.908239]  [<ffffffff810edfa6>] __ipipe_notify_syscall+0xc6/0x190
21 [   18.913452]  [<ffffffff811b4959>] ? do_sys_open+0x119/0x1f0
22 [   18.917617]  [<ffffffff810ee095>] ipipe_handle_syscall+0x25/0xb0
23 [   18.922827]  [<ffffffff81002a88>] do_syscall_64+0x28/0xf0
24 [   18.927068]  [<ffffffff81886d1b>] entry_SYSCALL_64_after_swapgs+0x58/0xc2
25 [   18.932349] Code: f6 83 88 11 00 00 02 75 ca 41 80 bc 24 85 09 00 00 00 75 29 41 81 64 24 3c ff df ff ff 5b 41 5c 41 5d 5d c3 b8 ff ff ff ff 89 c2 <48> 0f ae 29 f6 86 88 11 00 00 01
25 ↳ 48 89 f3 74 89 eb 9d 41 c6
26 84
27 [   19.002465] RIP  [<ffffffff81058c35>] xnarch_switch_fpu+0xc5/0x1d0
28 [   19.008716]  RSP <ffffc900017f7e28>
29 [   19.011913] --[ end trace cac76a4473c8f84e ]--

Mingyuan18, 2019-12-19:
[External_System_ID] ACRN-5864

fuzhongl, 2020-05-05:
@conghuic23 Can this git issue be closed?
Thanks!

closed 2020-06-28 by fuzhongl