projectacrn/acrn-hypervisor #4280: acrnctl stop VM1 and mouse pointer not visible

paspndevops, 2019-12-19:
Hi ,

When iam trying to stop /reset/suspend with acrnctl UOS not responded also let know how to resolve mouse pointer visibility on UOS.

Thanks !

Mingyuan18, 2019-12-20:
[External_System_ID] ACRN-5887

fuzhongl, 2019-12-20:
@paspndevops For UOS normal case acrnctl works well:
https://projectacrn.github.io/latest/misc/acrn-manager/README.html
You can add -f option if UOS is in abnormal status.
About UOS mouse cursor issue, please using acrn-kernel for SOS.
https://github.com/projectacrn/acrn-kernel

Thanks!

paspndevops, 2019-12-23:
@fuzhongl 

I tried to stop vm with -f but no response Please guide me how to compile acrn-kernel for NUC7i7BNB.(Which configuration file .config)

Thanks!

fuzhongl, 2019-12-23:
@paspndevops  Please refer following link; and  build acrn.efi for NUC7i7BNB with config-tool:
https://projectacrn.github.io/latest/tutorials/acrn_configuration_tool.html
Thanks!

paspndevops, 2019-12-28:
@fuzhongl 

UOS not **shutdown and reboot error** after adding RTVM parameters please provide the solution for UOS launcher script added here for acrn-dm:

acrn-dm -A -m $mem_size -s 0:0,hostbridge \
   --lapic_pt \
   --rtvm \
   -U 495ae2e5-2603-4d64-af76-d4bc5a8ec0e5 \
   -s 3,virtio-blk,./uosFC.img \
   -s 4,virtio-console,@stdio:stdio_port \
   $pm_channel $pm_by_vuart \
   --ovmf /usr/share/acrn/bios/OVMF.fd \
   hard_rtvm

Thanks !

fuzhongl, 2020-07-09:
@paspndevops  Could you please share RT kernel version?
Thanks!

fuzhongl, 2022-03-04:
Closed since it can't be reproduced with latest code and no feedback long time.

closed 2022-03-04 by fuzhongl