projectacrn/acrn-hypervisor #4315: UOS CPU Latency Fedora Server 30-rt9

paspndevops, 2019-12-27:
Hi,

I tried to capture cpu benchmark latency for 1hr on UBUNTU SOS and Fedora 30 server as UOS here is the following result:

here is my launcher script for your reference :

[Laun_UOS.txt](https://github.com/projectacrn/acrn-hypervisor/files/4004314/Laun_UOS.txt)

==========================
# Total: 017999984
# Min Latencies: 00007
# Avg Latencies: 00012
# Max Latencies: 03095
# Histogram Overflows: 00001
# Histogram Overflow at cycle number:
# Thread 0: 7286179
==============================

Thanks !

fuzhongl, 2019-12-27:
@paspndevops Please use SSD passthrough instead of virtio_blk in launch script; and share "uname -a " and "cat /proc/cmdline" in UOS.
Thanks!

paspndevops, 2019-12-27:
@fuzhongl 

[root@localhost ~]# cat /proc/version 
Linux version 5.2.17-rt9 (root@localhost.localdomain) (gcc version 9.2.1 20190827 (Red Hat 9.2.1-1) (GCC)) #2 SMP PREEMPT RT Wed Oct 16 17:15:26 PDT 2019

When i am trying to boot with SSD passtrough getting error for fedora-rt9-30 

Thanks!

fuzhongl, 2019-12-27:
> @fuzhongl
> 
> [root@localhost ~]# cat /proc/version
> Linux version 5.2.17-rt9 ([root@localhost.localdomain](mailto:root@localhost.localdomain)) (gcc version 9.2.1 20190827 (Red Hat 9.2.1-1) (GCC)) #2 SMP PREEMPT RT Wed Oct 16 17:15:26 PDT 2019
> 
> When i am trying to boot with SSD passtrough getting error for fedora-rt9-30
> 
> Thanks!

Maybe, it is because  fedora-rt9-30 kernel doesn't support NVMe SSD, please passthrough SATA SSD to UOS and install SOS in NVMe SSD.
Please cat /proc/cmdline; not cat /proc/version in UOS.
Thanks!

paspndevops, 2019-12-27:
@fuzhongl 

[root@localhost ~]# cat /proc/cmdline
BOOT_IMAGE=(hd0,gpt2)/vmlinuz-5.2.17-rt9 root=UUID=83fce51a-3126-4188-991a-470d882760ae ro rhgb quie

Could you please reply with modified launcher script.

Thanks!

paspndevops, 2019-12-27:
@fuzhongl 
root@intel-desktop:~#  lspci -v | grep -iE 'nvm|sata'
00:17.0 SATA controller: Intel Corporation Sunrise Point-LP SATA Controller [AHCI mode] (rev 21) (prog-if 01 [AHCI 1.0])
	Subsystem: Intel Corporation Sunrise Point-LP SATA Controller [AHCI mode]
	Capabilities: [a8] SATA HBA v1.0
3c:00.0 Non-Volatile memory controller: Intel Corporation Device 2522 (prog-if 02 [NVM Express])

fuzhongl, 2019-12-27:
@paspndevops Please follow [this link](
https://github.com/projectacrn/acrn-hypervisor/blob/master/devicemodel/samples/nuc/launch_hard_rt_vm.sh)
Add following cmdline in UOS:
rootwait nohpet console=hvc0 console=ttyS0 \
no_timer_check ignore_loglevel log_buf_len=16M consoleblank=0 \
clocksource=tsc tsc=reliable x2apic_phys processor.max_cstate=0 \
intel_idle.max_cstate=0 intel_pstate=disable mce=ignore_ce audit=0 \
isolcpus=nohz,domain,1 nohz_full=1 rcu_nocbs=1 nosoftlockup idle=poll \
irqaffinity=0

paspndevops, 2019-12-27:
@fuzhongl 
Thanks for your support now CPU latency little bit improved for 10Min but not expected i will try again with 1 hour and let you know the benchmark result.

Thanks !

paspndevops, 2019-12-27:
@fuzhongl 

For 20min i am getting following results:

 Total: 007499999
 Min Latencies: 00002
 Avg Latencies: 00006
 **Max Latencies: 00203**
 Histogram Overflows: 00000
 Histogram Overflow at cycle number:
 Thread 0:

# for pm setting
pm_channel="--pm_notify_channel uart "
pm_by_vuart="--pm_by_vuart tty,/dev/ttyS1"

acrn-dm -A -m $mem_size -s 0:0,hostbridge \
   -U 495ae2e5-2603-4d64-af76-d4bc5a8ec0e5 \
   -s 3,virtio-blk,./uosFC.img \
   -s 4,virtio-console,@stdio:stdio_port \
   $pm_channel $pm_by_vuart \
   --ovmf /usr/share/acrn/bios/OVMF.fd \
   -B "rootwait nohpet console=hvc0 console=ttyS0
no_timer_check ignore_loglevel log_buf_len=16M consoleblank=0
clocksource=tsc tsc=reliable x2apic_phys processor.max_cstate=0
intel_idle.max_cstate=0 intel_pstate=disable mce=ignore_ce audit=0
isolcpus=nohz,domain,1 nohz_full=1 rcu_nocbs=1 nosoftlockup idle=poll
irqaffinity=0"  hard_rtvm
}

fuzhongl, 2019-12-27:
@paspndevops Could you please "cat /proc/cmdline" to double check?
Then, try passthrough SATA SSD to UOS.
Thanks!

paspndevops, 2019-12-30:
@fuzhongl 

UOS stops at below point after passthrough SATA SSD to UOS:

root@intel-acrnubuntu:/home/intel/satya# bash launFed.sh 
cpu1 online=0
cpu2 online=0
cpu3 online=0
vm_create:  
VHM api version 1.0
vm_setup_memory: size=0x40000000
open hugetlbfs file /run/hugepage/acrn/huge_lv1/ /D279543825D611E8864ECB7A18B34643
open hugetlbfs file /run/hugepage/acrn/huge_lv2/ /D279543825D611E8864ECB7A18B34643
level 0 free/need pages:1/0 page size:0x200000
level 1 free/need pages:1/1 page size:0x40000000

try to setup hugepage with:
	level 0 - lowmem 0x0, biosmem 0x0, highmem 0x0
	level 1 - lowmem 0x40000000, biosmem 0x0, highmem 0x0
total_size 0x180000000

mmap ptr 0x0x7fd7d5006000 -> baseaddr 0x0x7fd800000000
mmap 0x40000000@0x7fd800000000
touch 1 pages with pagesz 0x40000000

really setup hugepage with:
	level 0 - lowmem 0x0, biosmem 0x0, highmem 0x0
	level 1 - lowmem 0x40000000, biosmem 0x0, highmem 0x0
vm_init_vdevs
No correct pm notify channel given
pci init hostbridge
pci init passthru
polling 34...
acrn-dm: passthru device 0/17/0 BAR 1: size[100] is expanded to page aligned [1000]

acrn-dm: passthru device 0/17/0 BAR 5: size[800] is expanded to page aligned [1000]

tpm: init_vtpm2:Invalid socket path!
Listening 34...
acpi_build failed, error=-1
pci deinit hostbridge
pci deinit passthru
vm_reset_ptdev_intx:0x10-b8, ioapic virpin=0.
mngr_client_new: Failed to accept from fd 34, err: Invalid argument
Stop listening 34...
Stop polling 34...
No correct pm notify channel given
launFed.sh: line 55: -s: command not found
======================UOS_launch.sh================

===========================================

# NVME pass-through
echo ${passthru_vpid["nvme"]} > /sys/bus/pci/drivers/pci-stub/new_id
echo ${passthru_bdf["nvme"]} > /sys/bus/pci/devices/${passthru_bdf["nvme"]}/driver/unbind
echo ${passthru_bdf["nvme"]} > /sys/bus/pci/drivers/pci-stub/bind

# for pm setting
pm_channel="--pm_notify_channel uart "
pm_by_vuart="--pm_by_vuart tty,/dev/ttyS1"

acrn-dm -A -m $mem_size -s 0:0,hostbridge \
   -U 495ae2e5-2603-4d64-af76-d4bc5a8ec0e5 \
   -s 2,passthru,0/17/0 \ 
   -s 3,virtio-blk,./uosFC.img \
   -s 4,virtio-console,@stdio:stdio_port \
   $pm_channel $pm_by_vuart \
   --ovmf /usr/share/acrn/bios/OVMF.fd \
   -B "rootwait nohpet console=hvc0 console=ttyS0
no_timer_check ignore_loglevel log_buf_len=16M consoleblank=0
clocksource=tsc tsc=reliable x2apic_phys processor.max_cstate=0
intel_idle.max_cstate=0 intel_pstate=disable mce=ignore_ce audit=0
isolcpus=nohz,domain,1 nohz_full=1 rcu_nocbs=1 nosoftlockup idle=poll
irqaffinity=0"  hard_rtvm
}

for i in `ls -d /sys/devices/system/cpu/cpu[1-99]`; do
        online=`cat $i/online`
        idx=`echo $i | tr -cd "[1-99]"`
        echo cpu$idx online=$online
        if [ "$online" = "1" ]; then
                echo 0 > $i/online
                echo $idx > /sys/class/vhm/acrn_vhm/offline_cpu
        fi
done

launch_hard_rt_vm

paspndevops, 2019-12-30:
@fuzhongl 

cat /proc/cmdline
BOOT_IMAGE=/boot/acrn/org.clearlinux.iot-lts2018-sos.4.19.78-98 root=UUID=25bc697e-3e39-4c0a-8b48-a75fdd191fd0 pci_devices_ignore=(0:18:1) console=tty0 console=ttyS0 rw rootwait ignore_loglevel no_timer_check consoleblank=0 i915.nuclear_pageflip=1 i915.avail_planes_per_pipe=0x010F0F i915.domain_plane_owners=0x011100001111 i915.enable_gvt=1 i915.enable_guc=0 hvlog=2M@0x1FE00000 quiet splash vt.handoff=1

Thanks!

fuzhongl, 2020-01-01:
@paspndevops  **acpi_build failed, error=-1** is in the log.
Could you please iasl  and acrn-dm version?
Thanks!

paspndevops, 2020-01-02:
@fuzhongl 

root@intel-desktop:~# /usr/sbin/iasl -v

Intel ACPI Component Architecture
ASL+ Optimizing Compiler/Disassembler version 20191018
Copyright (c) 2000 - 2019 Intel Corporation

root@intel-desktop:~# acrn-dm -v
DM version is: 1.4-unstable- (daily tag:), build by root@2019-11-25 09:11:52

Thanks!

fuzhongl, 2020-01-02:
@paspndevops 
root@kata-acrn-2~ # acrn-dm -v
DM version is: 1.5-unstable-5b4d676b-dirty (daily tag:**acrn-2019w48.1-140000p**), b                                                  uild by root@2019-11-28 06:26:55
Above is log from mine, Why no tag info in your acrn-dm -v log?
Thanks!

paspndevops, 2020-01-03:
@fuzhongl 
acrn-dm -v
DM version is: 1.5-unstable- (daily tag:), build by root@2020-01-02 15:04:53
-- Above is my DM Ver. and here i attached my latency for your refernce:

For Ubuntu { maxcpus=$2}

[image]

[image]

fuzhongl, 2020-01-03:
@paspndevops   Any difference between Ubuntu and clr-linux on your side?
The only difference is about kernel, but with same kernel cmdline, right?
Thanks!

paspndevops, 2020-01-03:
@fuzhongl 
Right common kernel for both clr-linux and Ubuntu-18 but in Ubuntu i am getting High latency for 1Hr comparing with clr-linux with 12Hrs.

Could you please provide any reason even i disabled all CRON jobs and timers in my Ubuntu Server , Still i am getting HIGH Latency.

I tested the Ubuntu Server on bare metal same NUC i got good result with Low Latency(With-out ACRN)

Thanks again for your quick reply !

fuzhongl, 2020-01-03:
@paspndevops  please help ""cat /proc/cmdline" in sos and uos.
We need this info fro double check.
Thanks!

paspndevops, 2020-01-03:
@fuzhongl 
## SOS ==>
root@intel-desktop:~# cat /proc/cmdline
BOOT_IMAGE=/boot/acrn/org.clearlinux.iot-lts2018-sos.4.19.78-98 root=/dev/mapper/ubuntu--vg-root pci_devices_ignore=(0:18:1) console=tty0 console=ttyS0 rw rootwait ignore_loglevel no_timer_check consoleblank=0 i915.nuclear_pageflip=1 i915.avail_planes_per_pipe=0x01010F i915.domain_plane_owners=0x011111110000 i915.enable_gvt=1 i915.enable_guc=0 hvlog=2M@0x1FE00000

## UOS ==>
BOOT_IMAGE=(hd0,gpt2)/vmlinuz-5.2.17-rt9 root=UUID=83fce51a-3126-4188-991a-470d882760ae ro rhgb quiet

paspndevops, 2020-01-08:
@fuzhongl 
[image]

Thanks!

fuzhongl, 2020-01-15:
@lyan3  Please help to answer this question: With the same kernel and cmdline on the same platfrom; but the latency of ubuntu rootfs is more bigger than clear linux rootfs.
Thanks!

paspndevops, 2020-02-24:
Thank you CPU latency issue resolved with Kernel boot arguments .

closed 2020-02-24 by paspndevops