projectacrn/acrn-hypervisor #4327: WiFi Config Error in Ubuntu18.04.1 LTS  SOS

paspndevops, 2020-01-02:
root@intel-desktop:~# iwconfig
lo        no wireless extensions.

virbr0    no wireless extensions.

sit0      no wireless extensions.

tap0      no wireless extensions.

virbr0-nic  no wireless extensions.

eno1      no wireless extensions.

acrn-br0  no wireless extensions.

root@intel-desktop:~# lsmod
Module                  Size  Used by
btusb                  45056  0
btrtl                  16384  1 btusb
btbcm                  16384  1 btusb
btintel                20480  1 btusb
bluetooth             565248  5 btrtl,btintel,btbcm,btusb
usbhid                 49152  0
ecdh_generic           24576  1 bluetooth
rtsx_pci_sdmmc         24576  0
snd_soc_skl           163840  0
e1000e                249856  0
snd_soc_skl_ipc        81920  1 snd_soc_skl
xhci_pci               16384  0
rtsx_pci               65536  1 rtsx_pci_sdmmc
tpm_tis                16384  0
snd_soc_sst_ipc        16384  1 snd_soc_skl_ipc
xhci_hcd              233472  1 xhci_pci
tpm_tis_core           20480  1 tpm_tis
snd_soc_sst_dsp        32768  1 snd_soc_skl_ipc
tpm                    53248  2 tpm_tis,tpm_tis_core
snd_hda_ext_core       24576  1 snd_soc_skl
snd_hda_core           81920  2 snd_hda_ext_core,snd_soc_skl
efivarfs               16384  1
root@intel-desktop:~# uname -a
Linux intel-desktop 4.19.78-98.iot-lts2018-sos #1 SMP PREEMPT Mon Nov 18 09:22:10 PST 2019 x86_64 x86_64 x86_64 GNU/Linux
root@intel-desktop:~# cat /etc/os-release
NAME="Ubuntu"
VERSION="18.04.1 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.1 LTS"

gvancuts, 2020-01-03:
What is your WiFi adapter make/model (`lspci` would help)?

Can you check in `dmesg` that it's recognized and enabled correctly?

paspndevops, 2020-01-06:
@gvancuts 
root@intel-desktop:~# lspci
00:00.0 Host bridge: Intel Corporation Celeron N3350/Pentium N4200/Atom E3900 Series Host Bridge (rev 0b)
00:02.0 VGA compatible controller: Intel Corporation Device 5927 (rev 06)
00:08.0 System peripheral: Intel Corporation Skylake Gaussian Mixture Model
00:14.0 USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21)
00:14.2 Signal processing controller: Intel Corporation Sunrise Point-LP Thermal subsystem (rev 21)
00:16.0 Communication controller: Intel Corporation Sunrise Point-LP CSME HECI #1 (rev 21)
00:17.0 SATA controller: Intel Corporation Sunrise Point-LP SATA Controller [AHCI mode] (rev 21)
00:1c.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port (rev f1)
00:1c.5 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #6 (rev f1)
00:1c.7 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #8 (rev f1)
00:1d.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #9 (rev f1)
00:1f.0 ISA bridge: Intel Corporation Device 9d4e (rev 21)
00:1f.2 Memory controller: Intel Corporation Sunrise Point-LP PMC (rev 21)
00:1f.3 Audio device: Intel Corporation Sunrise Point-LP HD Audio (rev 21)
00:1f.4 SMBus: Intel Corporation Sunrise Point-LP SMBus (rev 21)
00:1f.6 Ethernet controller: Intel Corporation Ethernet Connection (4) I219-V (rev 21)
3a:00.0 Network controller: Intel Corporation Wireless 8265 / 8275 (rev 78)
3b:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS5229 PCI Express Card Reader (rev 01)
3c:00.0 Non-Volatile memory controller: Intel Corporation Device 2522

Thanks!

gvancuts, 2020-01-06:
Can you also send the `dmesg` output?

paspndevops, 2020-01-06:
@gvancuts 

[log1.txt](https://github.com/projectacrn/acrn-hypervisor/files/4025996/log1.txt)

Here i attached my system log and i am using :
     description: Motherboard
       product: NUC7i7BNB
       vendor: Intel Corporation
       physical id: 0
       version: J31145-304
       serial: GEBN72200F0J
       slot: Default string
     *-firmware
          description: BIOS
          vendor: Intel Corp.
          physical id: 0
          version: BNKBL357.86A.0080.2019.0725.1139

Thanks!

gvancuts, 2020-01-06:
Thanks @paspndevops but the log file seems truncated. Can you try to capture a new one, that starts from the very beginning?

paspndevops, 2020-01-07:
@gvancuts 

Sorry for wrong log attached kindly find attached full dmesg log for your reference :

Thanks!
[log1.txt](https://github.com/projectacrn/acrn-hypervisor/files/4028844/log1.txt)

gvancuts, 2020-01-07:
Thanks. The `iwlwifi` driver should be loaded for this device. I wonder why it isn't. Can you try to load it manually (`sudo modprobe iwlwifi`)?

paspndevops, 2020-01-08:
@gvancuts 

I tried but not working with SOS when i tried with same Ubuntu with regular kernel it was working fine.

Thanks!

gvancuts, 2020-01-13:
What does it say when you try to load it in the SOS? Any specific error that can help us understand what is wrong?

paspndevops, 2020-01-23:
@gvancuts 
kernel 4.19.78-98.iot-lts2018-sos doesn't working with Realtek RTS5229

Thanks again for your support

gvancuts, 2020-02-03:
Hi @paspndevops , I've realised the Intel WiFi driver is turned off on the SOS kernel. Can you try to enable it and see if it makes the wifi operational.

The easiest would be to follow the build procedure described [here](https://projectacrn.github.io/latest/tutorials/building_acrn_in_docker.html#build-the-acrn-service-vm-kernel-in-docker) but with a different `kernel_config_uefi_sos` file that adds support for the `iwlwifi` driver. You may also need to download and install the firmware in case it's not readily available in the Ubuntu rootfs, you can download it here: https://wireless.wiki.kernel.org/_media/en/users/drivers/iwlwifi-8265-ucode-22.361476.0.tgz

Let us know if this helps!

gvancuts, 2020-02-03:
I have just rebuilt a kernel with the driver enabled, and ran it in a Clear Linux Service OS. I can now see the WiFi card (disclaimer: I have not tried to use it at all but at least it's now showing up whereas it wasn't before).

Here is the [sos-config-wifi.txt](https://github.com/projectacrn/acrn-hypervisor/files/4148503/sos-config-wifi.txt) file that I used with the ACRN kernel hosted [here](https://github.com/projectacrn/acrn-kernel)

paspndevops, 2020-02-04:
@gvancuts 

Thank you for your reply i will test and let you know the status..

fuzhongl, 2020-05-04:
@paspndevops ACRN doesn't support on NUC.
Will close this git issue now, if you really need ACRN to support WIFI on NUC, please contact with @terryzouhao .

closed 2020-05-04 by fuzhongl