projectacrn/acrn-hypervisor #4349: nvme nvme0: RPMBS No partitions

paspndevops, 2020-01-09:
Hi,

when i parse the Boot commands for NVMe parameters i am getting following error:

[    1.487820] nvme nvme0: RPMBS 0
[    1.488715] nvme nvme0: RPMBS No partitions
[    1.496687] GPT:Primary header thinks Alt. header is not at the end of the disk.
[    1.503090] GPT:20971519 != 28131327
[    1.504232] GPT:Alternate GPT header not at the end of the disk.
[    1.506435] GPT:20971519 != 28131327
[    1.507565] GPT: Use GNU Parted to correct GPT errors.
[    1.509174]  nvme0n1: p1 p2
[    1.646669] [drm] Cannot find any crtc or sizes
[    1.649960] md: Skipping autodetection of RAID arrays. (raid=autodetect will force)
[    1.652190] Waiting for root device /dev/vda3...

=========Launch.sh===========================

acrn-dm -A -m $mem_size -s 0:0,hostbridge \
  -s 2,pci-gvt -G "$2" \
  -s 5,virtio-console,@stdio:stdio_port \
  -s 6,virtio-hyper_dmabuf \
  -s 3,passthru,3c/0/0 \
  -s 7,virtio-rnd \
  $pm_channel $pm_by_vuart $pm_vuart_node \
  $logger_setting \
  --mac_seed $mac_seed \
  -U 38158821-5208-4005-b72a-8a609e4190d0 \
 -k /boot/acrn/org.clearlinux.iot-lts2018-preempt-rt.4.19.72-18 \
 -B "root=/dev/vda3 rw rootwait maxcpus=1 nohpet console=tty0 console=hvc0 \
 console=ttyS0 no_timer_check ignore_loglevel log_buf_len=16M \
 consoleblank=0 tsc=reliable i915.avail_planes_per_pipe=$4 \
 i915.enable_hangcheck=0 i915.nuclear_pageflip=1 i915.enable_guc_loading=0 \
 i915.enable_guc_submission=0 i915.enable_guc=0" $vm_name

gvancuts, 2020-01-09:
Where is the rootfs for your User VM?

Because you are using pass-through here, you should use something like `root=/dev/nvmen0n1p21 in your kernel command-line parameter (assuming the User VM is installed on that partition)

fuzhongl, 2020-02-13:
Base on following the log: the rootfs is in nvme0n1p2
nvme0n1 259:0 0 13.4G 0 disk
├─nvme0n1p1 259:1 0 512M 0 part
└─nvme0n1p2 259:2 0 9.5G 0 part

Should be root=/dev/nvme0n1p2

closed 2020-06-28 by fuzhongl