projectacrn/acrn-hypervisor #4366: Coffee Lake support ?

cwyark, 2020-01-15:
Hi , is there any plan to support Coffee Lake ? For autonomous robot controllers, they might pick more high computation rather than low power one. So I think supporting coffee lake might be a possible and good choice for ACRN, if robotic vendors they want to adopt such hypervisor technology to their software architecture.

terryzouhao, 2020-04-21:
Hi @cwyark , first thanks for your suggestion of Coffee Lake support on ACRN to extend ACRN usage in robotic scenario. Yes "Coffee Lake" is not in current  ACRN Supported HW List: https://projectacrn.github.io/latest/reference/hardware.html#hardware, but not means Coffee Lake cannot be supported on ACRN. Actually most CPU features are same with Whiskey Lake (4C/8T), so you can try to add "new board" config with "acrn config tool": read board->scenario/VM comfig->launch scripts.  
BTW, there maybe some problems of Coffee Lake (6C/12T) and your board config, feel free to submit questions in mailing list or issues in github, thanks.

fuzhongl, 2021-03-15:
enabled with v2.3

closed 2021-03-15 by fuzhongl