projectacrn/acrn-hypervisor #4369: RTVM option not working for Ubuntu/Fedora

paspndevops, 2020-01-16:
RTVM options are not working when we added into the UOS launcher script(--rtvm \ and --lapic_pt) for Ubuntu and Fedora30.

But same working fine with clr-linux UOS.

Errors :  
- Unable to Reboot/shutdown
- acrnctl stop not working
- latency not good with rtvm options for ubuntu/fedora

paspndevops, 2020-01-23:
@gvancuts 

I tried RTVM features with Fedora 30 server with rt9 as UOS and achieved latency Max 152 us for 3 Hrs using with below launcher script:

Kindly guide me how to TUNE latency on ACRN UOS

-	Result:

# Total: 029533223
# Min Latencies: 00002
# Avg Latencies: 00005
# Max Latencies: 00152
# Histogram Overflows: 00000
# Histogram Overflow at cycle number:
# Thread 0:

-	RTVM Parameters:

/usr/bin/acrn-dm -A -m $mem_size -s 0:0,hostbridge \
   --lapic_pt \
   --rtvm \
   --virtio_poll 1000000 \
   -U 495ae2e5-2603-4d64-af76-d4bc5a8ec0e5 \
    -s 2,virtio-blk,/root/RTVM/fc222.img \
   -s 3,virtio-console,@stdio:stdio_port \
   -s 8,virtio-net,tap1 \
   $pm_channel $pm_by_vuart \
-U 495ae2e5-2603-4d64-af76-d4bc5a8ec0e5 \
   --ovmf /usr/share/acrn/bios/OVMF.fd \
   -B "maxcpus=2 nohpet no_timer_check ignore_loglevel clocksource=tsc tsc=reliable x2apic_phys processor.max_cstate=0 \
intel_idle.max_cstate=0 intel_pstate=disable mce=ignore_ce audit=0 \
isolcpus=nohz,domain,1 nohz_full=1 rcu_nocbs=1 nosoftlockup idle=poll irqaffinity=0" hard_rtvm
}

Thanks in advance !

gvancuts, 2020-01-24:
Hi @paspndevops , are you saying that the exact same script and kernel but with a Clear Linux rootfs yields to much better results than when you switch to Fedora?

@fuzhongl provided recommendations in terms of kernel command-line arguments to be used in the User VM (see https://github.com/projectacrn/acrn-hypervisor/issues/4315#issuecomment-569204939), I see you're using them in the launch script (vis the `-B` argument). I'm really not sure however that this argument has any effect, instead you should configure these parameters via the Fedora bootloader (Grub). Can you try this? Please post the output of `cat /proc/cmdline` in the User OS to double-check these are in effect.

paspndevops, 2020-01-27:
@gvancuts  @fuzhongl 

-  Using with RTVM  in the launch script 'Unable to Reboot/shutdown' and 'acrnctl stop not working'

-  [root@localhost ~]# cat /proc/cmdline
BOOT_IMAGE=(hd0,gpt2)/vmlinuz-5.2.17-rt9 root=UUID=66ed30df-5959-4d29-b648-dc2926ca9b8b ro resume=UUID=a325009e-99ef-4d89-b4b6-b140340703a2 rhgb quiet
[root@localhost ~]# uname -a
Linux localhost.localdomain 5.2.17-rt9 #2 SMP PREEMPT RT Wed Oct 16 17:15:26 PDT 2019 x86_64 x86_64 x86_64 GNU/Linux

-  [root@localhost ~]# cat /etc/os-release
NAME=Fedora
VERSION="30 (Thirty)"
ID=fedora
VERSION_ID=30
VERSION_CODENAME=""
PLATFORM_ID="platform:f30"
PRETTY_NAME="Fedora 30 (Thirty)"
ANSI_COLOR="0;34"
LOGO=fedora-logo-icon

gvancuts, 2020-01-27:
@paspndevops can you modify your User VM image so that the Grub inside it includes the kernel parameters recommended by @fuzhongl ?

I haven't done that in Fedora 30 yet but a quick search indicates the right tool for the job would be `grubby` - see https://www.dedoimedo.com/computers/fedora-30-grub-persistent-changes.html for example.

paspndevops, 2020-01-28:
@gvancuts 
Thank you very much for your reply after modifying grubby i got some better latency metrics for 6Hrs. Let you know the result after final validation .

Thanks again!

gvancuts, 2020-01-28:
Sounds promising - thanks!

fuzhongl, 2020-02-23:
@paspndevops  Could you please close this ticket if no more issue?
Thanks!

paspndevops, 2020-02-24:
Hi,
Issue resolved boot kernel boot arguments thank you all

closed 2020-02-24 by paspndevops