projectacrn/acrn-hypervisor #4373: Pass-through of the PCI-serial problem

extramask93, 2020-01-20:
Hello, 
What we are trying to do is: Make serial port available to UOS (clear-linux)  via pass-through of the physical PCI serial (so it can talk to the external world). 
And what we did was:
* uart=disable in acrn.conf
* change sdc2/vm_configurations.c to:
.vuart[0] = {
	.type = VUART_PCI,
	.addr.port_base = {
				.b = 0x00,
				.d = 0x16,
				.f = 0x03,
	},
	.irq = 19,
}
in SOS_VM
And this works for SOS - we can communicate SOS->external device.
But leaving all above as is and making pass-through in the UOS launch file does not work(PCI serial is available in the UOS but no byte could be sent).
UOS ->| external device

Are we doing something wrong or is it a bug?
Thanks

For reference
vm_configurations.c : https://github.com/extramask93/acrn_config/blob/master/vm_configurations.c lines 31-38
launch_uos.sh: https://github.com/extramask93/acrn_config/blob/master/launch_uos.sh
line 123
device: NUC7i7DNHE + Serial controller: Intel Corporation Device 9d3d

gvancuts, 2020-01-21:
Hi @extramask93 , responses from the engineering team may be slow as this is the Chinese New Year.

Have you tried to also disable the use of the UART at the hypervisor level (i.e. not only in `acrn.conf` but also when starting `acrn.efi`)? A quick test would be to drop to an EFI shell and launch the hypervisor manually from there, by typing" `acrn.efi uart=disable`

extramask93, 2020-01-21:
@gvancuts 
Yes i have. To be honest i've made a mistake in the question above. Option 'uart=disable' was added to the acrn.efi at the hypervisor level not in the acrn.conf (communication for SOS would not work otherwise). 
Thanks :}

gvancuts, 2020-01-21:
Thanks for the clarification :)

I haven't played much with serial ports and stuff but have you tried to play with the `virtio-console` parameter (as described [here](https://projectacrn.github.io/latest/developer-guides/hld/virtio-console.html))? I.e. something like `-s 3,virtio-console,@tty:tty_port=/dev/ttyS1`?

A couple of additional thoughts/ideas for investigation:
- When tty is used, please make sure the redirected tty is sleeping, (e.g., by sleep 2d command), and will not read input from stdin before it is used by virtio-console to redirect guest output.
- I've seen reports that some devices do not work as expected if they do not support FLR, this may be the case for this PCI device?

fuzhongl, 2020-05-04:
@extramask93 Which platform are you using?
Could you please run "lspci -k " is SOS  after launching User VM?
Thanks!

closed 2020-06-28 by fuzhongl