projectacrn/acrn-hypervisor #4401: Compilation/make error on master branch

richlife98, 2020-02-14:
Hi there,
I'm trying to build ACRN codes/project downloaded on a public GitHub repository at https://github.com/projectacrn/acrn-hypervisor. However, there is a critical make error to terminate the make process as followings:

=====================================
Command: **make all BOARD=nuc6cayh SCENARIO=sdc RELEASE=0**
Error Msg:

-Wl,--start-group /home/wtbuild1604/workspace/acrn/acrn-hypervisor/build/hypervisor/modules/lib_mod.a /home/wtbuild1604/workspace/acrn/acrn-hypervisor/build/hypervisor/modules/boot_mod.a /home/wtbuild1604/workspace/acrn/acrn-hypervisor/build/hypervisor/modules/hw_mod.a /home/wtbuild1604/workspace/acrn/acrn-hypervisor/build/hypervisor/modules/vp_base_mod.a /home/wtbuild1604/workspace/acrn/acrn-hypervisor/build/hypervisor/modules/vp_dm_mod.a /home/wtbuild1604/workspace/acrn/acrn-hypervisor/build/hypervisor/modules/vp_trusty_mod.a /home/wtbuild1604/workspace/acrn/acrn-hypervisor/build/hypervisor/modules/vp_hcall_mod.a /home/wtbuild1604/workspace/acrn/acrn-hypervisor/build/hypervisor/modules/libdebug.a /home/wtbuild1604/workspace/acrn/acrn-hypervisor/build/hypervisor/modules/sys_init_mod.a -Wl,--end-group
/usr/bin/ld: warning: -z noreloc-overflow ignored.
/**usr/bin/ld: /home/wtbuild1604/workspace/acrn/acrn-hypervisor/build/hypervisor/modules/boot_mod.a(cpu_primary.o): relocation R_X86_64_32 against `boot_regs' can not be used when making a shared object; recompile with -fPIC
/home/wtbuild1604/workspace/acrn/acrn-hypervisor/build/hypervisor/modules/boot_mod.a: 無法加入符號: 數值不符**
collect2: error: ld returned 1 exit status
Makefile:463: recipe for target '/home/wtbuild1604/workspace/acrn/acrn-hypervisor/build/hypervisor/acrn.out' failed
make[1]: *** [/home/wtbuild1604/workspace/acrn/acrn-hypervisor/build/hypervisor/acrn.out] Error 1
make[1]: Leaving directory '/home/wtbuild1604/workspace/acrn/acrn-hypervisor/hypervisor'
Makefile:110: recipe for target 'hypervisor' failed
make: *** [hypervisor] Error 2

============================================
Is it a known issue and how to fix it??

Thanks

gvancuts, 2020-02-14:
What is your development environment? I did not see this when compiling on Clear Linux for example (which is the primary environment we use for testing)

richlife98, 2020-02-14:
Hi Gvancuts,
Thanks for the reply. It is build on Ubuntu 16.04 LTS. 
Basically, I follow the instructions on the below web-page to build ACRN.

https://projectacrn.github.io/latest/getting-started/building-from-source.html

gvancuts, 2020-02-14:
Did you update `binutils` to version 2.27?

If not, you can do it as follows:
```
wget https://mirrors.ocf.berkeley.edu/gnu/binutils/binutils-2.27.tar.gz \
    && tar xzvf binutils-2.27.tar.gz \
    && cd binutils-2.27 \
    && ./configure \
    && make \
    && make install \
    && cd .. \
    && rm -fr binutils-2.27 \
```

fuzhongl, 2020-05-04:
@richlife98 Could you please upgrade to Ubuntu 18.04?
Following Doc is verified on Ubuntu 18.04.
https://projectacrn.github.io/latest/getting-started/building-from-source.html

closed 2020-06-28 by fuzhongl