projectacrn/acrn-hypervisor #4423: Increase default value of MAX_PCI_DEV_NUM (Maximum number of MSI-X tables per device)

Matthias1102, 2020-02-24:
Some PCIe devices require MSI-X table sizes that are greater than ACRN's default value of 16. One example of such a device is the WD Black NVMe SSD. If such a device is present, ACRN does not boot and is trapped by an ASSERT statement in function pci_read_cap in file [hypervisor/hw/pci.c](https://github.com/projectacrn/acrn-hypervisor/blob/release_1.5/hypervisor/hw/pci.c#L448):
```
    ASSERT(pdev->msix.table_count <= CONFIG_MAX_MSIX_TABLE_NUM);
```
lspci shows that this device requires a MSI-X table size of 19:
```
$ lspci -s 02:00 -vv
02:00.0 Non-Volatile memory controller: Sandisk Corp WD Black NVMe SSD (prog-if 02 [NVM Express])
    Subsystem: Marvell Technology Group Ltd. Device 1093
    Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
    Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
    Latency: 0, Cache Line Size: 64 bytes
    Interrupt: pin A routed to IRQ 16
    NUMA node: 0
    Region 0: Memory at a1300000 (64-bit, non-prefetchable) [size=16K]
    ...
    Capabilities: [b0] MSI-X: Enable+ Count=19 Masked-
        Vector table: BAR=0 offset=00002000
        PBA: BAR=0 offset=00003000
    ...
```
To increase the number of supported PCIe devices, we suggest to change the default value of MAX_MSIX_TABLE_NUM in [hypervisor/arch/x86/Kconfig](https://github.com/projectacrn/acrn-hypervisor/blob/release_1.5/hypervisor/arch/x86/Kconfig#L298) from 16 to e.g. 32.

HSchroeder, 2020-02-24:
Ideally this would be detected when the $(board).xml is being created on the target and this value would be automatically updated in the generated defconfig when it is bigger than the ACRN default.

Matthias1102, 2020-02-25:
I agree with HSchroeder, having the MSI-X table size in the $(board).xml would be the ideal solution. However, the current $(board).xml configuration does not provide this information for PCIe devices. We only have:
```
    <PCI_DEVICE>
    : 
    02:00.0 Non-Volatile memory controller: Sandisk Corp WD Black NVMe SSD
    Region 0: Memory at a1300000 (64-bit, non-prefetchable) [size=16K]
    : 
    </PCI_DEVICE>
```
What is the best way to support PCIe devices that need MSI-X table sizes > 16 ?

gvancuts, 2020-02-25:
I believe the best course of action would be to define your own `board-name.config` and have that `Kconfig` parameter set to whatever is suitable for your board. (you can define your `*.config` file by using `make menuconfig` and then `make savedefconfig`)

HSchroeder, 2020-02-25:
That is the manual approach which we take today. ACRN config tool does exist to capture the board information and is creating the board.xml specificallly for our hardware. It just would be convenient if this device info would be captured in addition. Out of the board.xml ACRN tooling can already generate the board-name.config and it would stay fully automated. So maybe it is a feature request and not a bug blocking us.

NanlinXie, 2020-02-25:
Thanks @HSchroeder , we will add this configuration tool feature request as configuration tool enhancement.

fuzhongl, 2022-07-05:
MAX_PCI_DEV_NUM can be parsed in board inspector.

closed 2022-07-05 by fuzhongl