projectacrn/acrn-hypervisor #4433: Introduce SR-IOV feature for ACRN

liuyuan-GitHub, 2020-02-26:
SR-IOV(single root input/output virtualization) can isolate PCIe devices and
make the devices to have good performance, close to bare-metal. The SR-IOV
consists of two basic units, one is called PF(Physical Function) that supports
SRIOV PCIe extended capability and manages entire physical device, another is
called VF (Virtual Function) and is a "lightweight" PCIe function which is meant
to be given to a VM as a pass-through device. For more details, please refer to
PCI Express Base Specification Revision 4.0 Chapter 9.

[External_System_ID] ACRN-5933

ClaudZhang1995, 2020-02-27:
[External_System_ID] ACRN-5933

NanlinXie, 2020-06-11:
Patch merged and no regression. Close it.

closed 2020-06-11 by NanlinXie

wenlingz, 2020-09-10:
[External_System_ID] ACRN-5933