projectacrn/acrn-hypervisor #4496: Acrn Support Split Lock

taoyuhong, 2020-03-16:
When the destination of an atomic memory operation located in 2 cache lines, it is called a Splitlock Access. LOCK# bus signal is asserted for splitloc access which can reduce performance. #AC for Splitlock Access is a CPU feature, it allows rise alignment check exception #AC(0) instead of asserting LOCK#, that is helpful to detect Splitlock Access.

Consider Splitlock Accesses are bugs in HV, HV will report it and go though. Need to fix Splitlock in HV

taoyuhong, 2020-03-16:
[External_System_ID] ACRN-5915
[External_System_ID] ACRN-5916
[External_System_ID] ACRN-5917
[External_System_ID] ACRN-5918
[External_System_ID] ACRN-5919

NanlinXie, 2020-06-11:
Patch merged and no regression. Close it.

closed 2020-06-11 by NanlinXie