projectacrn/acrn-hypervisor #4503: doc: broken link to gechic portable monitor doc

dbkinder, 2020-03-18:
Our document https://projectacrn.github.io/latest/tutorials/agl-vms.html references documentation for the GeChic 1303I portable monitor that's no longer listed on their site (https://www.gechic.com/en/touch-monitor/).  What is the appropriate monitor to list now?  Perhaps the 1503I?

gvancuts, 2020-03-18:
/cc @kiendinh

kiendinh, 2020-03-19:
@dbkinder, I think the 1503I should work, since GeChic series have been using similar USB modules. I also used AGL with the GeChic 1502I, which is previous version of 1503I.

gvancuts, 2020-03-19:
> @dbkinder, I think the 1503I should work, since GeChic series have been using similar USB modules. I also used AGL with the GeChic 1502I, which is previous version of 1503I.

Unless someone has it (and time to test it), I'd recommend we use this one (and we can also add a note that we originally tested with a different one no longer available but this one **should** work).

deb-intel, 2020-03-19:
@dbkinder @gvancuts @NanlinXie @kiendinh Do you want me to go with this?

gvancuts, 2020-03-19:
> @dbkinder @gvancuts @NanlinXie @kiendinh Do you want me to go with this?

👍

fuzhongl, 2020-05-04:
@dbkinder Can this git issue be closed?
Thanks!

dbkinder, 2020-05-04:
@fuzhongl the link is still broken, so no. I can submit a PR that fixes the broken link (based on the comments) and then we can close this issue.

PR #4762 (2020-05-04): doc: fix broken link to touch monitor
Fixes: #4503
commit f9824261 (2020-05-04): doc: fix broken link to touch monitor

Fixes: #4503
closed 2020-05-04 by dbkinder